Koenkk/zigbee2mqtt #14498: [Feature request]: Add a setting to make the "Map" fixed (not moving by itself)

Photelegy, 2022-10-15:
### Is your feature request related to a problem? Please describe

**Please save the position of the devices in the map or add a switch/setting to enable this.**

With many devices in my home it would really help to be able to organize the Map in Zigbee2MQTT the way I want. 
This would really help to make it more individual clear.
So after I arranged them how they are in real-live (like a floor plan) I can easily see when there should be a better (more direct) connection or what connection should be stronger to figure out what and where could be a problem.

I think this should be fairly easy because the main functionality for this is already there and only needs to be saved.

### Describe the solution you'd like

I want to organize the Map approximately like the floor plan of my home. And this is possible like seen in this [screenshot](https://imgur.com/a/ka08jmV).
But after some time it goes back to the automatic organisation where everything is flouting and moving how it wants (every time it's different). So I just want that it keeps (saves) the positioning how I made it (maybe with a manual save-button).
And if there are new devices discovered they should just be added around it (so I can position them how I want).

### Describe alternatives you've considered

Maybe just store the position like a x-y-coordination-system with the host at [0; 0] or something like that and save there position in the settings. (So we could maybe even export the map-positioning by just copy-pasting the coordination-system)

The experting by copy-paste isn't really necessary.

### Additional context

I've asked on reddit for help about this: [my Reddit-Question] (https://www.reddit.com/r/homeassistant/comments/y4lpb1/how_to_fixate_the_map_in_zigbee2mqtt/?utm_source=share&utm_medium=web2x&context=3)
And got told to make a request here.

danruck, 2022-10-15:
This would be even better if there was an option to upload a floor plan image for the background.

closed 2022-11-22 by github-actions[bot]

jcwild, 2023-05-04:
I would really like this too.  Being able to save the map as it really exists would be a great feature.

polygon242, 2024-02-19:
i second the wish. 
router nodes in the (mean) ortientation from the coordinator would be a first start. 
i drag my routers in the place according to my floor plan and z2m should remember at least the bearing of the routers for the next time. The map should be less chaotic next time ;)

Comawhite54, 2024-05-11:
I confirm this would be nice. Also remembering the positions that you have set previously (or being able to reload them from a file)

MrJeromeV, 2024-11-20:
I totally agree with the previous posts.