Koenkk/zigbee2mqtt #14501: Unsupported device with Zigbee model 'TS011F' and manufacturer name '_TZ3000_typdpbpg'

rednasss, 2022-10-15:
### What happened?

Hello,

I am receiving multiple errors in the log while trying to add a Zigbee plug:
Warning Received message from unsupported device with Zigbee model 'TS011F' and manufacturer name '_TZ3000_typdpbpg'
Warning Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.
And this continues for a couple of times.

In the devices list it is shown as Manufacturer: Unsupported and Model: TS011F

But if I look in the devices list the TZ3000_typdpbpg is shown in it...
I am using Home Assistant with the Zigbee2mqtt add-on.
What can I do to fix this?

Thanks!

### Zigbee2MQTT version

1.18.1-1

### Adapter firmware version

0x26660700

### Adapter

Conbee II

JacekKac, 2022-10-15:
try 1.28.0-dev zigbee2mqtt

rednasss, 2022-10-17:
Thanks! Switched to the edge version of zigbee2mqtt in Home Assistant and it is now showing up correctly.
However, I am getting some errors right now.

In zigbee2mqtt I get the following errors for my smoke detectors:
`Error 2022-10-17 09:44:10 Failed to configure 'Smoke detector 1', attempt 1 (Error: Bind 0x0015bc0031014f05/35 genBasic from '0x00212effff051c07/1' failed (Error: status: 140) at DeconzAdapter.bind (/app/node_modules/zigbee-herdsman/src/adapter/deconz/adapter/deconzAdapter.ts:768:19) at Endpoint.bind (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:567:13) at Object.bind (/app/node_modules/zigbee-herdsman-converters/lib/reporting.js:40:9) at Object.configure (/app/node_modules/zigbee-herdsman-converters/devices/develco.js:425:13) at Configure.configure (/app/lib/extension/configure.ts:115:13))`

I see these errors also in the zigbee2mqtt log in Home Assistant, but also an error for the plug I am talking about:
```
Zigbee2MQTT:error 2022-10-17 09:41:35: Failed to configure '0xa4c1384ede67a06c', attempt 1 (Error: Read 0xa4c1384ede67a06c/1 genBasic(["manufacturerName","zclVersion","appVersion","modelId","powerSource",65534], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (no response received)
    at DeconzAdapter.sendZclFrameToEndpoint (/app/node_modules/zigbee-herdsman/src/adapter/deconz/adapter/deconzAdapter.ts:658:23)
    at runNextTicks (node:internal/process/task_queues:61:5)
    at processTimers (node:internal/timers:499:9)
    at Endpoint.sendRequest (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:299:20)
    at Endpoint.read (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:472:28)
    at Object.configure (/app/node_modules/zigbee-herdsman-converters/devices/tuya.js:2200:13)
    at Configure.configure (/app/lib/extension/configure.ts:115:13)
    at Configure.start (/app/lib/extension/configure.ts:68:13)
    at Controller.callExtensions (/app/lib/controller.ts:314:17)
    at Controller.start (/app/lib/controller.ts:153:9))
```

What can I do to fix these errors?

closed 2022-11-25 by github-actions[bot]