Koenkk/zigbee2mqtt #14510: Add Install Code Function Not Supported

mauschil, 2022-10-16:
### What happened?

I am trying to connect a Bosch Plug Compact BSP-FZ2 with my Zigbee2MQTT installation running on my Raspberry Pi 4 as an add-on in Home Assistant. I am running the Edge version of the add-on.

When trying to connect using the `Enter Install Code` functionality it will fail with the following error.

`Zigbee2MQTT:error 2022-10-16 15:22:42: Request 'zigbee2mqtt/bridge/request/install_code/add' failed with error: 'Add install code is not supported'`

`Zigbee2MQTT:debug 2022-10-16 15:22:42: Error: Add install code is not supported
    at DeconzAdapter.addInstallCode (/app/node_modules/zigbee-herdsman/src/adapter/deconz/adapter/deconzAdapter.ts:188:31)
    at Controller.addInstallCode (/app/node_modules/zigbee-herdsman/src/controller/controller.ts:230:28)
    at Zigbee.addInstallCode (/app/lib/zigbee.ts:314:29)
    at Bridge.installCodeAdd (/app/lib/extension/bridge.ts:258:27)
    at Bridge.onMQTTMessage (/app/lib/extension/bridge.ts:123:63)
    at EventEmitter.emit (node:events:525:35)
    at EventBus.emitMQTTMessage (/app/lib/eventBus.ts:109:22)
    at MQTT.onMessage (/app/lib/mqtt.ts:140:27)
    at WebSocket.<anonymous> (/app/lib/extension/frontend.ts:94:27)
    at WebSocket.emit (node:events:513:28)
Zigbee2MQTT:info  2022-10-16 15:22:42: MQTT publish: topic 'zigbee2mqtt/bridge/response/install_code/add', payload '{"data":{},"error":"Add install code is not supported","status":"error","transaction":"ulaaz-4"}'`

### What did you expect to happen?

The `Enter Install Code` to complete successfully and pair with the device.

### Zigbee2MQTT version

1.28.0-dev

### Adapter firmware version

26720700

### Adapter

ConBee II

danieledwardgeorgehitchcock, 2022-10-16:
It Looks like the install code functionality was added to the Deconz API in [this commit](https://github.com/dresden-elektronik/deconz-rest-plugin/commit/24c7ff9d3047b7a6bde698e56d734120b63025ee)

As for Zigbee2MQTT support, I think it has only been added so far for TI Z-Stack and Silicon Labs EZSP co-ordinators

MattWestb, 2022-10-17:
ZHA is also having support for install code for real Zigbee 3 devices as service in HA:
```
Service
Zigbee Home Automation: Permit
Allow nodes to join the Zigbee network.

Install Code
Install code of the joining device (must be used with source_ieee)
```

mauschil, 2022-10-17:
@danieledwardgeorgehitchcock Someone else reported getting it to work with a Sonoff Zigbee Dongle. Looks like Ill give that a try. https://github.com/Koenkk/zigbee2mqtt/issues/13069#issuecomment-1272267851

@MattWestb Unfortunately, it looks like ZHA isnt supporting the Bosch plug. When entering the Install Code and IEEE, it will throw a `NotImplementedError`.

MattWestb, 2022-10-17:
I think only EZSP have it implanted and perhaps de(F)CONZ but have not red the code.

danieledwardgeorgehitchcock, 2022-10-18:
Just done some further digging on this..

Looks like the function was stubbed out when adding install code support for EZSP and Z-Stack

[Here are the lines](https://github.com/Koenkk/zigbee-herdsman/blob/12fea9e30f0d4f603c8b317ab354e1a581a24432/src/adapter/deconz/adapter/deconzAdapter.ts#L187-L189)

MattWestb, 2022-10-18:
Then its only waiting for our Master **Barrista** @kirovilya getting time implanting it in Z2M for for EZSP coordinators (its on his ToDo list). Hi is having some Aqara devices that need it so i think hi is not sleeping well until its in place. 

You can adding support for Bosch plug in ZHA and it shall working OK (I think they is pity ZCL so shall not being so hard like most tuya devices).

kirovilya, 2022-10-18:
@MattWestb already implemented in dev/edge version z2m

gekberlin, 2022-11-21:
Any news on installing a device with z2m and deconz stick? Every time I try to install I still get the fault code: `'Add install code is not supported'`

Koenkk, 2022-11-21:
@gekberlin , @kirovilya is not maintaining the Deconz integration. Install codes are not supported by the Deconz stick AFAIK.

mmattel, 2022-11-27:
It seems that there is some progress, see:
https://github.com/dresden-elektronik/deconz-rest-plugin/pull/6565 (Add DDF for Bosch Thermostat II)

From my understanding, if they are adding a DDF, install codes must already work with the Conbee II, but I may be wrong.

I am also desperately waiting that z2m + Conbee II + install codes are working (for adding the Bosch Thermostat II)

closed 2023-01-04 by github-actions[bot]

opal06, 2023-03-17:
To bump this, it seems there has been some movement on deconz supporting install codes. Pull request has not been merged, but looks good: https://github.com/dresden-elektronik/deconz-rest-plugin/pull/6791

sniecha, 2023-03-19:
cant wait to hear its working :) my bosch thermostats are waiting :)

rouvenst, 2023-04-24:
Pull request is now merged into v2.22.0-beta of dresden-elektronik/deconz-rest-plugin.
Did you create a new issue yet to track the progress of supporting install codes with conbee stick @sniecha @opal06 ?

Hedda, 2023-05-12:
FYI, if using ConBee or RaspBee (deconz Zigbee Coordinator adapter from dresden elektronik) then see related issue -> https://github.com/Koenkk/zigbee2mqtt/issues/17492