Koenkk/zigbee2mqtt #14513: [Schneider dimmer and switches ] - No status received when using physical buttons (used to work many versions ago)

onewaylifestreet, 2022-10-16:
### What happened?

Schneider Dimmer and switch used to work many versions ago but now not receiving status when pressing physical button.
Sending command have no problems and it worked really well.

[U202DST600ZB](https://www.zigbee2mqtt.io/devices/U202DST600ZB.html#schneider%2520electric-u202dst600zb)
[U202SRY2KWZB](https://www.zigbee2mqtt.io/devices/U202SRY2KWZB.html#schneider%2520electric-u202sry2kwzb)
[U201DST600ZB](https://www.zigbee2mqtt.io/devices/U202DST600ZB.html#schneider%2520electric-u202dst600zb)
[U201SRY2KWZB](https://www.zigbee2mqtt.io/devices/U202SRY2KWZB.html#schneider%2520electric-u202sry2kwzb)

### What did you expect to happen?

When pressing physical button on device, it should report status back to zigbee2mqtt (onoff status, brightness, etc).

### How to reproduce it (minimal and precise)

It used to work many versions ago from when it was supported for the first time. But now it is not working.

### Zigbee2MQTT version

1.28.0-1

### Adapter firmware version

20220219

### Adapter

CC2652R

### Debug log

[log (1).txt](https://github.com/Koenkk/zigbee2mqtt/files/9795087/log.1.txt)

Koenkk, 2022-10-18:
Can you  try to reconfigure the device via the z2m frontend and see if that fixes it? (it is the yellow refresh icon)

onewaylifestreet, 2022-10-19:
Already tried but did’t help. Any idea?

Koenkk, 2022-10-19:
What is shown under the z2m frontend -> device -> reporting tab? Can you play a bit with the reporting interval?

onewaylifestreet, 2022-10-30:
As follows;
Thanks.
[image: image.png]

onewaylifestreet, 2022-10-30:
[image]

closed 2022-12-07 by github-actions[bot]