Koenkk/zigbee2mqtt #14515: [New device support]: Villeroy & Boch Home Automation Kit C0040000

tobiasfrank87, 2022-10-16:
### Link

https://www.villeroy-boch.de/bad-und-wellness/produkte/More-to-See-Lite-ZigBeeR-Home-Automation-Kit-fuer-Spiegel-Rechteck-C0040000.html

### Database entry

{"id":15,"type":"Router","ieeeAddr":"0x00124b002114263f","nwkAddr":15933,"manufId":0,"manufName":"TIEFFELUCE","powerSource":"Mains (single phase)","modelId":"EC1300","epList":[11,13],"endpoints":{"11":{"profId":49246,"epId":11,"devId":544,"inClusterList":[0,3,4,5,6,8,768],"outClusterList":[],"clusters":{"genBasic":{"attributes":{"modelId":"EC1300","manufacturerName":"TIEFFELUCE","powerSource":1,"zclVersion":1,"appVersion":1,"stackVersion":2,"hwVersion":1,"dateCode":"20131206","swBuildId":"1.1.1"}}},"binds":[],"configuredReportings":[],"meta":{}},"13":{"profId":49246,"epId":13,"devId":544,"inClusterList":[4096],"outClusterList":[4096],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":1,"stackVersion":2,"hwVersion":1,"dateCode":"20131206","swBuildId":"1.1.1","zclVersion":1,"interviewCompleted":true,"meta":{},"lastSeen":1665932138197,"defaultSendRequestWhen":"immediate"}

### Comments

This Zigbee Kit for Mirrors of Villeroy & Boch is not working. It looks like that the manufacturer of the Zigbee Kit is Tieffeluce. 
[image] It should be able to support white balancing.

### Supported color modes

coldwhite & warmwhite

### Color temperature range

{"colorTempPhysicalMin":155,"colorTempPhysicalMax":495}

Koenkk, 2022-10-18:
Added!

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2022-10-18 by Koenkk

waltf62, 2023-08-29:
I have the exact same kit and used to be able to connect to HUE bridge. Now it is not possible anymore. I got two additional mirrors and whenever I search for new lights with them switched on, the bridge looses connection and is unavailable for a few minutes. Soemthing chnaged with later software?

berndottgh, 2023-09-30:
> I have the exact same kit and used to be able to connect to HUE bridge. Now it is not possible anymore. I got two additional mirrors and whenever I search for new lights with them switched on, the bridge looses connection and is unavailable for a few minutes. Soemthing chnaged with later software?

HEy, have the same problem. Works with HA and old Ikea Bridge. With Hue no pairing. I am in contact with V&B they try to solve the problem. We have to wait.

waltf62, 2023-10-01:
V&B were entirely incompetent - they have absolutely no idea what they are talking about. The kit never worked properly, even when it originally connected with earlier SW version of the bridge. Every 20% or so it would not switch on. I now simply replaced the components with zigbee stuff from illuminize and it works perfectly every time. BTW, changed the wiring to permanent power as it was disconnected everytime you touched the sensor on the screen...

hayashin4me, 2023-12-31:
``

> V&B were entirely incompetent - they have absolutely no idea what they are talking about. The kit never worked properly, even when it originally connected with earlier SW version of the bridge. [...]

- hey Walt, have the same issue here with this device.
- can you maybe give me exact device name which component did u replace?
Thanks

berndottgh, 2023-12-31:
Hey, V&B updated the Product in Germany to a new firmware. You should try to get a new one from your seller.
I used meanwile a Controller from a Philips Hue Lightstrip. Works like a charme ….

hayashin4me, 2023-12-31:
[image]

hayashin4me, 2023-12-31:
[image]

waltf62, 2024-01-01:
[https://www.iluminize.com/en/shop/product/5110-zigbee-3-0-led-controller-multi-5-pulse-width-modulation-pwm-constant-voltage-for-light-strips-black-502?category=24#attr=2738,2741,2740](url)

waltf62, 2024-01-01:
[image]

waltf62, 2024-01-01:
Works flawlessly.