Koenkk/zigbee2mqtt #14519: 'Error: Cluster 'manuSpecificTuya_3' has no command '53280''

laoyuan0, 2022-10-17:
### What happened?

Hello, the data obtained from the device is as follows:
  type 'attributeReport', cluster 'genBasic', data '{"65506":31,"65508":0,"appVersion":69}' from endpoint 1 with groupID 0 
entity：{"ID":1,"profileID":260,"deviceID":4,"inputClusters":[0,3,4,5,6,57344,57345],"outputClusters":[25,10],"deviceNetworkAddress":15825,"deviceIeeeAddress":"0x0c4314fffe868e20","clusters":{"genBasic":{"attributes":{"65503":"???*\u0011","65506":31,"65508":0,"65533":2,"65534":0,"modelId":"TS0726","manufacturerName":"_TZ3000_dfl9kueg","powerSource":1,"zclVersion":3,"appVersion":69,"stackVersion":0,"hwVersion":1,"dateCode":""}},"genOnOff":{"attributes":{"20480":1,"onOff":1,"onTime":0,"offWaitTime":0,"tuyaBacklightMode":1,"moesStartUpOnOff":0}},"manuSpecificTuya_3":{"attributes":{"53264":0,"53280":0}}},"_binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00124b00237dfc19","endpointID":1}],"_configuredReportings":[{"cluster":6,"attrId":0,"minRepIntval":0,"maxRepIntval":3600,"repChange":0}],"meta":{},"pendingRequests":[]}

But I use:  meta.device.getEndpoint(1).command('manuSpecificTuya_3',0xD020,1, opt); throw : Error Publish 'set' 'tuya_data_point_test' to 'six key switch' failed: 'Error: Cluster 'manuSpecificTuya_3' has no command '53280''

The website of the switch is: https://developer.tuya.com/en/docs/iot/tuya-zigbee-scene-selector-access-standard?id=K9ik6zvma9qh6

### Zigbee2MQTT version

1.22.1

### Adapter firmware version

20211220

### Adapter

CC2652

laoyuan0, 2022-10-21:
It has been solved, thank you

closed 2022-10-21 by Koenkk

Ethanwain, 2022-11-19:
> It has been solved, thank you
能说说咋弄的吗？我也想把ts0726接入z2m，谢谢了