Koenkk/zigbee2mqtt #14520: Cannot configure Tuya smart switch w. power monitoring

MicheleSilla810, 2022-10-17:
### What happened?

I have a Aubess Tuya mini smart switch with power monitoring, like the dude from issue #14220 but i can't get z2m to recognise it
It's in the 1.28.0 added devices, but with that z2m version it says unsupported device
I tried the latest-dev, and it gets recognised, but only as a simple switch w/ power monitoring

here's the screenshot
[image]
Any clue?

Thanks

### Zigbee2MQTT version

1.28.0, 1.28.0-dev commit: ceeaaba

### Adapter firmware version

20211217

### Adapter

Slaesh CC2652RB

### Debug log

Debug in 1.28.0 stable version (where it doesn't connect)
warn  2022-10-17 15:45:31: Received message from unsupported device with Zigbee model 'TS000F' and manufacturer name '_TZ3000_xkap8wtb'

MicheleSilla810, 2022-10-30:
@Koenkk are there any updates regarding this? Thanks

Koenkk, 2022-10-30:
Could you check if the issue is fixed with the following external converter:

```js
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const ota = require('zigbee-herdsman-converters/lib/ota');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const utils = require('zigbee-herdsman-converters/lib/utils');
const e = exposes.presets;
const ea = exposes.access;

const definition =  {
    fingerprint: [{modelID: 'TS000F', manufacturerName: '_TZ3000_xkap8wtb'}],
    model: 'TS000F_power',
    description: 'Switch with power monitoring',
    vendor: 'TuYa',
    fromZigbee: [fz.on_off, fz.electrical_measurement, fz.metering, fz.ignore_basic_report],
    toZigbee: [tz.on_off],
    configure: async (device, coordinatorEndpoint, logger) => {
        const endpoint = device.getEndpoint(1);
        await endpoint.read('genBasic', ['manufacturerName', 'zclVersion', 'appVersion', 'modelId', 'powerSource', 0xfffe]);
        await reporting.bind(endpoint, coordinatorEndpoint, ['genOnOff', 'haElectricalMeasurement', 'seMetering']);
        await reporting.rmsVoltage(endpoint, {change: 5});
        await reporting.rmsCurrent(endpoint, {change: 50});
        await reporting.activePower(endpoint, {change: 10});
        await reporting.currentSummDelivered(endpoint);
        endpoint.saveClusterAttributeKeyValue('haElectricalMeasurement', {acCurrentDivisor: 1000, acCurrentMultiplier: 1});
        endpoint.saveClusterAttributeKeyValue('seMetering', {divisor: 100, multiplier: 1});
        device.save();
    },
    exposes: [e.switch(), e.power(), e.current(), e.voltage().withAccess(ea.STATE), e.energy()],
};

module.exports = definition;
```

- save this as file next to `configuration.yaml` as `ext_converter.js`
- add it to `configuration.yaml`:
```yaml
external_converters:
  - ext_converter.js
```
- start z2m, check if issue is fixed

MicheleSilla810, 2022-10-30:
Yes, it's kind of fixed. The 'current' field however doesn't work. It shows 158 amps, sometimes it goes to 500 when the load is turned on, sometimes it remains at 158A or so. Power and voltage seems fine otherwise

[image]
[image]

Koenkk, 2022-10-31:
Hmm strange, so it always stays at 158A or 500A? Then hardware probably doesn't support it or the fw is bugged.

MicheleSilla810, 2022-10-31:
No it actually goes a bit randomly, now it's showing 83 amps. The power in watts is correct and updates constantly. So maybe there is something with the current conversion

Koenkk, 2022-10-31:
@MicheleSilla810 I don't expect something is wrong with the conversion since it reports 158A when the value is 0W (then the A value should be 0). In case you have a TuYa gatway, can you check if it works with that?

MicheleSilla810, 2022-10-31:
I don't know, sometimes it jumps up and down with the power, sometimes it doesn't, strange.
No sadly i don't have a TuYa gateway, but i'll buy one to test this.
Does it need to be something special, or even random ones off AliExpress work?
like this "G06" one for 11€? https://it.aliexpress.com/item/1005001794162200.html

migsun, 2022-11-01:
Same here, after integrate the external converter now I can control my two switches.
This works, thank you for the given information.

But I get the similar strange values for power and current. My electrician would really cry at 78 Amps. ;)

[image]

Koenkk, 2022-11-01:
@MicheleSilla810 that is the correct gateway indeed.

sergeybelozorov, 2022-11-16:
Hi!

This is a working converter.

```
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    fingerprint: [{modelID: 'TS0001', manufacturerName: '_TZ3000_xkap8wtb'}],
        model: 'ZN369906_01',
        vendor: 'Aubess',
        description: 'Automatic Switch',
        toZigbee: [tz.on_off, tz.tuya_switch_power_outage_memory, tz.ts011f_plug_indicator_mode],
        fromZigbee: [fz.on_off, fz.electrical_measurement, fz.metering, fz.ignore_basic_report, fz.tuya_switch_power_outage_memory,fz.ts011f_plug_indicator_mode],
        exposes: [e.switch(), e.power(), e.current(), e.voltage().withAccess(ea.STATE),
                e.energy(), exposes.enum('power_outage_memory', ea.ALL, ['on', 'off', 'restore'])
                    .withDescription('Recover state after power outage'),
                exposes.enum('indicator_mode', ea.ALL, ['off', 'off/on', 'on/off']).withDescription('LED indicator mode')],
        options: [exposes.options.measurement_poll_interval()],
        configure: async (device, coordinatorEndpoint, logger) => {
            const endpoint = device.getEndpoint(1);
            await reporting.bind(endpoint, coordinatorEndpoint, ['genOnOff', 'haElectricalMeasurement', 'seMetering']);
            endpoint.saveClusterAttributeKeyValue('seMetering', {divisor: 100, multiplier: 1});
            endpoint.saveClusterAttributeKeyValue('haElectricalMeasurement', {
                acVoltageMultiplier: 1, acVoltageDivisor: 1, acCurrentMultiplier: 1, acCurrentDivisor: 1000, acPowerMultiplier: 1,
                acPowerDivisor: 1,
            });
            try {
                await reporting.currentSummDelivered(endpoint);
            } catch (error) {/* fails for some https://github.com/Koenkk/zigbee2mqtt/issues/11179 */}
            await endpoint.read('genOnOff', ['onOff', 'moesStartUpOnOff', 'tuyaBacklightMode']);
        },
        onEvent: tuya.onEventMeasurementPoll,
    };

module.exports = definition;
```

[image]

But without a **switch_type**

[image]

MicheleSilla810, 2022-11-16:
@sergeybelozorov hi!

the switch part works, but the power metering doesn't show up

[image]

sergeybelozorov, 2022-11-17:
Converter is not connected

MicheleSilla810, 2022-11-18:
It doesn't work. I did the same steps as with the converter from Koen. His converter does get recognised, and the device has no picture, but this converter doesn't, and the switch gets recognised normally as a TS000F

alworx, 2022-11-20:
I merged converter code from other devices and got the switch + power metering running with this custom converter:
```
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const ota = require('zigbee-herdsman-converters/lib/ota');
const utils = require('zigbee-herdsman-converters/lib/utils');
const e = exposes.presets;
const ea = exposes.access;

const fzLocal = {
    metering_skip_duplicate: {
        ...fz.metering,
        convert: (model, msg, publish, options, meta) => {
            if (utils.hasAlreadyProcessedMessage(msg)) return;
            return fz.metering.convert(model, msg, publish, options, meta);
        },
    },
    electrical_measurement_skip_duplicate: {
        ...fz.electrical_measurement,
        convert: (model, msg, publish, options, meta) => {
            if (utils.hasAlreadyProcessedMessage(msg)) return;
            return fz.electrical_measurement.convert(model, msg, publish, options, meta);
        },
    },
};

const definition = {
    fingerprint: [{modelID: 'TS000F', manufacturerName: '_TZ3000_xkap8wtb'}],
    model: 'TS000F_power',
    description: 'Smart relais (with power monitoring)',
    vendor: 'TuYa',
    whiteLabel: [],
    ota: ota.zigbeeOTA,
    fromZigbee: [fz.on_off, fz.electrical_measurement, fz.metering, fz.ignore_basic_report,
        fz.tuya_switch_power_outage_memory, fz.ts011f_plug_indicator_mode, fz.ts011f_plug_child_mode],
    toZigbee: [tz.on_off, tz.tuya_switch_power_outage_memory, tz.ts011f_plug_indicator_mode, tz.ts011f_plug_child_mode],
    configure: async (device, coordinatorEndpoint, logger) => {
        const endpoint = device.getEndpoint(1);
        await endpoint.read('genBasic', ['manufacturerName', 'zclVersion', 'appVersion', 'modelId', 'powerSource']);
        await reporting.bind(endpoint, coordinatorEndpoint, ['genOnOff', 'haElectricalMeasurement', 'seMetering']);
        await reporting.rmsVoltage(endpoint, {change: 5});
        await reporting.rmsCurrent(endpoint, {change: 50});
	try {
            await reporting.currentSummDelivered(endpoint);
        } catch (error) {/* fails for some https://github.com/Koenkk/zigbee2mqtt/issues/11179 */}
        endpoint.saveClusterAttributeKeyValue('haElectricalMeasurement', {acCurrentDivisor: 1000, acCurrentMultiplier: 1, acPowerDivisor: 1, acPowerMultiplier: 1});
        endpoint.saveClusterAttributeKeyValue('seMetering', {divisor: 100, multiplier: 1});
        device.save();
    },
    exposes: [e.switch(), e.power(), e.current(), e.voltage().withAccess(ea.STATE),
        e.energy(), exposes.enum('power_outage_memory', ea.ALL, ['on', 'off', 'restore'])
            .withDescription('Recover state after power outage'),
        exposes.enum('indicator_mode', ea.ALL, ['off', 'off/on', 'on/off', 'on'])
            .withDescription('Plug LED indicator mode'), e.child_lock()],
};

module.exports = definition;
```

Need to add the switch type though

MicheleSilla810, 2022-11-20:
Yes, i confirm this is working. The current reads about 0.08-0-07 amps more (also with no load), but the rest is working perfectly. 
[image]
Thanks a lot

migsun, 2022-11-20:
I have no luck with the converter :/

Zigbee2MQTT:error 2022-11-20 22:11:51: Failed to configure 'light_outdoor_01', attempt 1 (Error: Read 0xa4c138bc7ea3a6b3/1 genBasic(["manufacturerName","zclVersion","appVersion","modelId","powerSource"], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 40765 - 1 - 4 - 0 - 1 after 10000ms)
    at Timeout._onTimeout (/app/node_modules/zigbee-herdsman/src/utils/waitress.ts:64:35)
    at listOnTimeout (node:internal/timers:559:17)
    at processTimers (node:internal/timers:502:7))
Zigbee2MQTT:error 2022-11-20 22:11:52: Invalid message 'null', skipping...

alworx, 2022-11-30:
> I have no luck with the converter :/
>
> Zigbee2MQTT:error 2022-11-20 22:11:51: Failed to configure 'light_outdoor_01', attempt 1 (Error: Read 0xa4c138bc7ea3a6b3/1 [...]

had some troubles in the beginning as well - have you tried force removing and re-pairing the device?

migsun, 2022-12-01:
Yeah, looks better now. But I made a full reboot of the system and now it seems to work as designed. Thank you all.

alworx, 2022-12-02:
> Yes, i confirm this is working. The current reads about 0.08-0-07 amps more (also with no load), but the rest is working perfectly. 

Try `current_calibration`:
[image]

closed 2023-01-10 by github-actions[bot]