Koenkk/zigbee2mqtt #14523: Tuya TS0003_switch_module does not trigger individual channels

chwoa, 2022-10-17:
### What happened?

Hi!

I a pretty sure the [TS0003](https://www.zigbee2mqtt.io/devices/TS0003_switch_module.html) version has exactly the same issue that you fixed for its 2-Gang version in:
https://github.com/Koenkk/zigbee2mqtt/issues/13954

The manual switches do work fine, so it needs it needs some TuYa magic for that one too :) 

### What did you expect to happen?

Each gang to be switched individually.

### How to reproduce it (minimal and precise)

- Add TS0003 to your network.
- Toggle any of its switches from Zigbee2MQTT

### Zigbee2MQTT version

 1.28.0

### Adapter firmware version

20220219

### Adapter

SONOFF Zigbee 3.0 USB Dongle Plus-P

danieledwardgeorgehitchcock, 2022-10-18:
Could you test your theory by trying [this external converter](https://gist.github.com/danieledwardgeorgehitchcock/367a23270db3d1b6dcf5fd2b591909f3)?

chwoa, 2022-10-18:
Tested and confirmed! Thank you very much!
Lights switching on independently using your converter.

Maybe we need this for the TS004 too? I don't have one to test, though.

closed 2022-10-18 by Koenkk

mabadran, 2022-11-01:
Thanks all for your great support. 
I have the same problem with the ts0004 
https://www.zigbee2mqtt.io/devices/TS0004_switch_module.html#tuya-ts0004_switch_module
and currently using Zigbee2mqtt ver. 1.28.0

Is this already in the last official release?

danieledwardgeorgehitchcock, 2022-11-01:
Hi,

This is due to be in the November release (due any time now).

Please keep an eye out [here](https://github.com/Koenkk/zigbee2mqtt/releases) for the latest release - under the release notes, it will have a New supported devices section which should contain your device.

Koenkk, 2022-11-01:
It is out! 😄

broekieman, 2024-01-20:
I bought a Tuya Zigbee ts0004 and when push button 1 while button 4 is on, button 4 switches of. It's not possible to have multiple relays on at the same time.

Sconn7, 2024-08-13:
> I bought a Tuya Zigbee ts0004 and when push button 1 while button 4 is on, button 4 switches of. It's not possible to have multiple relays on at the same time. 

Is there a fix for this issue? Only one output ON at any one time!