Koenkk/zigbee2mqtt #14524: Sengled E1C-NB7 smart plug fails to interview

phidauex, 2022-10-18:
### What happened?

Sengled [E1C-NB7](https://www.zigbee2mqtt.io/devices/E1C-NB7.html) paired to TubeZB POE coordinator (CC2652P). Pairing appears successful, but interview fails, and device is left connected but nonfunctional. Attempts to switch the device result in a timeout error. LQI value populates, but other field values are `null`.

### What did you expect to happen?

Device would pair and interview, and then respond to switch commands normally.

### How to reproduce it (minimal and precise)

- Set Z2M to Pairing Allowed.
- Reset Sengled E1C-NB7 in proximity to the coordinator.
- After 5 purple flashes it will attempt to pair.
- After 3 purple flashes it is pairing.
- Z2M identifies the device.
- Interview fails.
- Device is left paired but nonfunctional.

### Zigbee2MQTT version

1.28

### Adapter firmware version

20220928

### Adapter

tubeszb-cc2652-poe-2022

### Debug log

```
2022-10-16T21:35:51.794Z zigbee-herdsman:controller:log Device '0xb0ce1814036bf443' joined
2022-10-16T21:35:51.794Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2022-10-16T21:35:51.794Z zigbee-herdsman:controller:log Device '0xb0ce1814036bf443' accepted by handler
2022-10-16T21:35:51.794Z zigbee-herdsman:controller:log Delete device '0xb0ce1814036bf443' joined, undeleting
Zigbee2MQTT:info  2022-10-16 15:35:51: Device '0xb0ce1814036bf443' joined
Zigbee2MQTT:info  2022-10-16 15:35:51: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0xb0ce1814036bf443","ieee_address":"0xb0ce1814036bf443"},"type":"device_joined"}'
Zigbee2MQTT:info  2022-10-16 15:35:51: MQTT publish: topic 'homeassistant/switch/0xb0ce1814036bf443/switch/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state","value_template":"{{ value_json.state }}"},{"topic":"zigbee2mqtt/0xb0ce1814036bf443/availability","value_template":"{{ value_json.state }}"}],"availability_mode":"all","command_topic":"zigbee2mqtt/0xb0ce1814036bf443/set","device":{"identifiers":["zigbee2mqtt_0xb0ce1814036bf443"],"manufacturer":"Sengled","model":"Smart plug with energy tracker (E1C-NB7)","name":"0xb0ce1814036bf443"},"name":"0xb0ce1814036bf443","payload_off":"OFF","payload_on":"ON","state_topic":"zigbee2mqtt/0xb0ce1814036bf443","unique_id":"0xb0ce1814036bf443_switch_zigbee2mqtt","value_template":"{{ value_json.state }}"}'
Zigbee2MQTT:info  2022-10-16 15:35:51: MQTT publish: topic 'homeassistant/sensor/0xb0ce1814036bf443/power/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state","value_template":"{{ value_json.state }}"},{"topic":"zigbee2mqtt/0xb0ce1814036bf443/availability","value_template":"{{ value_json.state }}"}],"availability_mode":"all","device":{"identifiers":["zigbee2mqtt_0xb0ce1814036bf443"],"manufacturer":"Sengled","model":"Smart plug with energy tracker (E1C-NB7)","name":"0xb0ce1814036bf443"},"device_class":"power","enabled_by_default":true,"entity_category":"diagnostic","name":"0xb0ce1814036bf443 power","state_class":"measurement","state_topic":"zigbee2mqtt/0xb0ce1814036bf443","unique_id":"0xb0ce1814036bf443_power_zigbee2mqtt","unit_of_measurement":"W","value_template":"{{ value_json.power }}"}'
Zigbee2MQTT:info  2022-10-16 15:35:51: MQTT publish: topic 'homeassistant/sensor/0xb0ce1814036bf443/energy/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state","value_template":"{{ value_json.state }}"},{"topic":"zigbee2mqtt/0xb0ce1814036bf443/availability","value_template":"{{ value_json.state }}"}],"availability_mode":"all","device":{"identifiers":["zigbee2mqtt_0xb0ce1814036bf443"],"manufacturer":"Sengled","model":"Smart plug with energy tracker (E1C-NB7)","name":"0xb0ce1814036bf443"},"device_class":"energy","enabled_by_default":true,"name":"0xb0ce1814036bf443 energy","state_class":"total_increasing","state_topic":"zigbee2mqtt/0xb0ce1814036bf443","unique_id":"0xb0ce1814036bf443_energy_zigbee2mqtt","unit_of_measurement":"kWh","value_template":"{{ value_json.energy }}"}'
Zigbee2MQTT:info  2022-10-16 15:35:51: MQTT publish: topic 'homeassistant/sensor/0xb0ce1814036bf443/linkquality/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state","value_template":"{{ value_json.state }}"},{"topic":"zigbee2mqtt/0xb0ce1814036bf443/availability","value_template":"{{ value_json.state }}"}],"availability_mode":"all","device":{"identifiers":["zigbee2mqtt_0xb0ce1814036bf443"],"manufacturer":"Sengled","model":"Smart plug with energy tracker (E1C-NB7)","name":"0xb0ce1814036bf443"},"enabled_by_default":false,"entity_category":"diagnostic","icon":"mdi:signal","name":"0xb0ce1814036bf443 linkquality","state_class":"measurement","state_topic":"zigbee2mqtt/0xb0ce1814036bf443","unique_id":"0xb0ce1814036bf443_linkquality_zigbee2mqtt","unit_of_measurement":"lqi","value_template":"{{ value_json.linkquality }}"}'
2022-10-16T21:35:51.850Z zigbee-herdsman:controller:log Device '0xb0ce1814036bf443' is already in database with different networkAddress, updating networkAddress
2022-10-16T21:35:51.851Z zigbee-herdsman:controller:log Interview '0xb0ce1814036bf443' start
Zigbee2MQTT:info  2022-10-16 15:35:51: Starting interview of '0xb0ce1814036bf443'
2022-10-16T21:35:51.879Z zigbee-herdsman:controller:device:log Interview - start device '0xb0ce1814036bf443'
2022-10-16T21:35:51.879Z zigbee-herdsman:controller:device:log Interview - skip node descriptor request for '0xb0ce1814036bf443', already got it
2022-10-16T21:35:51.879Z zigbee-herdsman:adapter:zStack:znp:SREQ --> ZDO - activeEpReq - {"dstaddr":16805,"nwkaddrofinterest":16805}
2022-10-16T21:35:51.879Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,4,37,5,165,65,165,65,36]
2022-10-16T21:35:51.880Z zigbee-herdsman:controller:log Device '0xb0ce1814036bf443' accepted by handler
2022-10-16T21:35:51.880Z zigbee-herdsman:controller:log Not interviewing '0xb0ce1814036bf443', completed 'false', in progress 'true'
Zigbee2MQTT:info  2022-10-16 15:35:51: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0xb0ce1814036bf443","ieee_address":"0xb0ce1814036bf443","status":"started"},"type":"device_interview"}'
2022-10-16T21:35:51.942Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,1,101,5,0,97]
2022-10-16T21:35:51.942Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,1,101,5,0,97]
2022-10-16T21:35:51.942Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 1 - 3 - 5 - 5 - [0] - 97
2022-10-16T21:35:51.942Z zigbee-herdsman:adapter:zStack:znp:SRSP <-- ZDO - activeEpReq - {"status":0}
2022-10-16T21:35:51.942Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2022-10-16T21:35:52.048Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,13,69,193,165,65,165,65,67,244,107,3,20,24,206,176,142,170]
2022-10-16T21:35:52.048Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,13,69,193,165,65,165,65,67,244,107,3,20,24,206,176,142,170]
2022-10-16T21:35:52.048Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 13 - 2 - 5 - 193 - [165,65,165,65,67,244,107,3,20,24,206,176,142] - 170
2022-10-16T21:35:52.048Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- ZDO - endDeviceAnnceInd - {"srcaddr":16805,"nwkaddr":16805,"ieeeaddr":"0xb0ce1814036bf443","capabilities":142}
2022-10-16T21:35:52.049Z zigbee-herdsman:controller:log Device announce '0xb0ce1814036bf443'
Zigbee2MQTT:info  2022-10-16 15:35:52: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0xb0ce1814036bf443","ieee_address":"0xb0ce1814036bf443"},"type":"device_announce"}'
2022-10-16T21:35:52.079Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2022-10-16T21:35:54.822Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,7,69,196,165,65,2,251,207,166,184,74]
2022-10-16T21:35:54.822Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,7,69,196,165,65,2,251,207,166,184,74]
2022-10-16T21:35:54.822Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 7 - 2 - 5 - 196 - [165,65,2,251,207,166,184] - 74
2022-10-16T21:35:54.822Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- ZDO - srcRtgInd - {"dstaddr":16805,"relaycount":2,"relaylist":[53243,47270]}
2022-10-16T21:35:54.822Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
[... 22 lines cut ...]
2022-10-16T21:35:57.886Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 1 - 3 - 4 - 1 - [0] - 100
2022-10-16T21:35:57.886Z zigbee-herdsman:adapter:zStack:znp:SRSP <-- AF - dataRequest - {"status":0}
[... cut ...]
2022-10-16T21:36:12.032Z zigbee-herdsman:controller:error Interview failed for '0xb0ce1814036bf443 with error 'Error: Interview failed because can not get active endpoints ('0xb0ce1814036bf443')'
Zigbee2MQTT:error 2022-10-16 15:36:12: Failed to interview '0xb0ce1814036bf443', device has not successfully been paired
Zigbee2MQTT:info  2022-10-16 15:36:12: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0xb0ce1814036bf443","ieee_address":"0xb0ce1814036bf443","status":"failed"},"type":"device_interview"}'
```

jerrm, 2022-10-18:
I've got 6 of that model plug.   Just got two more last week. I haven't had any issue with pairing. 

Sonoff ZBDongle-P with coordinator version 20220103. 

I know that's not particularly helpful.  Just adding a data point.

phidauex, 2022-10-18:
Thanks, that is a useful datapoint. Maybe I have a coordinator problem instead of a Z2M problem. At least this proves that the Z2M config file for the device works.

jerrm, 2022-10-18:
It's zigbee - could be the phase of the moon.

phidauex, 2022-10-19:
Well, despite the measurements, channel 11 seemed to be part of the problem. I moved to channel 25, and the two units were able to pair up, interview successfully, and now switch. They still do not report power or energy, but I'm going to leave them for a while and see if the network settles. Closing this issue for now, but I'll add any further troubleshooting tips for future searchers.

closed 2022-10-19 by phidauex