Koenkk/zigbee2mqtt #14526: eWeLink - WB01 - Push Button does not parse the string correctly

MiddleSiggy, 2022-10-18:
### Link

https://www.amazon.com/SNZB-01-Wireless-Supports-Connected-Required/dp/B08BF9YQK1/ref=sr_1_1?crid=1FR6RW31BHNCS&keywords=sonoff+push+button&qid=1666066700&sprefix=sonoff+push+button%2Caps%2C108&sr=8-1

### Database entry

{"id":35,"type":"EndDevice","ieeeAddr":"0x00124b002513268a","nwkAddr":22846,"manufId":0,"manufName":"eWeLink","powerSource":"Battery","modelId":"WB01","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":0,"inClusterList":[0,3,1],"outClusterList":[6,3],"clusters":{"genBasic":{"attributes":{"modelId":"WB01","manufacturerName":"eWeLink","zclVersion":0,"hwVersion":0}},"genPowerCfg":{"attributes":{"batteryVoltage":31,"batteryPercentageRemaining":200}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00124b0024c866e5","endpointID":1},{"cluster":1,"type":"endpoint","deviceIeeeAddress":"0x00124b0024c866e5","endpointID":1}],"configuredReportings":[{"cluster":1,"attrId":32,"minRepIntval":3600,"maxRepIntval":62000,"repChange":0},{"cluster":1,"attrId":33,"minRepIntval":3600,"maxRepIntval":62000,"repChange":0}],"meta":{}}},"appVersion":5,"hwVersion":0,"dateCode":"20211103","zclVersion":0,"interviewCompleted":true,"meta":{"configured":1796851078},"lastSeen":1666066437771,"defaultSendRequestWhen":"immediate"}

### Comments

It detects the device correctly, but when it tries to parse the push button it shows the string as being empty:

A Single and Double push of the button returns:
Empty string("")

Here are the events from the Logs in debug mode, you can see the single and double push
Info 2022-10-18 00:14:19MQTT publish: topic 'zigbee2mqtt/Push Sensor 10', payload '{"action":"single","battery":100,"last_seen":"2022-10-18T00:16:11-04:00","linkquality":3,"voltage":3000}'
Info 2022-10-18 00:14:22MQTT publish: topic 'zigbee2mqtt/Push Sensor 10', payload '{"action":"double","battery":100,"last_seen":"2022-10-18T00:16:14-04:00","linkquality":0,"voltage":3000}'

danieledwardgeorgehitchcock, 2022-10-18:
Actions on button devices are momentary so, in Z2M frontend, you may only see a quick flash of a device action before it returning to a blank string / N/A / empty.

From the logs, it looks as though the device is functioning as it should, and the correct action is being parsed

Koenkk, 2022-10-18:
Read https://www.zigbee2mqtt.io/guide/faq/#why-is-the-action-property-always-empty why this happens.

closed 2022-10-18 by Koenkk

MiddleSiggy, 2022-10-19:
I got it to work with a trigger object in node-red thank you