Koenkk/zigbee2mqtt #14534: [New device support]: Bosch Water Alarm

jonasled, 2022-10-18:
### Link

https://www.bosch-smarthome.com/de/de/produkte/geraete/wassermelder/

### Database entry

{     "id": 42,     "type": "Unknown",     "ieeeAddr": "0x84fd27fffec99588",     "nwkAddr": 62489,     "epList": [],     "endpoints": {},     "interviewCompleted": false,     "meta": {},     "lastSeen": 1666084358153,     "defaultSendRequestWhen": "immediate" }

### Comments

Hi,
I try to add support for the Bosch BWA-1 Water leak sensor. I've already set the install code, but the device keeps stuck in interview. My Z2M Version is the latest stable (1.28.0)

jonasled, 2022-10-18:
Maybe this device is a bit more compicated to pair than the other Bosch devices: https://forum.phoscon.de/t/bosch-water-sensor-bwa-1-how-to-set-up-using-install-code/1698/7

danieledwardgeorgehitchcock, 2022-10-18:
Hi,

What co-ordinator device are you using?

Also, some devices (such as the Outdoor Siren) have two modes for pairing (Smart and Other) - I haven't investigated your device in particular but does it have any dip switches or references in the manual to different setup methods? - when setting up my Outdoor Siren, one mode presented behaviour in a similar way to what your device is..

jonasled, 2022-10-18:
Hi, 
The coordinator uses a Silicon Labs EFR32MG21 (Sonoff Zigbee Bridge) chip.
The water Sensor doesn't have any switches, you can only insert batteries and press a button on the bottom for pairing.

danieledwardgeorgehitchcock, 2022-10-18:
Could you enable the debug logs and see what is reported during the install code add and pair process?

closed 2022-11-25 by github-actions[bot]