Koenkk/zigbee2mqtt #14535: [New device support]: Müller tint e14 45730

skrause-data, 2022-10-18:
### Link

https://tool.mueller-licht.net/web/pdf/?45730.pdf

### Database entry

{"id":35,"type":"Router","ieeeAddr":"0xa4c13800ac4c1048","nwkAddr":21915,"manufId":4635,"manufName":"MLI","powerSource":"Mains (single phase)","modelId":"Candle white+color","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":269,"inClusterList":[0,3,4,5,6,8,768,4096,4111,65376],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":0,"stackVersion":2,"hwVersion":0,"swBuildId":"MDC37501","zclVersion":3,"interviewCompleted":false,"meta":{},"lastSeen":1666090820617,"defaultSendRequestWhen":"immediate"}

### Comments

Yesterday I bought a Zigbee 3.0 / Bluetooth light bulb in Aldi (German supermarket discount store). According to the data sheet, this bulb seems to be produced exclusively for Aldi.

I can pair the bulb, but the subsequent interview fails.

The bulb seems to have a Zigbee and Bluethooth model. Other bulbs from manufacturer Müller-Licht seem to work with zigbee2mqtt, as you could read here.

LOG:

`info  2022-10-18 12:57:01: Starting interview of '0xa4c13800ac4c1048'
info  2022-10-18 12:57:01: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0xa4c13800ac4c1048","ieee_address":"0xa4c13800ac4c1048","status":"started"},"type":"device_interview"}'
debug 2022-10-18 12:57:03: Received MQTT message on 'zigbee2mqtt/bridge/request/device/configure' with data '{"id":"0xa4c13800ac4c1048","transaction":"69oku-2"}'
info  2022-10-18 12:57:03: MQTT publish: topic 'zigbee2mqtt/bridge/response/device/configure', payload '{"data":{"id":"0xa4c13800ac4c1048"},"error":"Device '0xa4c13800ac4c1048' cannot be configured","status":"error","transaction":"69oku-2"}'`

### External converter

```shell
SONOFF Zigbee 3.0 USB Dongle Plus TI CC2652P + CP2102N
Zigbee2MQTT Version 1.28.0
Coordinator Typ: zStack3x0
Coordinator Version: 20220219
```

### Supported color modes

white + color

### Color temperature range

1800-6500 K

Koenkk, 2022-10-19:
Please add the color temperature range as described on https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html#_3-adding-converter-s-for-your-device

skrause-data, 2022-10-20:
Hi Koenkk,

I cannot fully pair the bulb, since the interview fails. Thus, I am not able to read the color temperature.

Maybe, it is the same color temperature like the bulb's older versions without bluthooth: https://www.zigbee2mqtt.io/devices/44435.html

Koenkk, 2022-10-20:
Can you try pairing it closer to the coordinator?

skrause-data, 2022-10-20:
Hi,

I tried to put the bulb right next to the coordinator (5 cm distance) and I even deactivated my wifi in order to avoid possible interference. Could the bulb be broken?

Koenkk, 2022-10-21:
I'm not sure why it fails to pair maybe the bulb is broken indeed. Can you try upgrading your coordinator firmware? https://github.com/Koenkk/Z-Stack-firmware/blob/develop/coordinator/Z-Stack_3.x.0/bin/CC1352P2_CC2652P_launchpad_coordinator_20220928.zip

skrause-data, 2022-10-24:
Hi, 

I think the bulb was broken, I returned it. I bought the older [model](https://www.zigbee2mqtt.io/devices/404036_45327_45317.html#m%25C3%25BCller%2520licht-404036%252F45327%252F45317) that works fine now. Thank you anyway for all the input. :)

closed 2022-10-24 by Koenkk