Koenkk/zigbee2mqtt #14537: UNSUPPORTED_ATTRIBUTE - manuSpecificSamsungAccelerometer reporting update

GSzabados, 2022-10-18:
### What happened?

I wanted to update the reporting interval on a Samsung/SmartThings/CentraLite Multi Sensor (2015 model) but the result is UNSUPPORTED_ATTRIBUTE.

I don't know that is it supposed to work, but I guess it should because values provided in the reporting tab.

### What did you expect to happen?

Able to change reporting attributes. 

### How to reproduce it (minimal and precise)

Try to update reporting attributes. 

### Zigbee2MQTT version

1.28.0

### Adapter firmware version

20210708

### Adapter

zStack3x0

### Debug log

Error 2022-10-18 16:14:41Request 'zigbee2mqtt/bridge/request/device/configure_reporting' failed with error: 'ConfigureReporting 0x000d6f000bbd6160/1 manuSpecificSamsungAccelerometer([{"attribute":"z_axis","minimumReportInterval":10,"maximumReportInterval":"600","reportableChange":1}], {"sendWhen":"active","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":4362,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE')'
Info 2022-10-18 16:14:41MQTT publish: topic 'zigbee2mqtt/bridge/response/device/configure_reporting', payload '{"data":{},"error":"ConfigureReporting 0x000d6f000bbd6160/1 manuSpecificSamsungAccelerometer([{\"attribute\":\"z_axis\",\"minimumReportInterval\":10,\"maximumReportInterval\":\"600\",\"reportableChange\":1}], {\"sendWhen\":\"active\",\"timeout\":10000,\"disableResponse\":false,\"disableRecovery\":false,\"disableDefaultResponse\":true,\"direction\":0,\"srcEndpoint\":null,\"reservedBits\":0,\"manufacturerCode\":4362,\"transactionSequenceNumber\":null,\"writeUndiv\":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE')","status":"error","transaction":"qaqto-3"}'

GSzabados, 2022-10-19:
Indeed the device should support the attributes, because the herdsman converter has these lines for configuring it.

https://github.com/Koenkk/zigbee-herdsman-converters/blob/328e2a362e1f67cfc13dc91e1d939e2c5edd75a1/devices/smartthings.js#L86-L93

Unless the converter does the reporting setup wrongly.

GSzabados, 2022-10-19:
The original SmartThings device handler suggests that the device supports reporting configuration as well.

https://github.com/SmartThingsCommunity/SmartThingsPublic/blob/master/devicetypes/smartthings/smartsense-multi-sensor.src/smartsense-multi-sensor.groovy

user1o1, 2022-11-05:
I have the same problem with the "IM6001-MPP01" Samsung/SmartThings/Samjin Multi Sensor (2018 model) in contrast there are no problems with the "F-MLT-US-2" SmartThings/SmartThings Multi Sensor (2016 model) which I also have.

Who can help?

closed 2022-12-14 by github-actions[bot]

djashjones, 2023-07-01:
Same issue here

Madj42, 2024-02-20:
Did any of you get a resolution on this?  Same thing here.

djashjones, 2024-02-20:
In the process of swapping mine out. Fed up of the batteries lasting only 6-9 months.

GSzabados, 2024-02-20:
> lasting only 6-9 months

Did you mean 6-9 weeks.. ?

djashjones, 2024-02-20:
No, months. I even changed the reporting on the temperature to 0.5 degrees C change but it did not really increase battery life.