Koenkk/zigbee2mqtt #14541: TuYa TS0012_switch_module turns on both loads when only one switch is pressed

fvillena, 2022-10-18:
### What happened?

TuYa TS0012_switch_module 2 gang switch module - (without neutral)
When I click on one of the two switches on the dashboard (State Left or State Right) that refer to that switch, both loads turn on, not just one. Similarly, then clicking on one of the switches turns off both loads at the same time.

### What did you expect to happen?

Each of the switches (Left or Right) will only control its own load.

### Zigbee2MQTT version

1.28.0 commit: 03ba647

### Adapter firmware version

20220219

### Adapter

CC2652P2 Based Zigbee to PoE Coordinator 2022

### Debug log

[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/9814694/log.txt)

fernando155, 2022-10-22:
Same problem with _TZ3000_odzoiovu 3 Gang

I got that model: _TZ3000_bvrlqyj7 and it works fine (old paired) _TZ3000_odzoiovu is been paired today. I´m using the same z2m, just with a conbee2

fernando155, 2022-11-02:
So mine is been fix in the update from 1.28.0 to .1. As said before, I got other switches that was working as normal, and this one was a new addition....

Koenkk, 2022-11-02:
@mikelpsv can you provide a screenshot of the z2m frontend -> device -> about page

mikelpsv, 2022-11-03:
> @mikelpsv can you provide a screenshot of the z2m frontend -> device -> about page

[image]

ZholobovSS, 2022-11-03:
Same problem with same device _TZ3000_jl7qyupf. Upgrade z2m to 1.28.1 not fixed problem.

Koenkk, 2022-11-03:
@mikelpsv for the `TS0012_switch_module` I fixed it yesterday (https://github.com/Koenkk/zigbee2mqtt/issues/14750)  (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

@ZholobovSS same for your device.

Mickroz, 2022-11-26:
i have the same problem as OP, my log says:
```
Failed to configure '0xa4c13831b8aaa0ae', attempt 1 (Error: Bind 0xa4c13831b8aaa0ae/2 genOnOff from '0x00124b001936aa42/1' failed (AREQ - ZDO - bindRsp after 10000ms) at Timeout._onTimeout (/app/node_modules/zigbee-herdsman/src/utils/waitress.ts:64:35) at listOnTimeout (node:internal/timers:559:17) at processTimers (node:internal/timers:502:7))
```

slzzuk, 2022-12-04:
TS0012_switch_module in this z2m 1.28.4-1 version
The error in is fixed and it works fine. However, the same product 3gang model TS0013_switch_module still has the same problem.
thank you

Koenkk, 2022-12-04:
@slzzuk fixed!

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2022-12-04 by Koenkk