Koenkk/zigbee2mqtt #14542: Schneider Electric EER50000 wrong entities exposed

benoithcc, 2022-10-18:
### What happened?

When EER50000 is linked to zigbeee2mqtt, climate (occupied_heating_setpoint, local_temperature) and linkquality are exposed.

### What did you expect to happen?

EER50000 should expose switch (state), power, energy and linkquality as the Schneider Electric EER42000.
EER50000 is not a climate device but a relay device with power and energy meter (to control heater but it is only a relay)

### How to reproduce it (minimal and precise)

Pair a EER50000 device.

### Zigbee2MQTT version

1.28.0-1

### Adapter firmware version

20210708

### Adapter

Sonoff Zigbee 3.0 Plus

benoithcc, 2022-10-22:
Hi,
I think it should at least have the same behavior as the CCTFR6700, otherwise we can’t control the heater at all without an external temperature or manual mode (off/auto/heat) https://www.zigbee2mqtt.io/devices/CCTFR6700.html
Or can someone explain how to switch on or off the heater without those functionalities (it works well with the Schneider box)
Thanks!

closed 2022-11-29 by github-actions[bot]

coelhoricardo, 2022-12-15:
Hi, I have the same issue.

coelhoricardo, 2022-12-15:
@benoithcc can you reopen the issue ?

coelhoricardo, 2022-12-16:
(let's try to ping the original developer)
@pklokke Hi, can you help us ?

pklokke, 2022-12-16:
The EER50000 (aka HACT) has a thermostat application loaded, so it relies on temperature reports to control its Relay/Fil Pilote outputs.

The EER42000 (aka LACT) acts as you describe.

The HW between the two is identical, but changing the application over requires directly reprogramming the MSP430 application processor onboard.

The original principle for the HACT is that it should work directly with an RTS and form a heating control loop, which will operate even if the gateway is offline.  Note that any temperature sensor in principle should work, as long as it can be bound to the temperature measurement client cluster exposed on endpoint 0x0b.

coelhoricardo, 2022-12-16:
@benoithcc you can join the discussion on https://discord.com/channels/556563650429583360/1052136886912688199/1053262346988036136

benoithcc, 2022-12-17:
> https://discord.com/channels/556563650429583360/1052136886912688199/1053262346988036136
I see nothing there... is it the right link?

coelhoricardo, 2022-12-17:
Sorry, you have to join first : https://discord.gg/NyseBeK

benoithcc, 2022-12-17:
Perfect I joined the discussion.

coelhoricardo, 2023-02-03:
@benoithcc It's fixed on the latest release

benoithcc, 2023-02-03:
@coelhoricardo Thanks for the update, I will test it this weekend.

coelhoricardo, 2023-02-03:
@benoithcc I only tested, you should thank @pklokke ;-)