Koenkk/zigbee2mqtt #14554: Awox / EGLO Pir sensor EPIR_Zm

adultano, 2022-10-19:
### Link

https://www.eglo.com/en/accessory-connect-z-sensor-99106.html

### Database entry

Device 0xa4c138a0f146a96c „EPIR_Zm“ not described in states mapping

### Comments

I am using Iobroker zigbee adapter which links to zigbee2mqtt herdsman. The pir sensor device got paired but is missing the config i guess. I tries to map it based on other pir sensors but i failed. Attached the details from iobroker. I guess it could be added to awox.js but i dont know how. Can anyone assist please? Thanks in advance. 
[image]

Koenkk, 2022-10-20:
Iobroker users should create issues in the iobroker issue tracker, not the zigbee2mqtt one.

closed 2022-10-20 by Koenkk

adultano, 2022-10-20:
Well it is not an iobroker issue - is it?! It clearly results from a missing config in zigbee2mqtt...

Koenkk, 2022-10-20:
zigbee2mqtt != zigbee-herdsman

> Device 0xa4c138a0f146a96c „EPIR_Zm“ not described in states mapping

this is really an iobroker issue.

adultano, 2022-10-20:
My bad - apologies. Agreed - but zigbee-herdsman seems also supported by you I think. Let me check and repost on the correct forum then.