Koenkk/zigbee2mqtt #14555: [New device support]: Aubess Wifi ZigBee Mini Smart Switch With Energy Monitor Smart Home 2 Way Control Switches Module Via Alexa Google Home Alice

sdalee, 2022-10-19:
### Link

https://www.aliexpress.com/item/1005004683547255.html?spm=a2g0o.order_list.0.0.516d1802a3KmyV

### Database entry

{"id":102,"type":"Router","ieeeAddr":"0xa4c138e469f1a381","nwkAddr":45514,"manufId":4417,"manufName":"_TZ3000_xkap8wtb","powerSource":"Mains (single phase)","modelId":"TS000F","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":256,"inClusterList":[0,3,4,5,6,10,57344,1794,2820,4096,6280],"outClusterList":[25],"clusters":{"57344":{"attributes":{"53251":"AAAA"}},"genOnOff":{"attributes":{"32768":0,"onOff":1,"tuyaBacklightMode":1,"moesStartUpOnOff":2,"onTime":0,"offWaitTime":0}},"haElectricalMeasurement":{"attributes":{"rmsVoltage":231,"activePower":47,"rmsCurrent":333}},"seMetering":{"attributes":{"currentSummDelivered":[0,0]}},"genBasic":{"attributes":{"modelId":"TS000F","manufacturerName":"_TZ3000_xkap8wtb","powerSource":1,"zclVersion":3,"appVersion":65,"stackVersion":2,"hwVersion":0,"swBuildId":"0122052017"}},"manuSpecificTuya_3":{"attributes":{"switchType":0}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":65,"stackVersion":2,"hwVersion":0,"swBuildId":"0122052017","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1666195704325,"defaultSendRequestWhen":"immediate"}

### Comments

Unfortunately, I don't know how can I do device support in Home Assistant.

Koenkk, 2022-10-20:
`_TZ3000_xkap8wtb` has been added already in the dev branch  (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2022-10-20 by Koenkk

sdalee, 2022-10-20:
Thanx!

sdalee, 2022-11-02:
Dear Koenekk,

I've installed the last z2m version (1.28.1-1) this day. I tried this switch and it works as a switch well (thank you), but the power and energy monitoring functions don't appear in the configuration (Hassio zigbee2mqtt menu -> devices ->dvivice / Exposes) Could you help us to add the power and energy functions to this device.

Koenkk, 2022-11-02:
Working on it in https://github.com/Koenkk/zigbee2mqtt/issues/13929