Koenkk/zigbee2mqtt #14556: Device TS0601 not recognized

zsoltty91, 2022-10-19:
### What happened?

Earlier ticket closed, I commented on it.
https://github.com/Koenkk/zigbee2mqtt/issues/14441

### Zigbee2MQTT version

1.22.1

### Adapter firmware version

2022

### Adapter

slaesh

Koenkk, 2022-10-20:
Moved it to TS0601_thermostat!

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2022-10-20 by Koenkk

f0rdprefect, 2022-11-04:
I am on 1.28.2-dev commit: [4a2190f](https://github.com/Koenkk/zigbee2mqtt/commit/4a2190f) and see these messages when trying to add this Tuya Thermostat... How can I help to make it work?

```
2022-11-04 20:52:57Please see: https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html
warning
2022-11-04 20:52:57Received message from unsupported device with Zigbee model 'TS0601' and manufacturer name '_TZE200_bvu2wnxz'
```

adlerweb, 2022-11-12:
@f0rdprefect Device was removed in #14578 - as per https://github.com/Koenkk/zigbee-herdsman-converters/issues/4694 noone is working on getting support back again.

Exellent1988, 2022-11-28:
Any Progress on official support for those devices?

DoraDeu, 2022-12-12:
I am also still interested in an official support of this Avatto me167 (_TZE200_bvu2wnxz) devices.

Koenkk, 2022-12-12:
@DoraDeu `_TZE200_bvu2wnxz` is supported in the latest-dev  (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

adlerweb, 2022-12-13:
(for those following along: That was commit https://github.com/Koenkk/zigbee-herdsman-converters/commit/9d8cbc88f2328893d18d30d6996978d2effa015f - 6 days ago)