Koenkk/zigbee2mqtt #14557: Device detected as TS0502B but is a WZ1, another WZ1 is not detected See below

MiddleSiggy, 2022-10-19:
### Link

https://www.zigbee2mqtt.io/devices/WZ1.html#skydance-wz1

### Database entry

{"id":39,"type":"Router","ieeeAddr":"0xa4c138caed47972e","nwkAddr":5799,"manufId":4417,"manufName":"_TZB210_nfzrlz29","powerSource":"Mains (single phase)","modelId":"TS0502B","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":268,"inClusterList":[3,4,5,6,4096,8,768,61184,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"stackVersion":0,"dateCode":""}},"lightingColorCtrl":{"attributes":{"colorCapabilities":16,"colorTempPhysicalMin":153,"colorTempPhysicalMax":500}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":64,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":88764544},"lastSeen":1666202149241,"defaultSendRequestWhen":"immediate"}

### Comments

Not sure how to parse this one, but I purchased 3 WZ1 Zigbee Controllers pictured in the Link above. I opened all of them and they all look 100% the same on the inside, but all three report differently.

WZ1 #1 is detected as:
{"id":38,"type":"Router","ieeeAddr":"0xa4c138ea9f7cccaa","nwkAddr":63675,"manufId":4417,"manufName":"_TZB210_rkgngb5o","powerSource":"Mains (single phase)","modelId":"TS0501B","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":257,"inClusterList":[3,4,5,6,4096,8,768,61184,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"modelId":"TS0501B","manufacturerName":"_TZB210_rkgngb5o","powerSource":1,"zclVersion":3,"appVersion":64,"stackVersion":0,"hwVersion":1,"dateCode":""}},"lightingColorCtrl":{"attributes":{}},"genOnOff":{"attributes":{"onOff":1}},"genLevelCtrl":{"attributes":{"currentLevel":254}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00124b0024c866e5","endpointID":1},{"cluster":8,"type":"endpoint","deviceIeeeAddress":"0x00124b0024c866e5","endpointID":1}],"configuredReportings":[{"cluster":6,"attrId":0,"minRepIntval":0,"maxRepIntval":3600,"repChange":0},{"cluster":8,"attrId":0,"minRepIntval":0,"maxRepIntval":3600,"repChange":1}],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":64,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":-526939472},"lastSeen":1666201894696,"defaultSendRequestWhen":"immediate"}

WZ1 #2 is detected as: (Already posted in the Database Entry)
{"id":39,"type":"Router","ieeeAddr":"0xa4c138caed47972e","nwkAddr":5799,"manufId":4417,"manufName":"_TZB210_nfzrlz29","powerSource":"Mains (single phase)","modelId":"TS0502B","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":268,"inClusterList":[3,4,5,6,4096,8,768,61184,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"stackVersion":0,"dateCode":""}},"lightingColorCtrl":{"attributes":{"colorCapabilities":16,"colorTempPhysicalMin":153,"colorTempPhysicalMax":500}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":64,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":88764544},"lastSeen":1666202149241,"defaultSendRequestWhen":"immediate"}

WZ1 
[image]
#3 is not detected as all, and is not a supported device:
{"id":40,"type":"Router","ieeeAddr":"0xa4c138a36caf9722","nwkAddr":39870,"manufId":4417,"manufName":"_TZB210_nfzrlz29","powerSource":"Mains (single phase)","modelId":"TS0503B","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":268,"inClusterList":[3,4,5,6,4096,8,768,61184,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"stackVersion":0,"dateCode":""}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":64,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1666202213067,"defaultSendRequestWhen":"immediate"}

I hope this helps in trying to clarify the devices that are being detected.

Also the image for #2 is currently:
https://www.zigbee2mqtt.io/devices/TS0502B.html#tuya-ts0502b

But the actual device is a:
https://www.zigbee2mqtt.io/devices/TS0502B.html#tuya-ts0502b

Please let me know how I can help in enhancing the detection.

MiddleSiggy, 2022-11-09:
Do you need anything to correct this?

FutureCow, 2022-12-10:
Same problem here, got 3 new EcoDim devices: https://www.zigbee2mqtt.io/devices/ED-10042.html
But one is detected as Tuya TS0501B
[image]
[image]

closed 2023-01-18 by github-actions[bot]

LeoS3, 2023-10-13:
Has anyone found the solution to this problem?

LaTrappe, 2024-01-15:
I`m curious to hear if anyone has found a solution. My EcoDim / EcoBright is showing up as color device (the tuya one) as well, instead of color temp.

hank-droid, 2024-11-02:
i have the same issue. if paired via tradfri controller, it is recognised as rgbw and i can somehow use it, but the dimming is not very responsive