Koenkk/zigbee2mqtt #14559: [Feature request]: Integrate external Zigbee2MQTT on HomeAssistant

maframan, 2022-10-19:
### Is your feature request related to a problem? Please describe

Hi, i've installed zigbee2mqtt as conatiner in separeted raspberry where is not running home assistant.
How do I to integrate it in home assistant console? Is it possible

Thanks a lot for your support.

### Describe the solution you'd like

Integration external zigbee2mqtt installation on home assistant console

### Describe alternatives you've considered

N/A

### Additional context

N/A

danieledwardgeorgehitchcock, 2022-10-20:
Zigbee2MQTT uses MQTT as the layer to interface with Home Automation software.

As long as you have your MQTT broker set up correctly, and Home Assistant can connect to it, you should just need to enable the Home Assistant config entry in Zigbee2MQTT and everything should just work

maframan, 2022-10-20:
I'm referring to use zigbee2mqtt console on home Assistant console because with external zigbee2mqtt I loss integration that provide the zigbee2mqtt ha addon

danieledwardgeorgehitchcock, 2022-10-20:
If I understand correctly, you can make reference to the Zigbee2MQTT GUI from Home Assistant by configuring a `panel_iframe` in your Home Assistant configuration.yaml.

[See here](https://www.home-assistant.io/integrations/panel_iframe/#:~:text=The%20panel_iframe%20support%20allows%20you%20to%20add%20additional,cannot%20access%20HTTP%20sites%20through%20an%20iframe%20panel.)

Koenkk, 2022-10-20:
You probably want https://github.com/zigbee2mqtt/hassio-zigbee2mqtt/pull/400

maframan, 2022-10-20:
With panel_iframe I get below error:

[image]

Loic691, 2022-10-20:
@maframan 
This is not really a feature request... As explained few posts ago by @danieledwardgeorgehitchcock you can add external web reference to HA by adding iframe pannel.
It's more an integration request so you have to read or ask HA forum for this.
In few words, the Z2M frontend is HTTP only (maybe a next feature to ask to switch in HTTPS ?). 
If you have access in HTTP mode to your HA on your local network , the Z2M iframe should diplay sucessfully.
As your screen capture, and for security reason, in HA, it's impossible to display non secured HTTP web page (your Z2M for example) in a secured HTTPS HA.
There is several posts for this on the HA community with several solutions to this
In first time I sugget you to use your HA in local mode with HTTP.
After the other solutions are more complex.

maframan, 2022-10-20:
> @maframan This is not really a feature request... As explained few posts ago by @danieledwardgeorgehitchcock you can add external web reference to HA by adding iframe pannel. It's more an integration request so you have to read or ask HA forum for this. In few words, the Z2M frontend is HTTP only (maybe a next feature to ask to switch in HTTPS ?). If you have access in HTTP mode to your HA on your local network , the Z2M iframe should diplay sucessfully. As your screen capture, and for security reason, in HA, it's impossible to display non secured HTTP web page (your Z2M for example) in a secured HTTPS HA. There is several posts for this on the HA community with several solutions. In first time I suggest you to use your HA in local mode with HTTP. The other solutions are more complex.

Thanks a lot for your supoort and answer

closed 2022-10-20 by maframan