Koenkk/zigbee2mqtt #14563: [New device support]: New model of an already supported device

Lovasz-Akos, 2022-10-20:
### Link

not public yet

### Database entry

{"id":1,"type":"Coordinator","ieeeAddr":"0x0c4314fffe59a63d","nwkAddr":0,"manufId":0,"epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":48879,"inClusterList":[0,3,6,10,25,26,768],"outClusterList":[0,3,4,5,6,8,32,768,1024,1026,1029,1030,1280,2817,2819,2820,1794,4096,64513,64514],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41230,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"interviewCompleted":true,"meta":{},"lastSeen":null,"defaultSendRequestWhen":"immediate"} {"id":2,"type":"Router","ieeeAddr":"0xa4c1382f0f83723d","nwkAddr":42040,"manufId":4417,"manufName":"_TZ3000_yujkchbz","powerSource":"Mains (single phase)","modelId":"TS011F","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"inClusterList":[3,4,5,6,1794,2820,57345,57344,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"modelId":"TS011F","manufacturerName":"_TZ3000_yujkchbz","powerSource":1,"zclVersion":3,"appVersion":74,"stackVersion":0,"hwVersion":1,"dateCode":""}},"genOnOff":{"attributes":{"32768":0,"onOff":1,"onTime":0,"offWaitTime":0,"tuyaBacklightMode":1,"moesStartUpOnOff":0,"tuyaBacklightSwitch":1}},"manuSpecificTuya_3":{"attributes":{"53248":0,"53249":0,"53250":0,"53251":0,"53252":0,"53253":0,"powerOnBehavior":0,"switchType":0}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":74,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1666271020670,"defaultSendRequestWhen":"immediate"}

### Comments

I'd like to ask if there's a quick way to add a device that's basically just a facelift version of an already existing device, nothing changed about the internals or exposes.

Lovasz-Akos, 2022-10-20:
supposed to use same internals as [this plug](https://www.zigbee2mqtt.io/devices/TS011F_plug_3.html), with a new shell

Lovasz-Akos, 2022-10-20:
that could work, thanks :D 

at some point I'd like to create it's own entry as it looks different and the vendor will be Moes (and probably a different model number)

danieledwardgeorgehitchcock, 2022-10-20:
If that's the case, I'd just ad a whitelabel entry to the TS011F converter rather than adding a complete new entry

Koenkk, 2022-10-20:
Fixed by https://github.com/Koenkk/zigbee-herdsman-converters/pull/4806

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2022-10-20 by Koenkk