Koenkk/zigbee2mqtt #14564: Support for Click Smart CSPGM2075PW

jb123567, 2022-10-20:
### Link

https://click-smart.com/products/wireless-smart-switch

### Database entry

"id":30,"type":"EndDevice","ieeeAddr":"0xa4c1385fb4573458","nwkAddr":51481,"manufId":4417,"manufName":"_TZ3000_wkhizchq","powerSource":"Battery","modelId":"TS0042","epList":[1,2,3,4],"endpoints":{"1":{"profId":260,"epId":1,"devId":0,"inClusterList":[1,6,57344,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"modelId":"TS0042","manufacturerName":"_TZ3000_wkhizchq","powerSource":3,"zclVersion":3,"appVersion":67,"stackVersion":0,"hwVersion":1,"dateCode":""}},"genPowerCfg":{"attributes":{"batteryPercentageRemaining":200}}},"binds":[{"cluster":1,"type":"endpoint","deviceIeeeAddress":"0x00124b0021b77f81","endpointID":1},{"cluster":0,"type":"endpoint","deviceIeeeAddress":"0x00124b0021b77f81","endpointID":1}],"configuredReportings":[{"cluster":1,"attrId":33,"minRepIntval":3600,"maxRepIntval":62000,"repChange":0}],"meta":{}},"2":{"profId":260,"epId":2,"devId":0,"inClusterList":[6,1,6],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"3":{"profId":260,"epId":3,"devId":0,"inClusterList":[6,1,6],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"4":{"profId":260,"epId":4,"devId":0,"inClusterList":[6,1,6],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":67,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":821693351},"lastSeen":1666270250819,"defaultSendRequestWhen":"immediate"

### Comments

I’m not sure the issue, however the device identifies as a Tuya Double Switch Device, and exposes action, link quality and Battery however on pressing the either of the buttons the action string is empty. On examining the payload in debugging the action text is correct, it’s just not brought across to the action string.

I have produced a external converter, but not sure if that’s the correct thing to do? Or am I missing something really simple?

### External converter

```shell
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require("zigbee-herdsman-converters/lib/tuya");

const definition = {
    // Since a lot of Tuya devices use the same modelID, but use different data points
    // it's usually necessary to provide a fingerprint instead of a zigbeeModel
    fingerprint: [
        {
            // The model ID from: Device with modelID 'TS0601' is not supported
            // You may need to add \u0000 at the end of the name in some cases
            modelID: 'TS0042',
            // The manufacturer name from: Device with modelID 'TS0601' is not supported.
            manufacturerName: '_TZ3000_wkhizchq'
        },
    ],
    model: 'CSScenario',
    vendor: 'ClickSmartPlus',
    description: 'Scenario Switch',
    fromZigbee: [
        fz.ignore_basic_report, // Add this if you are getting no converter for 'genBasic'
        fz.tuya_data_point_dump, // This is a debug converter, it will be described in the next part
    ],
    toZigbee: [
        tz.tuya_data_point_test, // Another debug converter
    ],
    onEvent: tuya.onEventSetTime, // Add this if you are getting no converter for 'commandMcuSyncTime'
    configure: async (device, coordinatorEndpoint, logger) => {
        const endpoint = device.getEndpoint(1);
        await reporting.bind(endpoint, coordinatorEndpoint, ['genBasic']);
    },
    exposes: [e.action(), e.battery(),// Here you should put all functionality that your device exposes
    ],
};

module.exports = definition;
```

jb123567, 2022-10-20:
I have now got the external converter working, but Zigbee2MQTT still reports that the string is still empty, despite the payload being right.

jb123567, 2022-10-20:
Okay, I've been silly, realised that looking in the log book that the payload is actually working. Is it worth adding this device with natively compatible, is it's not advertised as a Tuya Device

Koenkk, 2022-10-21:
How does you external converter look like now? Then I can add support for it out-of-the-box.

jb123567, 2022-10-21:
> How does you external converter look like now? Then I can add support for it out-of-the-box.

I have produced the below external converter using the Tuya TS0042 as a template, it correctly identifies the device and identifies when the buttons is clicked, double clicked or held, however it doesn't differentiate between the two buttons, the native support does, what am I missing?

const fz =require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require("zigbee-herdsman-converters/lib/tuya");

const definition = {
    fingerprint: [
        {
            modelID:'TS0042',
            manufacturerName: '_TZ3000_wkhizchq'
        },
    ],
    model: 'CSPGM2075PW',
    vendor: 'Click Smart Plus',
    description: 'Scenario Switch',
    fromZigbee: [fz.tuya_on_off_action, fz.battery],
    toZigbee: [], 
    configure: tuya.configureMagicPacket,
    exposes: [e.battery(), e.action(['1_single', '1_double', '1_hold', '2_single', '2_double', '2_hold'])],
};

module.exports = definition;

Koenkk, 2022-10-22:
Can you provide the debug log of pressing the buttons which cannot be distinguished?

See https://www.zigbee2mqtt.io/guide/usage/debug.html on how to enable debug logging.

jb123567, 2022-10-22:
> Can you provide the debug log of pressing the buttons which cannot be distinguished?
> 
> 
> 
> See https://www.zigbee2mqtt.io/guide/usage/debug.html on how to enable debug logging.

Here we go, hopefully this is of use?

```

Debug 2022-10-22 21:18:07Received Zigbee message from 'Outside Light Switch', type 'raw', cluster 'genOnOff', data '{"data":[1,175,253,0],"type":"Buffer"}' from endpoint 1 with groupID 0
Info 2022-10-22 21:18:07MQTT publish: topic 'zigbee2mqtt/Outside Light Switch', payload '{"action":"single","battery":95,"linkquality":94,"voltage":2900}'
Info 2022-10-22 21:18:07MQTT publish: topic 'zigbee2mqtt/Outside Light Switch', payload '{"action":"","battery":95,"linkquality":94,"voltage":2900}'
Info 2022-10-22 21:18:07MQTT publish: topic 'zigbee2mqtt/Outside Light Switch/action', payload 'single'

```

Koenkk, 2022-10-23:
I see only one button click. I need to have to log of both button clicks resulting in the same action.

jb123567, 2022-10-23:
> I see only one button click. I need to have to log of both button clicks resulting in the same action.

Sorry

Here is button 1 being pressed

```

Debug 2022-10-23 21:32:20Received Zigbee message from 'Outside Light Switch', type 'raw', cluster 'genOnOff', data '{"data":[1,191,253,0],"type":"Buffer"}' from endpoint 1 with groupID 0
Info 2022-10-23 21:32:20MQTT publish: topic 'zigbee2mqtt/Outside Light Switch', payload '{"action":"single","battery":95,"linkquality":87,"voltage":2900}'
Info 2022-10-23 21:32:20MQTT publish: topic 'zigbee2mqtt/Outside Light Switch', payload '{"action":"","battery":95,"linkquality":87,"voltage":2900}'
Info 2022-10-23 21:32:20MQTT publish: topic 'zigbee2mqtt/Outside Light Switch/action', payload 'single'

```

Here is button 2 being pressed 

```

Debug 2022-10-23 21:30:05Received Zigbee message from 'Outside Light Switch', type 'raw', cluster 'genOnOff', data '{"data":[1,190,253,0],"type":"Buffer"}' from endpoint 2 with groupID 0
Info 2022-10-23 21:30:05MQTT publish: topic 'zigbee2mqtt/Outside Light Switch', payload '{"action":"single","battery":95,"linkquality":51,"voltage":2900}'
Info 2022-10-23 21:30:05MQTT publish: topic 'zigbee2mqtt/Outside Light Switch', payload '{"action":"","battery":95,"linkquality":51,"voltage":2900}'
Info 2022-10-23 21:30:05MQTT publish: topic 'zigbee2mqtt/Outside Light Switch/action', payload 'single'

```

Koenkk, 2022-10-24:
It should provided different actions when changing `    model: 'CSPGM2075PW',` to `    model: 'TS0042',` in your external converter.

jb123567, 2022-10-24:
> It should provided different actions when changing `    model: 'CSPGM2075PW',` to `    model: 'TS0042',` in your external converter.

#It does you're right. I was hoping for it to identify as the correct device, just to help people looking for it in the future, but no bother if it cant.

Koenkk, 2022-10-25:
I've marked it as a whitelabel now so it will appear on the supported device page with the next release. Can this be closed?

jb123567, 2022-10-25:
Yes thanks. Can I remove the external converter after the next update?

Koenkk, 2022-10-25:
Yes!

closed 2022-10-25 by Koenkk