Koenkk/zigbee2mqtt #14568: Devices leaving network shortly after pairing

mateuszkozakiewicz, 2022-10-20:
### What happened?

My IKEA devices are leaving the network shortly after pairing, the devices are supported:
https://www.zigbee2mqtt.io/devices/E2001_E2002.html - have 2 buttons, one has the problem, the other works
https://www.zigbee2mqtt.io/devices/LED1545G12.html - have 4 bulbs, 2 have the problem, others work

### What did you expect to happen?

After pairing device stays paired.

### How to reproduce it (minimal and precise)

Tested on Zigbee2MQTT 1.27.2, 1.28.0, latest-dev

Pairing is enabled on nearest router
Device enters pairing mode
Pairing is successful, device is fully functional
After 10 seconds device leaves the network with message "Device xxx left the network"

### Zigbee2MQTT version

1.28.0

### Adapter firmware version

6.7.10.0 build 423

### Adapter

Xiaomi Gateway 3

### Debug log

Zigbee2MQTT:info  2022-10-20 19:27:41: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"left_network","meta":{"friendly_name":"0x84ba20fffed6f94e"},"type":"device_removed"}'
Zigbee2MQTT:info  2022-10-20 19:27:41: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x84ba20fffed6f94e","ieee_address":"0x84ba20fffed6f94e"},"type":"device_leave"}'
Zigbee2MQTT:warn  2022-10-20 19:27:41: Device '0x84ba20fffed6f94e' left the network
Zigbee2MQTT:info  2022-10-20 19:27:41: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"left_network","meta":{"friendly_name":"0x84ba20fffed6f94e"},"type":"device_removed"}'
Zigbee2MQTT:info  2022-10-20 19:27:41: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x84ba20fffed6f94e","ieee_address":"0x84ba20fffed6f94e"},"type":"device_leave"}'
Zigbee2MQTT:warn  2022-10-20 19:27:41: Device '0x84ba20fffed6f94e' left the network
Zigbee2MQTT:info  2022-10-20 19:27:41: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"left_network","meta":{"friendly_name":"0x84ba20fffed6f94e"},"type":"device_removed"}'
Zigbee2MQTT:info  2022-10-20 19:27:41: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x84ba20fffed6f94e","ieee_address":"0x84ba20fffed6f94e"},"type":"device_leave"}'
Zigbee2MQTT:warn  2022-10-20 19:27:41: Device '0x84ba20fffed6f94e' left the network
Zigbee2MQTT:info  2022-10-20 19:27:41: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"left_network","meta":{"friendly_name":"0x84ba20fffed6f94e"},"type":"device_removed"}'
Zigbee2MQTT:info  2022-10-20 19:27:41: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x84ba20fffed6f94e","ieee_address":"0x84ba20fffed6f94e"},"type":"device_leave"}'
Zigbee2MQTT:warn  2022-10-20 19:27:41: Device '0x84ba20fffed6f94e' left the network
Zigbee2MQTT:info  2022-10-20 19:27:41: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"left_network","meta":{"friendly_name":"0x84ba20fffed6f94e"},"type":"device_removed"}'
Zigbee2MQTT:info  2022-10-20 19:27:41: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x84ba20fffed6f94e","ieee_address":"0x84ba20fffed6f94e"},"type":"device_leave"}'
Zigbee2MQTT:warn  2022-10-20 19:27:41: Device '0x84ba20fffed6f94e' left the network
Zigbee2MQTT:info  2022-10-20 19:27:40: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"left_network","meta":{"friendly_name":"0x003c84fffe0f0d4a"},"type":"device_removed"}'
Zigbee2MQTT:info  2022-10-20 19:27:40: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x003c84fffe0f0d4a","ieee_address":"0x003c84fffe0f0d4a"},"type":"device_leave"}'
Zigbee2MQTT:warn  2022-10-20 19:27:40: Device '0x003c84fffe0f0d4a' left the network
Zigbee2MQTT:info  2022-10-20 19:27:40: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"left_network","meta":{"friendly_name":"0x84ba20fffed6f94e"},"type":"device_removed"}'
Zigbee2MQTT:info  2022-10-20 19:27:40: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x84ba20fffed6f94e","ieee_address":"0x84ba20fffed6f94e"},"type":"device_leave"}'
Zigbee2MQTT:warn  2022-10-20 19:27:40: Device '0x84ba20fffed6f94e' left the network
Zigbee2MQTT:info  2022-10-20 19:27:40: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"left_network","meta":{"friendly_name":"0x84ba20fffed6f94e"},"type":"device_removed"}'
Zigbee2MQTT:info  2022-10-20 19:27:40: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x84ba20fffed6f94e","ieee_address":"0x84ba20fffed6f94e"},"type":"device_leave"}'
Zigbee2MQTT:warn  2022-10-20 19:27:40: Device '0x84ba20fffed6f94e' left the network
Zigbee2MQTT:info  2022-10-20 19:27:40: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"left_network","meta":{"friendly_name":"0x84ba20fffed6f94e"},"type":"device_removed"}'
Zigbee2MQTT:info  2022-10-20 19:27:40: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x84ba20fffed6f94e","ieee_address":"0x84ba20fffed6f94e"},"type":"device_leave"}'
Zigbee2MQTT:warn  2022-10-20 19:27:40: Device '0x84ba20fffed6f94e' left the network
Zigbee2MQTT:info  2022-10-20 19:27:40: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"left_network","meta":{"friendly_name":"0x003c84fffe0f0d4a"},"type":"device_removed"}'
Zigbee2MQTT:info  2022-10-20 19:27:40: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x003c84fffe0f0d4a","ieee_address":"0x003c84fffe0f0d4a"},"type":"device_leave"}'
Zigbee2MQTT:warn  2022-10-20 19:27:40: Device '0x003c84fffe0f0d4a' left the network
Zigbee2MQTT:info  2022-10-20 19:27:40: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"left_network","meta":{"friendly_name":"0x003c84fffe0f0d4a"},"type":"device_removed"}'
Zigbee2MQTT:info  2022-10-20 19:27:40: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x003c84fffe0f0d4a","ieee_address":"0x003c84fffe0f0d4a"},"type":"device_leave"}'
Zigbee2MQTT:warn  2022-10-20 19:27:40: Device '0x003c84fffe0f0d4a' left the network
Zigbee2MQTT:info  2022-10-20 19:27:40: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"left_network","meta":{"friendly_name":"0x003c84fffe0f0d4a"},"type":"device_removed"}'
Zigbee2MQTT:info  2022-10-20 19:27:40: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x003c84fffe0f0d4a","ieee_address":"0x003c84fffe0f0d4a"},"type":"device_leave"}'
Zigbee2MQTT:warn  2022-10-20 19:27:40: Device '0x003c84fffe0f0d4a' left the network
Zigbee2MQTT:info  2022-10-20 19:27:39: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"left_network","meta":{"friendly_name":"0x003c84fffe0f0d4a"},"type":"device_removed"}'
[... 27 lines cut ...]
Zigbee2MQTT:debug 2022-10-20 19:27:23: Skipping message, definition is undefined and still interviewing
Zigbee2MQTT:debug 2022-10-20 19:27:23: Received Zigbee message from '0x003c84fffe0f0d4a', type 'readResponse', cluster 'genBasic', data '{"swBuildId":"1.0.032"}' from endpoint 1 with groupID 0
Zigbee2MQTT:info  2022-10-20 19:27:22: Successfully configured '0x84ba20fffed6f94e'
Zigbee2MQTT:info  2022-10-20 19:27:22: MQTT publish: topic 'zigbee2mqtt/0x84ba20fffed6f94e', payload '{"color_mode":"xy","linkquality":152,"power_on_behavior":null,"state":null,"update":{"state":null},"update_available":null}'
Zigbee2MQTT:debug 2022-10-20 19:27:22: Received Zigbee message from '0x84ba20fffed6f94e', type 'readResponse', cluster 'lightingColorCtrl', data '{"colorTempPhysicalMax":454,"colorTempPhysicalMin":250}' from endpoint 1 with groupID 0
Zigbee2MQTT:info  2022-10-20 19:27:22: MQTT publish: topic 'zigbee2mqtt/0x84ba20fffed6f94e', payload '{"color_mode":"xy","linkquality":152,"power_on_behavior":null,"state":null,"update":{"state":null},"update_available":null}'
Zigbee2MQTT:debug 2022-10-20 19:27:22: Missing colorTempPhysicalMin and/or colorTempPhysicalMax for endpoint 0x84ba20fffed6f94e!
Zigbee2MQTT:debug 2022-10-20 19:27:22: Received Zigbee message from '0x84ba20fffed6f94e', type 'readResponse', cluster 'lightingColorCtrl', data '{"colorCapabilities":16}' from endpoint 1 with groupID 0
Zigbee2MQTT:info  2022-10-20 19:27:22: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_successful","meta":{"description":"TRADFRI LED bulb E12/E14 450/470 lumen, wireless dimmable white spectrum/chandelier opal white","friendly_name":"0x84ba20fffed6f94e","model":"LED1949C5","supported":true [... cut ...]
Zigbee2MQTT:info  2022-10-20 19:27:21: Starting interview of '0x84ba20fffed6f94e'
Zigbee2MQTT:info  2022-10-20 19:27:21: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":{"friendly_name":"0x84ba20fffed6f94e"},"type":"device_connected"}'
Zigbee2MQTT:info  2022-10-20 19:27:21: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x84ba20fffed6f94e","ieee_address":"0x84ba20fffed6f94e"},"type":"device_joined"}'
Zigbee2MQTT:info  2022-10-20 19:27:21: Device '0x84ba20fffed6f94e' joined

closed 2022-11-27 by github-actions[bot]

horni2, 2023-01-27:
Hi, was there any solution? I have the same issue, but not (only) for IKEA products. Tried already to pair TUYA and HUE...

horni2, 2023-01-27:
I´m using a Sonoff plugged in a nuc
[image]
(the new version). Tried already to restart Z2M and (in my case) Homeassistant. I also turned off the HUE bridge and our guest wifi. But nothing is an advantage.
[image]

What I recognized is that it doesn't matter to which rout I'm trying to pair. But when trying to pair with automatic detection the devices aren't found at all...
I also wonder because all the rest of the network is stable...

horni2, 2023-01-27:
Sorry for the layout above there went something wrong...

mateuszkozakiewicz, 2023-01-27:
There doesn't seem to be any indication in the logs as to the root cause so my general approach to this error is to try a few image tags behind, and latest-dev. In general I found 1.27.2 to be the most stable however it lacks support for some of my devices. 
I also tried rebooting/reflashing my coordinator, turning off wifi in my house, pairing with different router device, pairing directly to the coordinator but the most reliable solution is to simply restart z2m and hope it works this time. It might be my coordinator but the whole zigbee network is generally very unstable for me, I have to restart z2m container almost on a daily basis because it stops receiving messages even though the mqtt broker on the coordinator is working fine.

horni2, 2023-01-27:
OK, thanks. Then I continue to search  :)

Kaptensanders, 2023-02-09:
Same problem for all my Aqara Motion P1's --> Sonoff Dongle-E Router --> Sonoff Dongle-E Coordinator --> Z2M

-->device_joined
-->Configuring
--> Successfully configured
10s pass
-->device_leave

I'm playing with the idea that messages are processed or reaches the the coordinator or Z2M in the wrong order.
For other Aqara devices, when re-pairing (eg long-press reset), there is an initial device_leave instruction followed by device_joined, and the interview...

This started happening after adding the Sonoff Dongle-E Routers. Before, when pairing directly with the coordinator (also Sonoff Dongle-E), there was no issues.

kirovilya, 2023-02-11:
@mateuszkozakiewicz @Kaptensanders please check on the latest dev/edge version of z2m, I made some changes for ezsp adapter

mateuszkozakiewicz, 2023-02-13:
@kirovilya there is no change, I still have a device that is unable to be paired, it is IKEA STYRBAR button, I have 2 of these already in my network but 3rd will not join. The update helped with another issue where I was not able to put some of my routers in pairing mode but not this one.
```
info 2023-02-13 18:11:22Device '0x842e14fffe742c9a' joined
info 2023-02-13 18:11:22MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x842e14fffe742c9a","ieee_address":"0x842e14fffe742c9a"},"type":"device_joined"}'
info 2023-02-13 18:11:22MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":{"friendly_name":"0x842e14fffe742c9a"},"type":"device_connected"}'
info 2023-02-13 18:11:22Starting interview of '0x842e14fffe742c9a'
info 2023-02-13 18:11:22MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x842e14fffe742c9a","ieee_address":"0x842e14fffe742c9a","status":"started"},"type":"device_interview"}'
info 2023-02-13 18:11:22MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_started","meta":{"friendly_name":"0x842e14fffe742c9a"},"type":"pairing"}'
error 2023-02-13 18:11:42Failed to interview '0x842e14fffe742c9a', device has not successfully been paired
info 2023-02-13 18:11:42MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x842e14fffe742c9a","ieee_address":"0x842e14fffe742c9a","status":"failed"},"type":"device_interview"}'
info 2023-02-13 18:11:42MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_failed","meta":{"friendly_name":"0x842e14fffe742c9a"},"type":"pairing"}'
warning 2023-02-13 18:11:52Device '0x842e14fffe742c9a' left the network
info 2023-02-13 18:11:52MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x842e14fffe742c9a","ieee_address":"0x842e14fffe742c9a"},"type":"device_leave"}'
info 2023-02-13 18:11:52MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"left_network","meta":{"friendly_name":"0x842e14fffe742c9a"},"type":"device_removed"}'
warning 2023-02-13 18:11:52Device '0x842e14fffe742c9a' left the network
info 2023-02-13 18:11:53MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x842e14fffe742c9a","ieee_address":"0x842e14fffe742c9a"},"type":"device_leave"}'
info 2023-02-13 18:11:53MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"left_network","meta":{"friendly_name":"0x842e14fffe742c9a"},"type":"device_removed"}'
```

kirovilya, 2023-02-13:
@mateuszkozakiewicz can you record herdsman log of your error?

mateuszkozakiewicz, 2023-02-13:
@kirovilya 
```
2023-02-13T17:30:43.785Z zigbee-herdsman:adapter:ezsp:uart -+- waiting (7) success
2023-02-13T17:30:43.785Z zigbee-herdsman:adapter:ezsp:ezsp <== 0x5: {"_id_":5,"_cls_":"nop","_isRequest_":false}
2023-02-13T17:30:43.785Z zigbee-herdsman:adapter:ezsp:ezsp <== Frame: 5680010500
2023-02-13T17:30:43.785Z zigbee-herdsman:adapter:ezsp:uart --> [8520dd7e]
2023-02-13T17:30:43.785Z zigbee-herdsman:adapter:ezsp:uart --> ACK  (5)
2023-02-13T17:30:43.785Z zigbee-herdsman:adapter:ezsp:uart <-- DATA (4,7,0): 4714a1a9512a26747e
2023-02-13T17:30:43.785Z zigbee-herdsman:adapter:ezsp:uart <-- [7e]
2023-02-13T17:30:43.784Z zigbee-herdsman:adapter:ezsp:uart <-- [4714a1a9512a2674]
2023-02-13T17:30:43.773Z zigbee-herdsman:adapter:ezsp:uart -?- waiting (7)
2023-02-13T17:30:43.772Z zigbee-herdsman:adapter:ezsp:uart --> [641421a9512a00a47e]
2023-02-13T17:30:43.772Z zigbee-herdsman:adapter:ezsp:uart --> DATA (6,4,0): 5600010500
2023-02-13T17:30:43.772Z zigbee-herdsman:adapter:ezsp:ezsp ==> {"_cls_":"nop","_id_":5,"_isRequest_":true}
2023-02-13T17:30:43.772Z zigbee-herdsman:adapter:ezsp:ezsp ==> nop: null
2023-02-13T17:30:43.772Z zigbee-herdsman:adapter:ezsp:ezsp Time to watchdog ... 0
2023-02-13T17:30:42.885Z zigbee-herdsman:adapter:ezsp:uart -+- waiting (6) success
2023-02-13T17:30:42.885Z zigbee-herdsman:adapter:ezsp:ezsp <== 0x55: {"_id_":85,"_cls_":"setPolicy","_isRequest_":false,"status":0}
2023-02-13T17:30:42.885Z zigbee-herdsman:adapter:ezsp:ezsp <== Frame: 558001550000
2023-02-13T17:30:42.885Z zigbee-herdsman:adapter:ezsp:uart --> [8430fc7e]
2023-02-13T17:30:42.885Z zigbee-herdsman:adapter:ezsp:uart --> ACK  (4)
2023-02-13T17:30:42.885Z zigbee-herdsman:adapter:ezsp:uart <-- DATA (3,6,0): 3617a1a9012a15de1e7e
2023-02-13T17:30:42.884Z zigbee-herdsman:adapter:ezsp:uart <-- [1e7e]
2023-02-13T17:30:42.877Z zigbee-herdsman:adapter:ezsp:uart <-- [3617a1a9012a15de]
Zigbee2MQTT:info  2023-02-13 18:30:42: MQTT publish: topic 'zigbee2mqtt/bridge/response/permit_join', payload '{"data":{},"error":"ZdoRequest error","status":"error","transaction":"hzc8e-5"}'
    at Queue.executeNext (/app/node_modules/zigbee-herdsman/src/utils/queue.ts:32:32)
    at Object.func (/app/node_modules/zigbee-herdsman/src/adapter/ezsp/adapter/ezspAdapter.ts:240:32)
    at processTicksAndRejections (node:internal/process/task_queues:95:5)
    at Driver.zdoRequest (/app/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/driver.ts:572:19)
Zigbee2MQTT:debug 2023-02-13 18:30:42: Error: ZdoRequest error
Zigbee2MQTT:error 2023-02-13 18:30:42: Request 'zigbee2mqtt/bridge/request/permit_join' failed with error: 'ZdoRequest error'
2023-02-13T17:30:42.858Z zigbee-herdsman:adapter:ezsp:erro zdoRequest error
    at Queue.executeNext (/app/node_modules/zigbee-herdsman/src/utils/queue.ts:32:32)
    at Object.func (/app/node_modules/zigbee-herdsman/src/adapter/ezsp/adapter/ezspAdapter.ts:240:32)
    at Driver.zdoRequest (/app/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/driver.ts:568:21)
    at Driver.request (/app/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/driver.ts:465:25)
    at processTicksAndRejections (node:internal/process/task_queues:95:5)
    at Driver.networkIdToEUI64 (/app/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/driver.ts:602:19)
2023-02-13T17:30:42.858Z zigbee-herdsman:adapter:ezsp:erro Request error Error: Unrecognized nodeId:52627: Error: Unrecognized nodeId:52627
2023-02-13T17:30:42.855Z zigbee-herdsman:adapter:ezsp:uart -?- waiting (6)
2023-02-13T17:30:42.855Z zigbee-herdsman:adapter:ezsp:uart --> [531721a9012a15b15907277e]
2023-02-13T17:30:42.855Z zigbee-herdsman:adapter:ezsp:uart --> DATA (5,3,0): 5500015500000300
2023-02-13T17:30:42.854Z zigbee-herdsman:adapter:ezsp:ezsp ==> {"_cls_":"setPolicy","_id_":85,"_isRequest_":true,"policyId":0,"decisionId":3}
2023-02-13T17:30:42.854Z zigbee-herdsman:adapter:ezsp:uart -+- waiting (5) success
2023-02-13T17:30:42.854Z zigbee-herdsman:adapter:ezsp:ezsp <== 0x61: {"_id_":97,"_cls_":"lookupEui64ByNodeId","_isRequest_":false,"status":1,"eui64":{"type":"Buffer","data":[0,3,1,35,1,1,0,0]}}
2023-02-13T17:30:42.854Z zigbee-herdsman:adapter:ezsp:ezsp <== Frame: 5480016100010000010123010300
2023-02-13T17:30:42.854Z zigbee-herdsman:adapter:ezsp:uart --> [83401b7e]
2023-02-13T17:30:42.854Z zigbee-herdsman:adapter:ezsp:uart --> ACK  (3)
2023-02-13T17:30:42.854Z zigbee-herdsman:adapter:ezsp:uart <-- DATA (2,5,0): 2516a1a9352a14b259954b06ab5692d66e7e
2023-02-13T17:30:42.853Z zigbee-herdsman:adapter:ezsp:uart <-- [59954b06ab5692d66e7e]
2023-02-13T17:30:42.849Z zigbee-herdsman:adapter:ezsp:uart <-- [2516a1a9352a14b2]
2023-02-13T17:30:42.829Z zigbee-herdsman:adapter:ezsp:ezsp ==> setPolicy: {"policyId":0,"decisionId":3}
2023-02-13T17:30:42.829Z zigbee-herdsman:adapter:ezsp:ezsp Set EzspPolicyId.TRUST_CENTER_POLICY = 3
2023-02-13T17:30:42.829Z zigbee-herdsman:adapter:ezsp:uart -?- waiting (5)
2023-02-13T17:30:42.828Z zigbee-herdsman:adapter:ezsp:uart --> [421621a9352a867fc34b7e]
2023-02-13T17:30:42.828Z zigbee-herdsman:adapter:ezsp:uart --> DATA (4,2,0): 540001610093cd
2023-02-13T17:30:42.828Z zigbee-herdsman:adapter:ezsp:ezsp ==> {"_cls_":"lookupEui64ByNodeId","_id_":97,"_isRequest_":true,"nodeId":52627}
2023-02-13T17:30:42.828Z zigbee-herdsman:adapter:ezsp:uart -+- waiting (4) success
2023-02-13T17:30:42.828Z zigbee-herdsman:adapter:ezsp:ezsp <== 0xaf: {"_id_":175,"_cls_":"addTransientLinkKey","_isRequest_":false,"status":0}
[... cut ...]
2023-02-13T17:30:42.808Z zigbee-herdsman:adapter:ezsp:ezsp ==> addTransientLinkKey: {"partner":{"_value":{"type":"Buffer","data":[255,255,255,255,255,255,255,255]}},"transientKey":{"contents":{"type":"Buffer","data":[xxx]}}}
Zigbee2MQTT:info  2023-02-13 18:30:42: Zigbee: allowing new devices to join via Light Bulb IKEA Study 1.
Zigbee2MQTT:debug 2023-02-13 18:30:42: Received MQTT message on 'zigbee2mqtt/bridge/request/permit_join' with data '{"device":"Light Bulb IKEA Study 1","time":254,"transaction":"hzc8e-5","value":true}'
```
The issue seems to be quite random because I have 4 of these bulbs and 2 have this problem 2 do not. On 1.30.1 all of them work.

kirovilya, 2023-02-13:
@mateuszkozakiewicz Thanks, I'll see how I can fix it

kirovilya, 2023-02-13:
@mateuszkozakiewicz Are you trying to start pairing mode on a specific router device?
In your case, the following happens:
when trying to send a command to the device, the full address of the device is requested from stack (networkIdToEUI64), but the stack reports that it cannot determine the address.
check that there is a connection on the map with a device with this address.

mateuszkozakiewicz, 2023-02-13:
Indeed there is connection, the device reacts to all commands/queries except permit_join and behavior depends on the build version.

primeroz, 2023-03-03:
I am having the same issue , joining some [BTICINO K4003C](https://www.zigbee2mqtt.io/devices/K4003C_L4003C_N4003C_NT4003C.html)

they just `join` then immediately `leave` the network
```
./2023-03-02.05-22-30/log.txt:info  2023-03-03 12:04:57: Device '0x000474000119ec9e' joined
./2023-03-02.05-22-30/log.txt:info  2023-03-03 12:04:58: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x000474000119ec9e","ieee_address":"0x000474000119ec9e"},"type":"device_joined"}'
./2023-03-02.05-22-30/log.txt:info  2023-03-03 12:04:58: MQTT publish: topic 'zigbee2mqtt/0x000474000119ec9e/availability', payload '{"state":"online"}'
./2023-03-02.05-22-30/log.txt:info  2023-03-03 12:04:58: Starting interview of '0x000474000119ec9e'
./2023-03-02.05-22-30/log.txt:info  2023-03-03 12:04:58: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x000474000119ec9e","ieee_address":"0x000474000119ec9e","status":"started"},"type":"device_interview"}'
./2023-03-02.05-22-30/log.txt:info  2023-03-03 12:04:59: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x000474000119ec9e","ieee_address":"0x000474000119ec9e"},"type":"device_announce"}'
./2023-03-02.05-22-30/log.txt:info  2023-03-03 12:04:59: Successfully interviewed '0x000474000119ec9e', device has successfully been paired
./2023-03-02.05-22-30/log.txt:info  2023-03-03 12:04:59: Device '0x000474000119ec9e' is supported, identified as: BTicino Light switch with neutral (K4003C/L4003C/N4003C/NT4003C)
./2023-03-02.05-22-30/log.txt:info  2023-03-03 12:05:00: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"definition":{"description":"Light switch with neutral","exposes":[{"features":[{"access":7,"description":"On/off state of the switch","name":"state","property":"state","type":"binary","value_off":"OFF","value_on":"ON","value_toggle":"TOGGLE"}],"type":"switch"},{"access":1,"description":"Triggered action (e.g. a button click)","name":"action","property":"action","type":"enum","values":["identify","on","off"]},{"access":7,"description":"Enables the LED when the light is turned off, allowing to see the switch in the dark","name":"led_in_dark","property":"led_in_dark","type":"binary","value_off":"OFF","value_on":"ON"},{"access":7,"description":"Enables the LED when the light is turned on","name":"led_if_on","property":"led_if_on","type":"binary","value_off":"OFF","value_on":"ON"},{"access":2,"description":"Blinks the built-in LED to make it easier to find the device","name":"identify","property":"identify","type":"enum","values":["blink"]},{"access":1,"description":"Link quality (signal strength)","name":"linkquality","property":"linkquality","type":"numeric","unit":"lqi","value_max":255,"value_min":0}],"model":"K4003C/L4003C/N4003C/NT4003C","options":[{"access":2,"description":"State actions will also be published as 'action' when true (default false).","name":"state_action","property":"state_action","type":"binary","value_off":false,"value_on":true}],"supports_ota":false,"vendor":"BTicino"},"friendly_name":"0x000474000119ec9e","ieee_address":"0x000474000119ec9e","status":"successful","supported":true},"type":"device_interview"}'
./2023-03-02.05-22-30/log.txt:info  2023-03-03 12:05:00: Configuring '0x000474000119ec9e'
./2023-03-02.05-22-30/log.txt:info  2023-03-03 12:05:00: MQTT publish: topic 'homeassistant/switch/0x000474000119ec9e/switch/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state","value_template":"{{ value_json.state }}"},{"topic":"zigbee2mqtt/0x000474000119ec9e/availability","value_template":"{{ value_json.state }}"}],"availability_mode":"all","command_topic":"zigbee2mqtt/0x000474000119ec9e/set","device":{"identifiers":["zigbee2mqtt_0x000474000119ec9e"],"manufacturer":"BTicino","model":"Light switch with neutral (K4003C/L4003C/N4003C/NT4003C)","name":"0x000474000119ec9e","sw_version":"001c"},"name":"0x000474000119ec9e","payload_off":"OFF","payload_on":"ON","state_topic":"zigbee2mqtt/0x000474000119ec9e","unique_id":"0x000474000119ec9e_switch_zigbee2mqtt","value_template":"{{ value_json.state }}"}'
./2023-03-02.05-22-30/log.txt:info  2023-03-03 12:05:00: MQTT publish: topic 'homeassistant/sensor/0x000474000119ec9e/action/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state","value_template":"{{ value_json.state }}"},{"topic":"zigbee2mqtt/0x000474000119ec9e/availability","value_template":"{{ value_json.state }}"}],"availability_mode":"all","device":{"identifiers":["zigbee2mqtt_0x000474000119ec9e"],"manufacturer":"BTicino","model":"Light switch with neutral (K4003C/L4003C/N4003C/NT4003C)","name":"0x000474000119ec9e","sw_version":"001c"},"enabled_by_default":true,"icon":"mdi:gesture-double-tap","name":"0x000474000119ec9e action","state_topic":"zigbee2mqtt/0x000474000119ec9e","unique_id":"0x000474000119ec9e_action_zigbee2mqtt","value_template":"{{ value_json.action }}"}'
./2023-03-02.05-22-30/log.txt:info  2023-03-03 12:05:00: MQTT publish: topic 'homeassistant/switch/0x000474000119ec9e/switch_led_in_dark/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state","value_template":"{{ value_json.state }}"},{"topic":"zigbee2mqtt/0x000474000119ec9e/availability","value_template":"{{ value_json.state }}"}],"availability_mode":"all","command_topic":"zigbee2mqtt/0x000474000119ec9e/set/led_in_dark","device":{"identifiers":["zigbee2mqtt_0x000474000119ec9e"],"manufacturer":"BTicino","model":"Light switch with neutral (K40 [... cut ...]
./2023-03-02.05-22-30/log.txt:info  2023-03-03 12:05:00: MQTT publish: topic 'zigbee2mqtt/0x000474000119ec9e', payload '{"action":null,"led_if_on":null,"led_in_dark":null,"led_when_off":"OFF","led_when_on":"OFF","linkquality":54,"state":null}'
./2023-03-02.05-22-30/log.txt:info  2023-03-03 12:05:00: Successfully configured '0x000474000119ec9e'
./2023-03-02.05-22-30/log.txt:warn  2023-03-03 12:05:05: Device '0x000474000119ec9e' left the network
./2023-03-02.05-22-30/log.txt:info  2023-03-03 12:05:06: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x000474000119ec9e","ieee_address":"0x000474000119ec9e"},"type":"device_leave"}'
```

primeroz, 2023-03-03:
@Koenkk do you reckon this issue should be reopened ? it was just closed as stale ... thanks :pray:

mateuszkozakiewicz, 2023-03-03:
> please check on the latest dev/edge version of z2m, I made some changes for ezsp adapter

In my case this helped on all problematic devices I had, I don't have new devices to confirm though, because of the random nature of the issue I was hesitant to call it fixed

kirovilya, 2023-03-03:
@mateuszkozakiewicz @primeroz what version of z2m? 1.30.2 ?

mateuszkozakiewicz, 2023-03-03:
`latest-dev` from 3 weeks ago was working fine, this tag was since overwritten I suppose.

primeroz, 2023-03-03:
@kirovilya running 1.30.2-1 on Home Assistant

I tried pairing other devices and they work so i am really only having issues with this device BTICINO K4003C , but only some ... since i paired others fine this morning.

Is strange that in the logs i see `All good, paired and configured` and right after that they leave the network

[image]

kirovilya, 2023-03-03:
@primeroz :( don't know what else to do. I need traffic sniff and herdsman-logs  when the device leaves...

primeroz, 2023-03-03:
@kirovilya `herdsman-logs` i guess are not the same as the zigbee2mqtt logs i posted before ? i do i collect them ? 

Also, if you can point me how to do the sniff i will :pray:

kirovilya, 2023-03-03:
@primeroz no. it looks like https://github.com/Koenkk/zigbee2mqtt/issues/14568#issuecomment-1428369814

https://www.zigbee2mqtt.io/guide/usage/debug.html#zigbee-herdsman-debug-logging

primeroz, 2023-03-03:
thanks, will get on it after lunch :+1:

primeroz, 2023-03-03:
@kirovilya i hope these are the logs you were looking for, i can't find them in any file just in the ui and they scroll very fast

the device is `0x000474000119ecb6` 

```
2023-03-03T14:28:32.116Z zigbee-herdsman:adapter:zStack:znp:SREQ --> ZDO - bindReq - {"dstaddr":25007,"srcaddr":"0x000474000119ecb6","srcendpoint":1,"clusterid":3,"dstaddrmode":3,"dstaddress":"0x00124b00237e065b","dstendpoint":1}
2023-03-03T14:28:32.116Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,23,37,33,175,97,182,236,25,1,0,116,4,0,1,3,0,3,91,6,126,35,0,75,18,0,1,182]
2023-03-03T14:28:32.135Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,1,101,33,0,69]
2023-03-03T14:28:32.135Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,1,101,33,0,69]
2023-03-03T14:28:32.135Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 1 - 3 - 5 - 33 - [0] - 69
2023-03-03T14:28:32.135Z zigbee-herdsman:adapter:zStack:znp:SRSP <-- ZDO - bindReq - {"status":0}
2023-03-03T14:28:32.135Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2023-03-03T14:28:32.194Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,3,69,161,175,97,0,41]
2023-03-03T14:28:32.194Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,3,69,161,175,97,0,41]
2023-03-03T14:28:32.194Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 3 - 2 - 5 - 161 - [175,97,0] - 41
2023-03-03T14:28:32.194Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- ZDO - bindRsp - {"srcaddr":25007,"status":0}
2023-03-03T14:28:32.195Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2023-03-03T14:28:32.196Z zigbee-herdsman:controller:database:log Writing database to '/config/zigbee2mqtt/database.db'
2023-03-03T14:28:32.217Z zigbee-herdsman:controller:endpoint Bind 0x000474000119ecb6/1 genOnOff from '0x00124b00237e065b/1'
2023-03-03T14:28:32.217Z zigbee-herdsman:adapter:zStack:znp:SREQ --> ZDO - bindReq - {"dstaddr":25007,"srcaddr":"0x000474000119ecb6","srcendpoint":1,"clusterid":6,"dstaddrmode":3,"dstaddress":"0x00124b00237e065b","dstendpoint":1}
2023-03-03T14:28:32.217Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,23,37,33,175,97,182,236,25,1,0,116,4,0,1,6,0,3,91,6,126,35,0,75,18,0,1,179]
2023-03-03T14:28:32.242Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,1,101,33,0,69]
2023-03-03T14:28:32.242Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,1,101,33,0,69]
2023-03-03T14:28:32.243Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 1 - 3 - 5 - 33 - [0] - 69
2023-03-03T14:28:32.243Z zigbee-herdsman:adapter:zStack:znp:SRSP <-- ZDO - bindReq - {"status":0}
2023-03-03T14:28:32.243Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2023-03-03T14:28:32.311Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,28,68,129,0,0,6,0,175,97,1,1,0,147,0,98,87,23,0,0,8,24,1,6,1,0,0,88,2,137,198,18]
2023-03-03T14:28:32.311Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,28,68,129,0,0,6,0,175,97,1,1,0,147,0,98,87,23,0,0,8,24,1,6,1,0,0,88,2,137,198,18]
2023-03-03T14:28:32.314Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [177,254,3,69,161,175,97,0,41]
2023-03-03T14:28:32.314Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,28,68,129,0,0,6,0,175,97,1,1,0,147,0,98,87,23,0,0,8,24,1,6,1,0,0,88,2,137,198,18,177,254,3,69,161,175,97,0,41]
2023-03-03T14:28:32.315Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 28 - 2 - 4 - 129 - [0,0,6,0,175,97,1,1,0,147,0,98,87,23,0,0,8,24,1,6,1,0,0,88,2,137,198,18] - 177
2023-03-03T14:28:32.315Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - incomingMsg - {"groupid":0,"clusterid":6,"srcaddr":25007,"srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"linkquality":147,"securityuse":0,"timestamp":1529698,"transseqnumber":0,"len":8,"data":{"type":"Buffer","data":[24,1,6,1,0,0,88,2]}}
2023-03-03T14:28:32.320Z zigbee-herdsman:controller:log Received 'zcl' data '{"frame":{"Header":{"frameControl":{"frameType":0,"manufacturerSpecific":false,"direction":1,"disableDefaultResponse":true,"reservedBits":0},"transactionSequenceNumber":1,"manufacturerCode":null,"commandIdentifier":6},"Payload":[{"direction":1,"attrId":0,"timeout":600}],"Command":{"ID":6,"name":"configReport","parameters":[{"name":"direction","type":32},{"name":"attrId","type":33},{"name":"dataType","type":32,"conditions":[{"type":"directionEquals","value":0}]},{"name":"minRepIntval","type":33,"conditions":[{"type":"directionEquals","value":0}]},{"name":"maxRepIntval","type":33,"conditions":[{"type":"directionEquals","value":0}]},{"name":"repChange","type":1000,"conditions":[{"type":"directionEquals","value":0},{"type":"dataTypeValueTypeEquals","value":"ANALOG"}]},{"name":"timeout","type":33,"conditions":[{"type":"directionEquals","value":1}]}],"response":7}},"address":25007,"endpoint":1,"linkquality":147,"groupID":0,"wasBroadcast":false,"destinationEndpoint":1}'
2023-03-03T14:28:32.322Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,3,69,161,175,97,0,41]
2023-03-03T14:28:32.322Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 3 - 2 - 5 - 161 - [175,97,0] - 41
2023-03-03T14:28:32.322Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- ZDO - bindRsp - {"srcaddr":25007,"status":0}
2023-03-03T14:28:32.323Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2023-03-03T14:28:32.324Z zigbee-herdsman:controller:database:log Writing database to '/config/zigbee2mqtt/database.db'
[... cut ...]
2023-03-03T14:28:39.280Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- ZDO - leaveInd - {"srcaddr":25007,"extaddr":"0x000474000119ecb6","request":0,"removechildren":0,"rejoin":0}
2023-03-03T14:28:39.281Z zigbee-herdsman:controller:log Device leave '0x000474000119ecb6'
2023-03-03T14:28:39.281Z zigbee-herdsman:controller:log Removing device from database '0x000474000119ecb6'
2023-03-03T14:28:39.282Z zigbee-herdsman:controller:database:log Writing database to '/config/zigbee2mqtt/database.db'
Zigbee2MQTT:warn  2023-03-03 15:28:39: Device '0x000474000119ecb6' left the network
```

I don't think i can do the sniffing since i don't have a spare zigbee device

kirovilya, 2023-03-03:
@primeroz Ok. Now I see that the device itself has decided to leave the network...
```<-- ZDO - leaveInd```
Hmm. why did it decide to leave...

primeroz, 2023-03-03:
Yeah :) 

I do have one of this that joined and worked just fine ... i have been wondering if maybe is a firmware difference but i can't see what firmware this one that decide to leave has

primeroz, 2023-03-03:
@kirovilya 

not sure if it helps but this is my configuration

```
data_path: /config/zigbee2mqtt
socat:
  enabled: false
  master: pty,raw,echo=0,link=/tmp/ttyZ2M,mode=777
  slave: tcp-listen:8485,keepalive,nodelay,reuseaddr,keepidle=1,keepintvl=1,keepcnt=5
  options: "-d -d"
  log: true
mqtt: {}
serial:
  port: /dev/ttyUSB0
zigbee_shepherd_devices: false
zigbee_herdsman_debug: false
```

my coordinator is a `ZigStar Stick v4 Zigbee Coordinator on CC2652P` https://www.zigbee2mqtt.io/guide/adapters/#recommended

primeroz, 2023-03-03:
@kirovilya would a `limit of devices` connected to the coordinator show up like this ? I don't think so ( since the device joins and then leave the network and also since i only have 19 devices connected to the coordintator between routers and end devices ) but just thought i would ask. 

this is driving me crazy :(

primeroz, 2023-03-03:
Should i create a new issue since this one is actually closed ? 

Let me know if there is any troubleshotting i can do that might help you identify the issue ( other than the sniffing part which i can't really unless i buy a new device )

primeroz, 2023-03-04:
@kirovilya  Just answering myself ... in my case i had to `repress`  the reset button ( the same one used to retrigger pairing ) as soon as the device showed up in the Z2M UI before it left (  a window of a few second ) . this way it stayed paired and joined 

I also tried with a `Conbee 2` and the exact same behaviour was observed so nothing to do with Z2M 

Should i add this to the docs for the device ?

kirovilya, 2023-03-05:
@primeroz I think yes, you can add a note on the device page.
I do not know the obvious reasons why it leaves the network, perhaps it did not receive any response from the coordinator during the interview, or "asleep" (and therefore did not receive it).

mateuszkozakiewicz, 2023-03-05:
I came to a similar conclusion, the device is not aware that the pairing process has completed from coordinator side and tried again which results in coordinator kicking it out. For me however pressing the pair button again/pulling the battery/disallowing joining did not help.

primeroz, 2023-03-05:
For me it worked on about 6 devices. I Just press it a few times when the led turn red a d the devices shows up in th Z2m ui

cklam2, 2023-03-09:
I had the same issue with the Aqara Roller blinds driver when using the new Sonoff Zigbee 3.0 Plus-E ('ezsp' adapter in configuration.yaml). I tried both regular and edge version of Zigbee2Mqtt but no luck.
When I switched back to my old CC2531 stick, adding the same device was no issue at all. So guess it's either the Sonoff or the adapter.

akrawiecpgs, 2023-03-15:
The same behavior 
```2023-03-15T21:40:09.749Z zigbee-herdsman:adapter:ezsp:ezsp <== 0x24: {"_id_":36,"_cls_":"trustCenterJoinHandler","_isRequest_":false,"newNodeId":3228,"newNodeEui64":{"type":"Buffer","data":[164,193,56,206,8,194,71,222]},"status":2,"policyDecision":3,"parentOfNewNodeId":65535}
2023-03-15T21:40:09.749Z zigbee-herdsman:adapter:ezsp:debg Device left network request received: 3228 a4c138ce08c247de
2023-03-15T21:40:09.749Z zigbee-herdsman:controller:log Device leave '0xa4c138ce08c247de'
Zigbee2MQTT:warn  2023-03-15 22:40:09: Device 'SWTZ26' left the network```

Zigbee2MQTT: 1.29.1-1 installed and upgraded to  1.30.2-1
Device : 'curtain blinder relay' SWTZ26 Tuya
Coordinator : Zigbee 3.0  Sonoff Dongle-E

bartzz, 2023-04-06:
I'm having the same issue with one device. I tried removing it completely from Z2M, checking "retain" to `true`, but the device still leaves the network in 1-24h (mostly within 1-2 hours)

```
Zigbee2MQTT:info  2023-04-06 09:35:23: MQTT publish: topic 'zigbee2mqtt/korytarz', payload '{"battery":63,"humidity":59.77,"last_seen":"2023-04-06T07:35:23.343Z","linkquality":108,"power_outage_count":2078,"pressure":1010.6,"temperature":19.81,"voltage":2945}'

Zigbee2MQTT:warn  2023-04-06 09:39:38: Device 'korytarz' left the network

Zigbee2MQTT:info  2023-04-06 09:39:38: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"korytarz","ieee_address":"0x00158d0008025b89"},"type":"device_leave"}'

```

adams-family, 2023-04-10:
I'm having the same issue with an IKEA Tradrfri switch, somewhere between 1-24 hours it decides to leave the network:

```
warn  2023-04-09 22:12:08: Device 'IkeaSwitchKitchen' left the network
```

I'm using a Sonoff Zigbee coordinator connected to a RaspberryPi directly via USB. Zigbee2Mqtt runs in a docker container, version: `1.30.1 (commit #eb878d3)`.

Before this setup, I was using Tube's ZB bridge over ethernet and a much order version of Zigbee2Mqtt (like 1-2 years old) and I had no issues with the same devices.

Would appreciate someone's help on this. I have 'wife acceptance issues' with my setup 😂

Foristkirito, 2023-05-10:
same issue with aqara fp1

jmaragon, 2023-05-20:
I was having the same issues with some Netatmo cover switches... the solution from @primeroz about repressing the reset button once the device joined indeed worked for me.

primeroz, 2023-05-20:
:+1: i did not have problems since then , I have about 15 switches working great

eligao, 2024-02-07:
Had a same issue with STYRBAR E2002. Pulling the battery when it was about to leave the network did the trick.

patpac9, 2024-02-22:
> Had a same issue with STYRBAR E2002. Pulling the battery when it was about to leave the network did the trick.

Can confirm did the same thing with my Tint Bulb. I just cut the power right after it joined the network.
Since then it stayed in my network. 👍🏻 Took me 2h filled with pure anger and hate to figure this out.

eligao, 2024-03-11:
Seems fixed in the latest release? Not sure which commit though.

mateuszkozakiewicz, 2024-03-11:
For me migrating to a different coordinator (SMLIGHT SLZB-06) fixed all issues. It's been months since I had to do check or debug anything.

Satalicious, 2024-05-15:
not fixed yet, experiencing this on the latest firmware.

AkshayRao27, 2024-07-02:
I just "upgraded" to `ember` and now one single Aqara H1 Switch (double rocker, with neutral, EU) is having this issue. I even tried force-removing it before it could leave by itself and then pairing it again, but that did not help. The rest of my Aqara H1 Switches work fine, it's just this one single switch that's problematic. 

```
Zigbee2MQTT version
    [1.39.0](https://github.com/Koenkk/zigbee2mqtt/releases/tag/1.39.0) commit: [unknown](https://github.com/Koenkk/zigbee2mqtt/commit/unknown)

Coordinator type
    EmberZNet

Coordinator revision
    7.4.2 [GA]
    ```

CodeForLove83, 2024-09-04:
Hi,

I'm wondering if the issue is related to missing response for:

- cluster_id = 0x000
- command = 0x00 (Read Attribute)
- attribute = 0xf000

The following pictures shows a successful pairing with deconz:
Switch device sends:

[image]

It receives the response:

[image]

I haven't investigated how to fix z2m to handle this request.
I'm wondering if @Koenkk can help us on this.

CodeForLove83, 2024-09-05:
@Koenkk, am I wrong or z2m is supposed to handle the attribute 0xf000?
I found this piece of [code](https://github.com/Koenkk/zigbee-herdsman-converters/blob/b0a673ff81245f03f1f30b9dc8023ab59363c230/src/index.ts#L442) but I'm not sure it's correct, a 0 is missing?

`                **const payload = {0xf00: {value: 23, type: 35}};**`

Shouldn't it be?:
`                **const payload = {0xf000: {value: 23, type: 35}};**`

Koenkk, 2024-09-05:
@CodeForLove83 it looks like a `0` is missing indeed. Can you try to apply this change locally? e.g. if you are running in Docker:

```bash
docker exec -it CONTAINER_NAME sh
apk add nano
nano node_modules/zigbee-herdsman-converters/index.js
# Make the change
exit
docker restart CONTAINER_NAME
```

jlaska, 2024-09-06:
@Koenkk Moved a Aqara contact sensor to a new coordinator and decommissioned the old coordinator.  Started experiencing this issue where the device would leave the network recently after joining.  Testing with the [provided change](https://github.com/Koenkk/zigbee2mqtt/issues/14568#issuecomment-2330683086) on top of [1.39.0](https://github.com/Koenkk/zigbee2mqtt/releases/tag/1.39.0) commit: [0326926](https://github.com/Koenkk/zigbee2mqtt/commit/0326926).  The device remains on the network and appears to be updating successfully for the last 4 hours.  Previously it would leave the network after a few minutes.

Koenkk, 2024-09-06:
@jlaska this fix is only for Ubisys devices.

Koenkk, 2024-09-06:
@jlaska do you have the debug of - 5 mins before the leave up to the leave?

See [this](https://www.zigbee2mqtt.io/guide/configuration/logging.html) on how to enable debug logging.

jlaska, 2024-09-06:
> @jlaska do you have the debug of - 5 mins before the leave up to the leave?
> 
> See [this](https://www.zigbee2mqtt.io/guide/configuration/logging.html) on how to enable debug logging.

I'll recreate with debugging enabled.  Would you like that here, or tracked as a new issue?

jlaska, 2024-09-07:
@Koenkk Recreated the issue with debug logging enabled.  The device leaving the network this time is ...

```json
{"friendly_name":"Kids Bathroom PersonH Motion","ieee_address":"0x00158d0006a24cac"}
``` 

Debug logs for ~5 mins before leaving the network available at https://gist.github.com/jlaska/1cddc54e5eefd77260d0f2183938aab0

Koenkk, 2024-09-08:
The log starts with the leave `z2m: Device 'Kids Bathroom PersonH Motion' left the network`, is nothing logged 5 mins before this?

jlaska, 2024-09-08:
Hi @Koenkk .. the log is sorted in reverse with newest messages at the top.  It starts at `01:00:00.625` and ends with the device leaving at `01:06:59.466`

jlaska, 2024-09-08:
@Koenkk Updated the [debug log](https://gist.github.com/jlaska/1cddc54e5eefd77260d0f2183938aab0#file-explore-logs-2024-09-07-08_32_07-txt-L175-L182) to sort as expected. Let me know if that's what you need.

Koenkk, 2024-09-09:
I don't see any strange in the logs here, either it's an issue of the device or another devices kicks it from the network (which you can figure out via [sniffing](https://www.zigbee2mqtt.io/advanced/zigbee/04_sniff_zigbee_traffic.html))

jlaska, 2024-09-09:
@Koenkk Thank you for looking!  A thought ... I am using a former zzh coordinator (CC2652R Stick), reflashed as a router and added to a new zigbee network.  My disappearing devices didn't start happening until I added the reflashed router.  I'm realizing now that flashing may not erase the NVRAM.  I'll explore that and [sniffing](https://www.zigbee2mqtt.io/advanced/zigbee/04_sniff_zigbee_traffic.html).  Thanks!

oberschwabe, 2024-09-16:
same issue here with BSEED 1G Switch and currently with a PowerPlug
Zigbee2MQTT version : 1.40.1 commit: 403d3c09
Coordinator type : zStack3x0
Coordinator revision : 20230507

any idea how to resolve this?

[ts011f_plug.log](https://github.com/user-attachments/files/17018224/ts011f_plug.log)