Koenkk/zigbee2mqtt #14570: Lonsonho dimmers misbehaving

yasapl, 2022-10-20:
### What happened?

Recently I have started getting issues with Lonsonho dimmers QS-Zigbee-D02-TRIAC-2C-L.
Switching on from Z2M randromly works. It is usually possible to switch it off but rarely on, especially following the use of physical buttons. When switch it on in z2m, it will show on but lights will not come up. Refreshing switch status in z2m will flip it back to off position. Then, turning light with the physical button will make them fully dimmed which in my case mean they don’t lit at all, only in z2m I can see they are on and brightness is set to 0.
When not touching z2m the dimmer works perfectly fine.
They worked fine for a couple of months and only recently started “playinig” with me. I have two of these dimmers and both are with the same issues.

Tried dev addon - same effect

### What did you expect to happen?

the dimmer works correctly using z2m

### How to reproduce it (minimal and precise)

I don’t know 🙁

### Zigbee2MQTT version

1.28 HA addon

### Adapter firmware version

20220219

### Adapter

Sonoff zigbee 3.0 usb dongle plus

yasapl, 2022-10-29:
Now I understand as I have played with these options too. Thanks!

dariusz-goc, 2024-12-06:
For me the transitions aren't working at all with z2m 1.42.0-dev commit: [639e3bb](https://github.com/Koenkk/zigbee2mqtt/commit/639e3bb), zlzb-06 firmware 20240710. The light just turns off immediately. Any ideas?