Koenkk/zigbee2mqtt #14571: unsupported device with Zigbee model 'TS0601' and manufacturer name '_TZE200_yojqa8xn'

hasifm2010, 2022-10-21:
### Link

https://www.aliexpress.com/item/1005003944885010.html?spm=a2g0o.order_list.0.0.21ef1802vpsuEp

### Database entry

Received message from unsupported device with Zigbee model 'TS0601' and manufacturer name '_TZE200_yojqa8xn'

### Comments

Warning 2022-10-21 11:04:10Received message from unsupported device with Zigbee model 'TS0601' and manufacturer name '_TZE200_yojqa8xn'
Warning 2022-10-21 11:04:10Please see: https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html

Koenkk, 2022-10-21:
Could you update to the latest dev  (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) and check if the issue is fixed with the following external converter:

```js
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const ota = require('zigbee-herdsman-converters/lib/ota');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const utils = require('zigbee-herdsman-converters/lib/utils');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    fingerprint: tuya.fingerprint('TS0601', ['_TZE200_ggev5fsl', '_TZE200_u319yc66', '_TZE200_yojqa8xn']),
    model: 'TS0601_gas_sensor',
    vendor: 'TuYa',
    description: 'gas sensor',
    fromZigbee: [tuya.fzDataPoints],
    toZigbee: [tuya.tzDataPoints],
    configure: tuya.configureMagicPacket,
    exposes: [e.gas(), tuya.exposes.selfTest(), tuya.exposes.selfTestResult(), tuya.exposes.faultAlarm(), tuya.exposes.silence()],
    meta: {
        tuyaDatapoints: [
            [1, 'gas', tuya.valueConverter.true0ElseFalse],
            [8, 'self_test', tuya.valueConverter.raw],
            [9, 'self_test_result', tuya.valueConverter.selfTestResult],
            [11, 'fault_alarm', tuya.valueConverter.trueFalse],
            [16, 'silence', tuya.valueConverter.raw],
        ],
    },
};

module.exports = definition;

```

- save this as file next to `configuration.yaml` as `ext_converter.js`
- add it to `configuration.yaml`:
```yaml
external_converters:
  - ext_converter.js
```
- start z2m, check if device works

hasifm2010, 2022-10-22:
Thanks.. Zigbee2mqtt edge can learn device now after following above steps. However, i am getting below warning in logs.

Warning 2022-10-22 09:34:41Datapoint 2 not defined for '_TZE200_yojqa8xn' with data {"dp":2,"datatype":2,"data":{"type":"Buffer","data":[0,0,0,2]}}

Koenkk, 2022-10-22:
I'm not sure what this value means; to find out: https://www.zigbee2mqtt.io/advanced/support-new-devices/03_find_tuya_data_points.html#_7-find-your-devices

Ucsus, 2022-10-29:
> 

[14:34:32] INFO: Preparing to start...
[14:34:33] INFO: Socat not enabled
[14:34:38] INFO: Starting Zigbee2MQTT...
/app/dist/util/externally-loaded.js:20
    exposes: [e.gas(), tuya.exposes.selfTest(), tuya.exposes.selfTestResult(), tuya.exposes.faultAlarm(), tuya.exposes.silence()],
                                    ^
TypeError: tuya.exposes.selfTest is not a function
    at /app/dist/util/externally-loaded.js:20:37
    at Script.runInContext (node:vm:141:12)
    at Script.runInNewContext (node:vm:146:17)
    at Object.runInNewContext (node:vm:306:38)
    at loadModuleFromText (/app/lib/util/utils.ts:148:8)
    at loadModuleFromFile (/app/lib/util/utils.ts:155:12)
    at Object.getExternalConvertersDefinitions (/app/lib/util/utils.ts:165:25)
    at getExternalConvertersDefinitions.next (<anonymous>)
    at new ExternalConverters (/app/lib/extension/externalConverters.ts:12:20)
    at new Controller (/app/lib/controller.ts:84:58)

Koenkk, 2022-10-29:
@Ucsus you are probably not using the latest dev branch: https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html

Ucsus, 2022-10-31:
> @Ucsus you are probably not using the latest dev branch: https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html

I'm using HA, edge version

Koenkk, 2022-10-31:
Great!

closed 2022-10-31 by Koenkk

pauloon, 2022-11-05:
Did anyone could improve the converter to get the gas reading on the Zigbee version?
I've also have the wifi one and it gives me the readings...

This converter works but looks a little glitched. I get many errors on Z2M when detection is triggered,

sercio, 2022-11-08:
Hello,
today I have received the tuya zigbee gateway.... the datapoints are more different hen the converter ... I have found:

1 Gas detection state		
2 Detected Gas
6 Alarm Ringtone
7 Alarm Time
8 Auto-detect
9 Auto-detect Result
10 Preheat

I don't understand how to find the datatype and the values for create a correct converter... can anyone help me?

Koenkk, 2022-11-09:
@sercio can you provide the data/database.db entry of your device so I can write the converter?

sercio, 2022-11-09:
> 

this is the entry in z2m database.db

{"id":2,"type":"Router","ieeeAddr":"0xa4c1381aea7ee7a6","nwkAddr":2556,"manufId":4417,"manufName":"_TZE200_yojqa8xn","powerSource":"Mains (single phase)","modelId":"TS0601","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[4,5,61184,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65503":"\u0019��)\u0017S\u0000\u0000\u0000eT\u0000\u0000\u0000\u0012","65506":54,"65508":1,"modelId":"TS0601","manufacturerName":"_TZE200_yojqa8xn","powerSource":1,"zclVersion":3,"appVersion":70,"stackVersion":0,"hwVersion":1,"dateCode":""}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":70,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1662022924671,"defaultSendRequestWhen":"immediate"}

Do you need other information?

Koenkk, 2022-11-09:
With the following converter https://gist.github.com/Koenkk/86ea747a683752fd6ff9390abf9fda55 everything should be exposed in the z2m frontend -> device -> state. However these are the raw values, we need to map them to more human friendly ones.

For each data point, can you give more context on what it does and what the possible options are?

sercio, 2022-11-10:
OK, I can try. There is a tool or a method or I must to try all the options?

sercio, 2022-11-11:
OK, I try.

1 Gas detection state        Normal   (no gas) 
2 Detected Gas    0.00LEL   (level of gas)
6 Alarm Ringtone 1 (number of alarm ringtone from 1 to 5 )
7 Alarm Time   180s  (alarm duration from 1 to 180s)
8 Auto-detect  (auto  test: value on, off)
9 Auto-detect Result (auto test Result value processing, success)
10 Preheat  (value on, off)

This is what I understand using  app and tuya device debug

Koenkk, 2022-11-11:
Thanks, can you check what appears in the z2m frontend -> device -> state with the following external conveter? https://gist.github.com/Koenkk/86ea747a683752fd6ff9390abf9fda55

sercio, 2022-11-11:
OK,
but it's impossible for me before 24/11. I'll send you the information as soon as possible.
Thanks

hasifm2010, 2022-11-11:
this is what i am getting when i use this converter https://gist.github.com/Koenkk/86ea747a683752fd6ff9390abf9fda55

{
    "detected_gas": 1,
    "gas": false,
    "last_seen": "2022-11-12T00:59:58+04:00",
    "linkquality": 69,
    "self_test": false,
    "self_test_result": "success",
    "silence": false
}

Masterz69, 2022-11-30:
Some time ago found somewhere DP info for '_TZE200_yojqa8xn'
```
sensor_status:   1 - enum (0-detected,1-not detected)
sensor_value:    2 - raw, DivideBy10
alarm_ringtone:  6
alarm_time:      7
self_checking:   8 - work with tuya.exposes.selfTest() & tuya.valueConverter.raw
checking_result: 9 - work with tuya.exposes.selfTestResult() & tuya.valueConverter.selfTestResult
preheat:        10
alarm_switch:   13
muffling:       16 - raw
```
usable exposes & converters checked by me. Still working on alarm_ringtone & time, as well alarm_switch.

Masterz69, 2022-11-30:
tIMHO tuya.tz.datapoints have whitelist for keys used in meta.tuyaDatapoints in converter.
Details described [here](https://gist.github.com/Koenkk/86ea747a683752fd6ff9390abf9fda55?permalink_comment_id=4386497#gistcomment-4386497).

Shortly so far my little uderstanding of code: external converter tuyaDatapoints definition ```[13, 'alarm_switch', tuya.valueConverter.raw]``` does not pass tuya.tz.datapoints with message 'No converter available ...'.

Dinth, 2023-01-07:
According to Blakadders website this should be supported, but its not detected correctly in my (edge) installation:
[image]

Koenkk, 2023-01-08:
`_TZE200_yojqa8xn` is supported in z2m 1.29.1

Dinth, 2023-01-08:
It actually is, thanks!

sercio, 2023-02-08:
Hi,
I receive the gas status unknown into Home assistant after update (now 1.30.1 zigbee2mqtt version). Analyzing the status into zigbee2mqtt I see:

{
    "alarm_ringtone": "1",
    "alarm_time": 180,
    "gas": "false",
    "gas_value": 0,
    "linkquality": 47,
    "preheat": false,
    "self_test": false,
    "self_test_result": "success"
}

I suppose the problem is that the gas attribute arrive as a text string and not as false/true value....
can you check?

Masterz69, 2023-02-10:
> 

Looks issue to be opened.
I still have 1.29.2 Z2M version - all looks fine, see below.
IMHO something changed in newer version which broke sensor reading.

```
{
    "alarm_ringtone": "RingTone 2",
    "alarm_switch": false,
    "alarm_time": 120,
    "gas": "not detected",
    "gas_value": 0,
    "last_seen": "2023-02-10T18:09:51+02:00",
    "linkquality": 78,
    "preheat": false,
    "self_test": false,
    "self_test_result": "success",
    "silence": null
}
```

[image]

Masterz69, 2023-02-10:
> It actually is, thanks!

Hi, have You upgraded Z2M to 1.30.1+ ? 
If Yes - have same issue as @sercio in https://github.com/Koenkk/zigbee2mqtt/issues/14571#issuecomment-1422426960 ?

Masterz69, 2023-02-10:
Oops, just checked binary_sensor. In 1.29.2 it also have 'unknown' state :(

[image]

Masterz69, 2023-02-14:
Hi, @sercio, @Dinth, @hasifm2010 !
Can You check latest dev branch in Your system ?
Possibly issues fixed in https://github.com/Koenkk/zigbee2mqtt/issues/16615#issuecomment-1428346422.

sercio, 2023-02-14:
Yes, I can test tomorrow.
Thanks

Inviato da Outlook per Android<https://aka.ms/AAb9ysg>

Possibly issues fixed in #16615 (comment)<https://github.com/Koenkk/zigbee2mqtt/issues/16615#issuecomment-1428346422>.

sercio, 2023-02-15:
OK, tested dev brach and for me all is solved:

Correct Value on gas attribute, no error when try to set Alarm Ringtone and Alarm Time.

klipper-vp, 2023-04-09:
Zigbee Manufacturer
_TZE204_yojqa8xn
Zigbee Model
TS0601
[image]

klipper-vp, 2023-04-09:
I have the same problem as not supported

alex-shl, 2023-04-26:
I have the same problem :(

programmieramt, 2023-04-28:
Hello, I have the same device but again with another manufacturer name:
`Received message from unsupported device with Zigbee model 'TS0601' and manufacturer name '_TZE204_sxm7l9xa`

36grad, 2023-05-10:
> Hello, I have the same device but again with another manufacturer name: `Received message from unsupported device with Zigbee model 'TS0601' and manufacturer name '_TZE204_sxm7l9xa`

Exactly the same issue here, ordered 1 week ago from Ali. Seems they keep happily changing the manufacturer ID.

Roman-RV78, 2023-07-13:
Hello. There is a problem with gas detected, the value is always false. Can anyone confirm this with this device? As I understand when pressing self_test, gas detect should be true. Z2M 1.32.1-1

Dinth, 2023-07-13:
Can you try testing in a real life scenario? 
For me spraying Isopropyl alcohol or even putting a large amount of wine into a pine was triggering alarms in the past.

Roman-RV78, 2023-07-13:
I have two such devices, they were both tested for gas as well. The devices themselves emit a beep, but gas detect is always false

mauriceatkinson, 2023-08-18:
I have the same issue, I purchased the TS0601 and am seeing the follwoing error message:

Zigbee2MQTT:warn  2023-08-18 10:36:30: Received message from unsupported device with Zigbee model 'TS0601' and manufacturer name '_TZE200_eo6xhfbo'
'
I see the TS0601 being supported but the manufacturers name is different.

mattaipm, 2025-01-15:
After upgrade z2m to version [2.0.0]   devices TS0601 is registered.  Can't be controlled
Status
{
    "anti_scaling": "ON",
    "away_mode": "OFF",
    "battery_low": false,
    "child_lock": "LOCK",
    "current_heating_setpoint": 18,
    "frost_detection": "ON",
    "heating": "OFF",
    "last_seen": 1736955819649,
    "linkquality": 240,
    "local_temperature": 19.5,
    "local_temperature_calibration": 0,
    "preset_mode": "none",
    "running_state": "idle",
    "system_mode": "auto",
    "window_detection": "ON"
}

Console

const definition = {
    zigbeeModel: ['TS0601'],
    model: 'TS0601',
    vendor: '_TZE284_3yp57tby',
    description: 'Automatically generated definition',
    extend: [],
    meta: {},
};

module.exports = definition;

drascom, 2025-10-01:
i have TS0601 model door sensor which is not recognized at zigbee2mqtt. used config below and it worked . put code inside `/config/zigbee2mqtt/external_converters/device_code_door.js` then restart and re-pair device 
it may help someone..
Tuya reuse “TS0601” for many products; DP numbers often vary. If your logs show, say, contact on DP 101 and battery on 2, just swap those in the array.

// tze284_u8ouaqsz_door.js
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const e = exposes.presets;
const ea = exposes.access;

module.exports = {
  fingerprint: [{modelID: 'TS0601', manufacturerName: '_TZE284_u8ouaqsz'}],
  model: 'TS0601',
  vendor: 'Tuya',
  description: 'Door sensor (u8ouaqsz) with optional siren',
  fromZigbee: [tuya.fz.datapoints],
  toZigbee: [tuya.tz.datapoints],
  configure: tuya.configureMagicPacket,
  exposes: [
    e.contact(),
    e.battery(),
    e.binary('alarm', ea.STATE_SET, 'ON', 'OFF'),
    e.binary('strobe', ea.STATE_SET, 'ON', 'OFF'),
    e.numeric('volume', ea.STATE_SET).withUnit('%').withValueMin(1).withValueMax(100),
    e.numeric('duration', ea.STATE_SET).withUnit('s').withValueMin(3).withValueMax(180),
  ],
  meta: {
    // Keep the DPs that already work for you. If you used my earlier mapping and it works, leave it.
    tuyaDatapoints: [
      [1,   'contact',  tuya.valueConverter.trueFalse], // your logs show correct toggling
      [2,   'battery',  tuya.valueConverter.raw],       // reports 100 -> use as %
      [101, 'alarm',    tuya.valueConverter.onOff],
      [102, 'strobe',   tuya.valueConverter.onOff],
      [103, 'volume',   tuya.valueConverter.raw],
      [104, 'duration', tuya.valueConverter.raw],
    ],
  },
};