Koenkk/zigbee2mqtt #14572: Error: Failure send addTransientLinkKey

kaciker, 2022-10-21:
### What happened?

While I was in version 1.27.2-1, everythin was working well
I upgraded to 1.27.2-2 and this porblem arrieved, my zigbee2mqtt is down after works from several minuts.
Tha last lines in the log are:

**_Error: Failure send addTransientLinkKey:{"type":"Buffer","data":[59,0,1,175,0,255,255,255,255,255,255,255,255,90,105,103,66,101,101,65,108,108,105,97,110,99,101,48,57]}
    at /app/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/ezsp.ts:507:23
    at Queue.executeNext (/app/node_modules/zigbee-herdsman/src/utils/queue.ts:32:32)_**

I resetored previous version and worked
When version 1.28.0 was abailible, I upgraded again, but I had the same problem.
Again I restored the version 1.27.2-1 and started to work again.

### What did you expect to happen?

Make work

### Zigbee2MQTT version

> 1.27.2-1   (Inside HA) 

### Adapter firmware version

6.7.8.0 build 373

### Adapter

EZSP v8  (Sonoff Zigbee wifi)

kirovilya, 2022-10-21:
Try to update to 1.28.0-dev (edge) version

kaciker, 2022-10-21:
> Try to update to 1.28.0-dev (edge) version

OK, I installed it, and I'm checking the log

kaciker, 2022-10-21:
> Try to update to 1.28.0-dev (edge) version

Hello
Unfortunatelly, after a cople of hours working, sytem drops again :

**_Zigbee2MQTT:info  2022-10-21 12:08:02: Logging to console and directory: '/config/zigbee2mqtt/log/2022-10-21.12-08-02' filename: log.txt
Zigbee2MQTT:info  2022-10-21 12:08:02: Starting Zigbee2MQTT version 1.28.0-dev (commit #939fdf0)
Zigbee2MQTT:info  2022-10-21 12:08:02: Starting zigbee-herdsman (0.14.66)
Zigbee2MQTT:error 2022-10-21 12:08:17: Error while starting zigbee-herdsman
Zigbee2MQTT:error 2022-10-21 12:08:17: Failed to start zigbee
Zigbee2MQTT:error 2022-10-21 12:08:17: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start.html for possible solutions
Zigbee2MQTT:error 2022-10-21 12:08:17: Exiting...
Zigbee2MQTT:error 2022-10-21 12:08:17: Error: Failure send networkInit:{"type":"Buffer","data":[22,0,1,23,0]}
    at /app/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/ezsp.ts:510:23
    at Queue.executeNext (/app/node_modules/zigbee-herdsman/src/utils/queue.ts:32:32)_**

kirovilya, 2022-10-21:
It is very strange. looks like a connection problem. but it's better to write a debug log to understand the problem
https://www.zigbee2mqtt.io/guide/usage/debug.html#zigbee-herdsman-debug-logging

kaciker, 2022-10-23:
> It is very strange. looks like a connection problem. but it's better to write a debug log to understand the problem
> 
> https://www.zigbee2mqtt.io/guide/usage/debug.html#zigbee-herdsman-debug-logging

Hello, I activate the option debug, but I can not see additional information in this log. 
This is the Darius today after 12h working 

[23:03:52] INFO: Preparing to start...
[23:03:52] INFO: Socat not enabled
[23:03:53] INFO: Starting Zigbee2MQTT...
Zigbee2MQTT:debug 2022-10-22 23:03:55: Loaded state from file /config/zigbee2mqtt/state.json
Zigbee2MQTT:info  2022-10-22 23:03:55: Logging to console and directory: '/config/zigbee2mqtt/log/2022-10-22.23-03-55' filename: log.txt
Zigbee2MQTT:debug 2022-10-22 23:03:55: Removing old log directory '/config/zigbee2mqtt/log/2022-10-22.21-41-38'
Zigbee2MQTT:info  2022-10-22 23:03:55: Starting Zigbee2MQTT version 1.28.0-dev (commit #939fdf0)
Zigbee2MQTT:info  2022-10-22 23:03:55: Starting zigbee-herdsman (0.14.66)
Zigbee2MQTT:debug 2022-10-22 23:03:55: Using zigbee-herdsman with settings: '{"adapter":{"concurrent":null,"delay":null,"disableLED":false},"backupPath":"/config/zigbee2mqtt/coordinator_backup.json","databaseBackupPath":"/config/zigbee2mqtt/database.db.backup","databasePath":"/config/zigbee2mqtt/database.db","network":{"channelList":[11],"extendedPanID":[221,221,221,221,221,221,221,221],"networkKey":"HIDDEN","panID":6754},"serialPort":{"adapter":"ezsp","path":"tcp://192.168.31.20:8888"}}'
Zigbee2MQTT:error 2022-10-22 23:04:00: Error while starting zigbee-herdsman
Zigbee2MQTT:error 2022-10-22 23:04:00: Failed to start zigbee
Zigbee2MQTT:error 2022-10-22 23:04:00: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start.html for possible solutions
Zigbee2MQTT:error 2022-10-22 23:04:00: Exiting...
Zigbee2MQTT:error 2022-10-22 23:04:00: Error: Failure send version:{"type":"Buffer","data":[1,0,1,0,0,8]}
    at /app/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/ezsp.ts:510:23
    at Queue.executeNext (/app/node_modules/zigbee-herdsman/src/utils/queue.ts:32:32)

kaciker, 2022-10-23:
New attempt, working only 2h after taht this msg

**_Zigbee2MQTT:debug 2022-10-23 13:07:01: Saving state to file /config/zigbee2mqtt/state.json
Error: Failure send addTransientLinkKey:{"type":"Buffer","data":[0,0,175,255,255,255,255,255,255,255,255,90,105,103,66,101,101,65,108,108,105,97,110,99,101,48,57]}
    at /app/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/ezsp.ts:510:23
    at Queue.executeNext (/app/node_modules/zigbee-herdsman/src/utils/queue.ts:32:32)_**

chollier, 2022-10-25:
I am experiencing the same issue, please let us know if there is further debugging we can provide

kaciker, 2022-10-25:
> I am experiencing the same issue, please let us know if there is further debugging we can provide

In my lasts test, system was only few seconds working before go down.

In the meantime, that the issue is solved I switched to version no edge 1.27.2-1. No issues for me in this version

kirovilya, 2022-10-25:
@kaciker I apologize for the delay. but I need a herdsman log... did you run this command:
`DEBUG=zigbee-herdsman* npm start`
Your console should display a lot of information that needs to be written to a file.

kaciker, 2022-10-25:
> DEBUG=zigbee-herdsman* npm start

Don't worry, I understand.
I have HA, I don't know how to extract the full file log.

After run with the option in zigbee_herdsman_debug: true , zigbee start, and ends in 2 minutes. This is the end of the log
Something about the driver, right ?

Zigbee2MQTT:info  2022-10-25 20:41:06: MQTT publish: topic 'zigbee2mqtt/Nevera Int ', payload '{"battery":57,"humidity":62.07,"linkquality":160,"power_outage_count":38857,"pressure":947.2,"temperature":-6.37,"voltage":2935}'
Zigbee2MQTT:debug 2022-10-25 20:41:06: Received Zigbee message from 'Nevera Int ', type 'attributeReport', cluster 'msRelativeHumidity', data '{"measuredValue":6789}' from endpoint 1 with groupID 0
Zigbee2MQTT:info  2022-10-25 20:41:06: MQTT publish: topic 'zigbee2mqtt/Nevera Int ', payload '{"battery":57,"humidity":67.89,"linkquality":160,"power_outage_count":38857,"pressure":947.2,"temperature":-6.37,"voltage":2935}'
Zigbee2MQTT:debug 2022-10-25 20:41:06: Received Zigbee message from 'Nevera Int ', type 'attributeReport', cluster 'msPressureMeasurement', data '{"measuredValue":947,"scale":-1,"scaledValue":9472}' from endpoint 1 with groupID 0
Zigbee2MQTT:info  2022-10-25 20:41:06: MQTT publish: topic 'zigbee2mqtt/Nevera Int ', payload '{"battery":57,"humidity":67.89,"linkquality":160,"power_outage_count":38857,"pressure":947.2,"temperature":-6.37,"voltage":2935}'
Zigbee2MQTT:debug 2022-10-25 20:41:09: Received Zigbee message from 'Xiaomi Presence', type 'attributeReport', cluster 'msOccupancySensing', data '{"occupancy":1}' from endpoint 1 with groupID 0
Zigbee2MQTT:info  2022-10-25 20:41:09: MQTT publish: topic 'zigbee2mqtt/Xiaomi Presence', payload '{"battery":97,"linkquality":144,"occupancy":true,"power_outage_count":297,"voltage":2995}'
Zigbee2MQTT:debug 2022-10-25 20:41:15: Received Zigbee message from 'Xiaomi Presence', type 'attributeReport', cluster 'msOccupancySensing', data '{"occupancy":1}' from endpoint 1 with groupID 0
Zigbee2MQTT:info  2022-10-25 20:41:15: MQTT publish: topic 'zigbee2mqtt/Xiaomi Presence', payload '{"battery":97,"linkquality":144,"occupancy":true,"power_outage_count":297,"voltage":2995}'
Zigbee2MQTT:debug 2022-10-25 20:41:17: Received Zigbee message from 'Nevera Int ', type 'attributeReport', cluster 'msRelativeHumidity', data '{"measuredValue":7270}' from endpoint 1 with groupID 0
Zigbee2MQTT:info  2022-10-25 20:41:17: MQTT publish: topic 'zigbee2mqtt/Nevera Int ', payload '{"battery":57,"humidity":72.7,"linkquality":164,"power_outage_count":38857,"pressure":947.2,"temperature":-6.37,"voltage":2935}'
Zigbee2MQTT:debug 2022-10-25 20:41:17: Received Zigbee message from 'Nevera Int ', type 'attributeReport', cluster 'msPressureMeasurement', data '{"measuredValue":947,"scale":-1,"scaledValue":9472}' from endpoint 1 with groupID 0
Zigbee2MQTT:info  2022-10-25 20:41:17: MQTT publish: topic 'zigbee2mqtt/Nevera Int ', payload '{"battery":57,"humidity":72.7,"linkquality":160,"power_outage_count":38857,"pressure":947.2,"temperature":-6.37,"voltage":2935}'
**_Error: Unrecognized nodeId:38169
    at Driver.networkIdToEUI64 (/app/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/driver.ts:590:19)
    at processTicksAndRejections (node:internal/process/task_queues:96:5)
    at Driver.handleRouteError (/app/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/driver.ts:402:22)_**

kaciker, 2022-11-02:
@kirovilya did you find the time to check this log? What dou you think about it ?

kirovilya, 2022-11-02:
@kaciker it is not a "herdsman log". it should have a lot of lowlevel data (uart packets). it is issued by the command that I wrote above.

Vladimir1604, 2022-11-02:
hi!
I ran into the same problem
Here are the logs:

Zigbee2MQTT:error 2022-11-02 14:54:23: Publish 'set' 'state' to 'Свет в коридоре' failed: 'Error: Command 0xa4c138d6d47bbff6/2 genOnOff.on({}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (sendZclFrameToEndpointInternal error)'
Zigbee2MQTT:debug 2022-11-02 14:54:23: Error: Command 0xa4c138d6d47bbff6/2 genOnOff.on({}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (sendZclFrameToEndpointInternal error)
    at EZSPAdapter.sendZclFrameToEndpointInternal (/app/node_modules/zigbee-herdsman/src/adapter/ezsp/adapter/ezspAdapter.ts:465:19)
    at Queue.executeNext (/app/node_modules/zigbee-herdsman/src/utils/queue.ts:32:32)
Zigbee2MQTT:info  2022-11-02 14:54:23: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Publish 'set' 'state' to 'Свет в коридоре' failed: 'Error: Command 0xa4c138d6d47bbff6/2 genOnOff.on({}, {\"sendWhen\":\"immediate\",\"timeout\":10000,\"disableResponse\":false,\"disableRecovery\":false,\"disableDefaultResponse\":false,\"direction\":0,\"srcEndpoint\":null,\"reservedBits\":0,\"manufacturerCode\":null,\"transactionSequenceNumber\":null,\"writeUndiv\":false}) failed (sendZclFrameToEndpointInternal error)'","meta":{"friendly_name":"Свет в коридоре"},"type":"zigbee_publish_error"}'
Zigbee2MQTT:debug 2022-11-02 14:54:30: Received MQTT message on 'zigbee2mqtt/bridge/request/networkmap' with data '{"routes":false,"transaction":"y87r4-1","type":"raw"}'
Zigbee2MQTT:info  2022-11-02 14:54:30: Starting network scan (includeRoutes 'false')
Zigbee2MQTT:error 2022-11-02 14:54:40: Failed to execute LQI for 'Coordinator'
Zigbee2MQTT:debug 2022-11-02 14:54:40: Error: ZdoRequest error
    at Driver.zdoRequest (/app/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/driver.ts:575:19)
    at request (/app/node_modules/zigbee-herdsman/src/adapter/ezsp/adapter/ezspAdapter.ts:275:32)
    at Object.func (/app/node_modules/zigbee-herdsman/src/adapter/ezsp/adapter/ezspAdapter.ts:300:28)
    at Queue.executeNext (/app/node_modules/zigbee-herdsman/src/utils/queue.ts:32:32)
Zigbee2MQTT:error 2022-11-02 14:54:50: Failed to execute LQI for 'Розетки в детской'
Zigbee2MQTT:debug 2022-11-02 14:54:50: Error: ZdoRequest error
    at Driver.zdoRequest (/app/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/driver.ts:575:19)
    at request (/app/node_modules/zigbee-herdsman/src/adapter/ezsp/adapter/ezspAdapter.ts:275:32)
    at Object.func (/app/node_modules/zigbee-herdsman/src/adapter/ezsp/adapter/ezspAdapter.ts:300:28)
    at Queue.executeNext (/app/node_modules/zigbee-herdsman/src/utils/queue.ts:32:32)
Zigbee2MQTT:info  2022-11-02 14:54:50: Network scan finished
Zigbee2MQTT:info  2022-11-02 14:54:50: MQTT publish: topic 'zigbee2mqtt/bridge/response/networkmap', payload '{"data":{"routes":false,"type":"raw","value":{"links":[],"nodes":[{"definition":null,"failed":["lqi"],"friendlyName":"Coordinator","ieeeAddr":"0x2c1165fffee5460e","lastSeen":1667381760530,"networkAddress":0,"type":"Coordinator"},{"definition":{"description":"Wall switch module","model":"WHD02","supports":"switch (state), power_on_behavior, switch_type, linkquality","vendor":"TuYa"},"failed":["lqi"],"friendlyName":"Розетки в детской","ieeeAddr":"0xa4c13856867a7307","lastSeen":1667386483307,"manufacturerName":"_TZ3000_mx3vgyea","modelID":"TS000F","networkAddress":58099,"type":"Router"},{"definition":{"description":"Door sensor","model":"TS0203","supports":"contact, battery_low, tamper, battery, voltage, linkquality","vendor":"TuYa"},"friendlyName":"Входная дверь","ieeeAddr":"0xa4c138ef068da632","lastSeen":1667331038360,"manufacturerName":"_TZ3000_oxslv1c9","modelID":"TS0203","networkAddress":16390,"type":"EndDevice"},{"definition":{"description":"MiJia door & window contact sensor","model":"MCCGQ01LM","supports":"battery, contact, voltage, power_outage_count, linkquality","vendor":"Xiaomi"},"friendlyName":"Дверь гардероба","ieeeAddr":"0x00158d000423b318","lastSeen":1667333561144,"manufacturerName":"LUMI","modelID":"lumi.sensor_magnet","networkAddress":43435,"type":"EndDevice"},{"definition":{"description":"2 gang switch module - (without neutral)","model":"TS0012_switch_module","supports":"switch (state), power_on_behavior, switch_type, linkquality","vendor":"TuYa"},"friendlyName":"Свет в коридоре","ieeeAddr":"0xa4c138d6d47bbff6","lastSeen":1667388429300,"manufacturerName":"_TZ3000_jl7qyupf","modelID":"TS0012","networkAddress":5919,"type":"EndDevice"},{"definition":{"description":"Water leak detector","model":"TS0207_water_leak_detector","supports":"water_leak, battery_low, battery, linkquality","vendor":"TuYa"},"friendlyName":"Датчик протечки в ванной","ieeeAddr":"0xa4c138646b0ed3ed","lastSeen":1667325282523,"manufacturerName":"_TZ3000_kyb656no","modelID":"TS0207","networkAddress":55995,"type":"EndDevice"},{"definition":{"description":"Water leak detector","model":"TS0207_water_leak_detector","supports":"water_leak, battery_low, battery, linkquality","vendor":"TuYa"},"friendlyName":"Датчик протечки на кухне","ieeeAddr":"0xa4c138cf0b034f0f","lastSeen":1667323168662,"manufac [... cut ...] isableRecovery\":false,\"disableDefaultResponse\":false,\"direction\":0,\"srcEndpoint\":null,\"reservedBits\":0,\"manufacturerCode\":null,\"transactionSequenceNumber\":null,\"writeUndiv\":false}) failed (sendZclFrameToEndpointInternal error)'","meta":{"friendly_name":"Свет в коридоре"},"type":"zigbee_publish_error"}'
Zigbee2MQTT:debug 2022-11-02 15:42:35: Saving state to file /config/zigbee2mqtt/state.json
Zigbee2MQTT:debug 2022-11-02 15:47:35: Saving state to file /config/zigbee2mqtt/state.json

kaciker, 2022-11-02:
Hello, thanks @kirovilya I use HA Suppervised, and the information is not clear in the descriptiopn for how to activate this log, I think that hass been not

Just I see that in configuration tab there switch that said "sow not used configuration options " if you active that, in the top of the configutration appears a switch button with "zigbee_herdsman_debug" that you can activate. 
Now I attach the full log

[log (1).txt](https://github.com/Koenkk/zigbee2mqtt/files/9920947/log.1.txt)

kirovilya, 2022-11-02:
@kaciker this is again not a herdsman log. it should look different, like this:

```
kirov@kirov-home:~/Projects/zigbee2mqtt$ DEBUG=zigbee* npm start

> zigbee2mqtt@1.28.1-dev start /home/kirov/Projects/zigbee2mqtt
> node index.js

		Zigbee2MQTT requires node version ^14 || ^16 || ^18, you are running v12.22.12!

Zigbee2MQTT:debug 2022-11-02 13:43:09: Loaded state from file /home/kirov/Projects/zigbee2mqtt/data/state.json
Zigbee2MQTT:info  2022-11-02 13:43:09: Logging to console and directory: '/home/kirov/Projects/zigbee2mqtt/data/log/2022-11-02.13-43-09' filename: log.txt
Zigbee2MQTT:debug 2022-11-02 13:43:09: Removing old log directory '/home/kirov/Projects/zigbee2mqtt/data/log/2022-10-29.13-50-52'
Zigbee2MQTT:info  2022-11-02 13:43:09: Starting Zigbee2MQTT version 1.28.1-dev (commit #c3f6e793)
Zigbee2MQTT:info  2022-11-02 13:43:09: Starting zigbee-herdsman (0.14.68)
Zigbee2MQTT:debug 2022-11-02 13:43:09: Using zigbee-herdsman with settings: '{"adapter":{"concurrent":null,"delay":null,"disableLED":false},"backupPath":"/home/kirov/Projects/zigbee2mqtt/data/coordinator_backup.json","databaseBackupPath":"/home/kirov/Projects/zigbee2mqtt/data/database.db.backup","databasePath":"/home/kirov/Projects/zigbee2mqtt/data/database.db","network":{"channelList":[11],"extendedPanID":[221,221,221,221,221,221,221,221],"networkKey":"HIDDEN","panID":6756},"serialPort":{"adapter":"ezsp","path":"tcp://192.168.1.35:8888"}}'
  zigbee-herdsman:controller:log Starting with options '{"network":{"networkKeyDistribute":false,"networkKey":[1,3,5,7,9,11,13,15,0,2,4,6,8,10,12,13],"panID":6756,"extendedPanID":[221,221,221,221,221,221,221,221],"channelList":[11]},"serialPort":{"path":"tcp://192.168.1.35:8888","adapter":"ezsp"},"databasePath":"/home/kirov/Projects/zigbee2mqtt/data/database.db","databaseBackupPath":"/home/kirov/Projects/zigbee2mqtt/data/database.db.backup","backupPath":"/home/kirov/Projects/zigbee2mqtt/data/coordinator_backup.json","adapter":{"disableLED":false,"concurrent":null,"delay":null}}' +0ms
  zigbee-herdsman:adapter:ezsp:uart Opening TCP socket with 192.168.1.35:8888 +0ms
  zigbee-herdsman:adapter:ezsp:uart Socket connected +7ms
  zigbee-herdsman:adapter:ezsp:uart Socket ready +0ms
  zigbee-herdsman:adapter:ezsp:uart Uart reseting +0ms
  zigbee-herdsman:adapter:ezsp:uart --> Write reset +1ms
  zigbee-herdsman:adapter:ezsp:uart --> [1ac038bc7e] +0ms
  zigbee-herdsman:adapter:ezsp:uart <-- [c1020b0a527e] +0ms
  zigbee-herdsman:adapter:ezsp:uart <-- RSTACK: c1020b0a527e +1s
  zigbee-herdsman:adapter:ezsp:uart RSTACK Version: 2 Reason: RESET_SOFTWARE frame: c1020b0a527e +0ms
  zigbee-herdsman:adapter:ezsp:ezsp ==> version: {"desiredProtocolVersion":4} +0ms
  zigbee-herdsman:adapter:ezsp:ezsp ==> {"_cls_":"version","_id_":0,"_isRequest_":true,"desiredProtocolVersion":4} +1ms
  zigbee-herdsman:adapter:ezsp:uart --> DATA (0,0,0): 00000004 +2ms
  zigbee-herdsman:adapter:ezsp:uart --> [004221a850ed2c7e] +1s
  zigbee-herdsman:adapter:ezsp:uart -?- waiting (1) +1ms
  zigbee-herdsman:adapter:ezsp:uart <-- [0142a1a85c2895d52e8b7e] +54ms
  zigbee-herdsman:adapter:ezsp:uart <-- DATA (0,1,0): 0142a1a85c2895d52e8b7e +50ms
  zigbee-herdsman:adapter:ezsp:uart --> ACK  (1) +1ms
  zigbee-herdsman:adapter:ezsp:uart --> [8160597e] +51ms
  zigbee-herdsman:adapter:ezsp:ezsp <== Frame: 00800008028067 +53ms
  zigbee-herdsman:adapter:ezsp:ezsp <== 0x0: {"_id_":0,"_cls_":"version","_isRequest_":false,"protocolVersion":8,"stackType":2,"stackVersion":26496} +0ms
  zigbee-herdsman:adapter:ezsp:uart -+- waiting (1) success +1ms
  zigbee-herdsman:adapter:ezsp:ezsp Switching to eszp version 8 +0ms
  zigbee-herdsman:adapter:ezsp:ezsp ==> version: {"desiredProtocolVersion":8} +1ms
  zigbee-herdsman:adapter:ezsp:ezsp ==> {"_cls_":"version","_id_":0,"_isRequest_":true,"desiredProtocolVersion":8} +0ms
  zigbee-herdsman:adapter:ezsp:uart --> DATA (1,1,0): 010001000008 +1ms
  zigbee-herdsman:adapter:ezsp:uart --> [7d314321a9542a1dc97f7e] +2ms
  zigbee-herdsman:adapter:ezsp:uart -?- waiting (2) +0ms
  zigbee-herdsman:adapter:ezsp:uart <-- [1243a1a9542a1db0d9f362737e] +53ms
  zigbee-herdsman:adapter:ezsp:uart <-- DATA (1,2,0): 1243a1a9542a1db0d9f362737e +50ms
  zigbee-herdsman:adapter:ezsp:uart --> ACK  (2) +0ms
  zigbee-herdsman:adapter:ezsp:uart --> [82503a7e] +50ms
  zigbee-herdsman:adapter:ezsp:ezsp <== Frame: 018001000008028067 +50ms
  zigbee-herdsman:adapter:ezsp:ezsp <== 0x0: {"_id_":0,"_cls_":"version","_isRequest_":false,"protocolVersion":8,"stackType":2,"stackVersion":26496} +1ms
  zigbee-herdsman:adapter:ezsp:uart -+- waiting (2) success +1ms
  zigbee-herdsman:adapter:ezsp:ezsp Set EzspConfigId.CONFIG_SECURITY_LEVEL = 5 +1ms
  zigbee-herdsman:adapter:ezsp:ezsp ==> setConfigurationValue: {"configId":13,"value":5} +0ms
  zigbee-herdsman:adapter:ezsp:ezsp ==> {"_cls_":"setConfigurationValue","_id_":83,"_isRequest_":true,"configId":13,"value":5} +0ms
  zigbee-herdsman:adapter:ezsp:uart --> DATA (2,2,0): 02000153000d0500 +1ms
  zigbee-herdsman:adapter:ezsp:uart --> [224021a9072a7d38b7596ed17e] +3ms
  zigbee-herdsman:adapter:ezsp:uart -?- waiting (3) +1ms
  zigbee-herdsman:adapter:ezsp:uart <-- [2340a1a9072a159ab77e] +54ms
  zigbee-herdsman:adapter:ezsp:uart <-- DATA (2,3,0): 2340a1a9072a159ab77e +51ms
  zigbee-herdsman:adapter:ezsp:uart --> ACK  (3) +0ms
  zigbee-herdsman:adapter:ezsp:uart --> [83401b7e] +51ms
  zigbee-herdsman:adapter:ezsp:ezsp <== Frame: 028001530000 +52ms
[... cut ...]
  zigbee-herdsman:adapter:ezsp:uart --> [8610be7e] +53ms
  zigbee-herdsman:adapter:ezsp:ezsp <== Frame: 158001aa000007750106070800aa +53ms
  zigbee-herdsman:adapter:ezsp:ezsp <== 0xaa: {"_id_":170,"_cls_":"getValue","_isRequest_":false,"status":0,"value":{"type":"Buffer","data":[117,1,6,7,8,0,170]}} +1ms
  zigbee-herdsman:adapter:ezsp:uart -+- waiting (6) success +1ms
  zigbee-herdsman:adapter:ezsp:ezsp Got EzspValueId.VALUE_VERSION_INFO = � +0ms
  zigbee-herdsman:adapter:ezsp:driv EmberZNet version: 6.7.8.0 build 373 +55ms
```

kirovilya, 2022-11-02:
@Vladimir1604 I do not see what led to errors in your log, maybe a problem with connecting the stick? are you using a usb extension cable?
please record [herdsman log](https://www.zigbee2mqtt.io/guide/usage/debug.html#zigbee-herdsman-debug-logging) too.

kaciker, 2022-11-02:
Hello, @kirovilya HA don't understand this code " DEBUG=zigbee* npm start", I need to follow the process, descrived in my previous mail for HA.  the log is visible in the tab log, but is not wrote in the log folder. So I on ly can copy the buffer in screen, like that, but the more apperas in the botton, the more is deleted in the top..

[image]

This is the curent buffer htat I can copy

Zigbee2MQTT:debug 2022-11-02 17:32:29: Received MQTT message on 'homeassistant/sensor/Dell-Latitude7290/Dell-Latitude7290_almacenamiento_total_disk_count/config' with data '{"availability_topic":"homeassistant/sensor/Dell-Latitude7290/availability","icon":"mdi:harddisk","unique_id":"Dell-Latitude7290_almacenamiento_total_disk_count","device":{"identifiers":"hass.agent-Dell-Latitude7290","manufacturer":"LAB02 Research","model":"Microsoft Windows NT 10.0.22621.0","name":"Dell-Latitude7290","sw_version":"2022.13.0"},"name":"Dell-Latitude7290_almacenamiento Total Disk Count","state_topic":"homeassistant/sensor/Dell-Latitude7290/Dell-Latitude7290_almacenamiento/Dell-Latitude7290_almacenamiento_total_disk_count/state"}'
Zigbee2MQTT:debug 2022-11-02 17:32:29: Received MQTT message on 'homeassistant/sensor/Dell-Latitude7290/Dell-Latitude7290_almacenamiento/Dell-Latitude7290_almacenamiento_c/state' with data 'C'
Zigbee2MQTT:debug 2022-11-02 17:32:29: Received MQTT message on 'homeassistant/sensor/Dell-Latitude7290/Dell-Latitude7290_almacenamiento/Dell-Latitude7290_almacenamiento_c/attributes' with data '{
  "Name": "C",
  "Label": "-",
  "FileSystem": "NTFS",
  "TotalSizeMB": 511317.0,
  "AvailableSpaceMB": 390432.0,
  "UsedSpaceMB": 120885.0,
  "AvailableSpacePercentage": 76,
  "UsedSpacePercentage": 24
}'
Zigbee2MQTT:debug 2022-11-02 17:32:29: Received MQT
T message on 'homeassistant/sensor/Dell-Latitude7290/Dell-Latitude7290_almacenamiento/Dell-Latitude7290_almacenamiento_total_disk_count/state' with data '1'
Zigbee2MQTT:debug 2022-11-02 17:32:30: Received MQTT message on 'homeassistant/sensor/Dell-Latitude7290/Dell-Latitude7290_usodememoria/state' with data '64.48'
Zigbee2MQTT:debug 2022-11-02 17:32:38: Received MQTT message on 'homeassistant/sensor/Dell-Latitude7290/Dell-Latitude7290_cargadecpu/state' with data '11'
2022-11-02T16:32:47.921Z zigbee-herdsman:adapter:ezsp:uart <-- [057d5eb1a9]
2022-11-02T16:32:47.921Z zigbee-herdsman:adapter:ezsp:uart <-- [7d312a15b658964e24ab5593499c72af15]
2022-11-02T16:32:47.922Z zigbee-herdsman:adapter:ezsp:uart <-- [f45b9874f5]
2022-11-02T16:32:47.924Z zigbee-herdsman:adapter:ezsp:uart <-- [de6483fc]
2022-11-02T16:32:47.924Z zigbee-herdsman:adapter:ezsp:uart <-- [7d5e16a8e9e90c]
2022-11-02T16:32:47.925Z zigbee-herdsman:adapter:ezsp:uart <-- [7e]
2022-11-02T16:32:47.925Z zigbee-herdsman:adapter:ezsp:uart <-- DATA (0,5,0): 057eb1a9112a15b658964e24ab5593499c72af15f45b9874f5de6483fc7e16a8e9e90c7e000000
2022-11-02T16:32:47.925Z zigbee-herdsman:adapter:ezsp:uart --> ACK  (1)
2022-11-02T16:32:47.925Z zigbee-herdsman:adapter:ezsp:uart --> [8160597e]
2022-11-02T16:32:47.925Z zigbee-herdsman:adapter:ezsp:ezsp <== Frame: 3c9001450000040102040101000100003c88be1995ffff0818070a0000290f0224d211
2022-11-02T16:32:47.926Z zigbee-herdsman:adapter:ezsp:ezsp <== 0x45: {"_id_":69,"_cls_":"incomingMessageHandler","_isRequest_":false,"type":0,"apsFrame":{"profileId":260,"clusterId":1026,"sourceEndpoint":1,"destinationEndpoint":1,"options":256,"groupId":0,"sequence":60},"lastHopLqi":136,"lastHopRssi":-66,"sender":38169,"bindingIndex":255,"addressIndex":255,"message":{"type":"Buffer","data":[24,7,10,0,0,41,15,2]}}
2022-11-02T16:32:47.926Z zigbee-herdsman:adapter:ezsp:debg processMessage: {"messageType":0,"apsFrame":{"profileId":260,"clusterId":1026,"sourceEndpoint":1,"destinationEndpoint":1,"options":256,"groupId":0,"sequence":60},"lqi":136,"rssi":-66,"sender":38169,"bindingIndex":255,"addressIndex":255,"message":{"type":"Buffer","data":[24,7,10,0,0,41,15,2]}}
2022-11-02T16:32:47.926Z zigbee-herdsman:controller:log Received 'zcl' data '{"frame":{"Header":{"frameControl":{"frameType":0,"manufacturerSpecific":false,"direction":1,"disableDefaultResponse":true,"reservedBits":0},"transactionSequenceNumber":7,"manufacturerCode":null,"commandIdentifier":10},"Payload":[{"attrId":0,"dataType":41,"attrData":527}],"Command":{"ID":10,"name":"report","parameters":[{"name":"attrId","type":33},{"name":"dataType","type":32},{"name":"attrData","type":1000}]}},"address":38169,"endpoint":1,"linkquality":136,"groupID":0,"wasBroadcast":false,"destinationEndpoint":1}'
Zigbee2MQTT:debug 2022-11-02 17:32:47: Received Zigbee message from 'Nevera Int ', type 'attributeReport', cluster 'msTemperatureMeasurement', data '{"measuredValue":527}' from endpoint 1 with groupID 0
Zigbee2MQTT:info  2022-11-02 17:32:47: MQTT publish: topic 'zigbee2mqtt/Nevera Int ', payload '{"battery":63,"humidity":73.03,"linkquality":136,"power_outage_count":38857,"pressure":950.2,"temperature":5.27,"voltage":2945}'
2022-11-02T16:32:47.931Z zigbee-herdsman:adapter:ezsp:uart <-- [157d5eb1a97d312a15b658914e24ab5593499c73af15f45b98]
2022-11-02T16:32:47.935Z zigbee-herdsman:adapter:ezsp:uart <-- [74f5de6b83fc]
Zigbee2MQTT:debug 2022-11-02 17:32:47: Received Zigbee message from 'Nevera Int ', type 'attributeReport', cluster 'msRelativeHumidity', data '{"measuredValue":6682}' from endpoint 1 with groupID 0
[... cut ...]
2022-11-02T16:32:54.010Z zigbee-herdsman:adapter:ezsp:uart --> [87009f7e]
2022-11-02T16:32:54.011Z zigbee-herdsman:adapter:ezsp:ezsp <== Frame: 3e90013f0000a76d04010600010840010000e90c0000
2022-11-02T16:32:54.011Z zigbee-herdsman:adapter:ezsp:ezsp <== 0x3f: {"_id_":63,"_cls_":"messageSentHandler","_isRequest_":false,"type":0,"indexOrDestination":28071,"apsFrame":{"profileId":260,"clusterId":6,"sourceEndpoint":1,"destinationEndpoint":8,"options":320,"groupId":0,"sequence":233},"messageTag":12,"status":0,"message":{"type":"Buffer","data":[]}}

kirovilya, 2022-11-02:
@kaciker now the correct log, but this is not enough, you need to record a similar log before the errors occur in order to see their cause.
I don't know how to do it in your case

kaciker, 2022-11-02:
> @kaciker now the correct log, but this is not enough, you need to record a similar log before the errors occur in order to see their cause. I don't know how to do it in your case

Now system down, again, and this is tha last log 

2022-11-02T17:31:28.125Z zigbee-herdsman:adapter:ezsp:uart -+- waiting (6) success
2022-11-02T17:31:28.136Z zigbee-herdsman:adapter:ezsp:uart <-- [366fb1a97d312a15b25990ca25aa]
2022-11-02T17:31:28.140Z zigbee-herdsman:adapter:ezsp:uart <-- [1593499c21d8abedce9874bec26389fc40]
2022-11-02T17:31:28.143Z zigbee-herdsman:adapter:ezsp:uart <-- [3ea3ea22]
2022-11-02T17:31:28.144Z zigbee-herdsman:adapter:ezsp:uart <-- [606f88ffc7d8d5d46986463aa9f6763ba6fe75824298482616b1e670301c2e07bbe6ca6188419f49984b9ff2c2d2d76131a9539748347d33f5bea2e9fd7e466fb1a96b2a15b259944a21aa5592099d4e27c5e8ce67a7317e]
2022-11-02T17:31:28.144Z zigbee-herdsman:adapter:ezsp:uart <-- DATA (3,6,0): 366fb1a9112a15b25990ca25aa1593499c21d8abedce9874bec26389fc403ea3ea22606f88ffc7d8d5d46986463aa9f6763ba6fe75824298482616b1e670301c2e07bbe6ca6188419f49984b9ff2c2d2d76131a95397483413f5bea2e9fd7e0000
2022-11-02T17:31:28.144Z zigbee-herdsman:adapter:ezsp:uart --> ACK  (4)
2022-11-02T17:31:28.144Z zigbee-herdsman:adapter:ezsp:uart --> [8430fc7e]
2022-11-02T17:31:28.144Z zigbee-herdsman:adapter:ezsp:ezsp <== Frame: 2d9001450000000004800000400100006fff000000ffff43040000003e010401efbe00070000030006000a0019001a000003140000030004000500060008002000000300040204050406040005010b030b040b0207001001fc02fcc652
2022-11-02T17:31:28.144Z zigbee-herdsman:adapter:ezsp:ezsp <== 0x45: {"_id_":69,"_cls_":"incomingMessageHandler","_isRequest_":false,"type":0,"apsFrame":{"profileId":0,"clusterId":32772,"sourceEndpoint":0,"destinationEndpoint":0,"options":320,"groupId":0,"sequence":111},"lastHopLqi":255,"lastHopRssi":0,"sender":0,"bindingIndex":255,"addressIndex":255,"message":{"type":"Buffer","data":[4,0,0,0,62,1,4,1,239,190,0,7,0,0,3,0,6,0,10,0,25,0,26,0,0,3,20,0,0,3,0,4,0,5,0,6,0,8,0,32,0,0,3,0,4,2,4,5,4,6,4,0,5,1,11,3,11,4,11,2,7,0,16,1,252,2,252]}}
2022-11-02T17:31:28.144Z zigbee-herdsman:adapter:ezsp:uart <-- DATA (4,6,0): 466fb1a96b2a15b259944a21aa5592099d4e27c5e8ce67a7317e
2022-11-02T17:31:28.144Z zigbee-herdsman:adapter:ezsp:uart --> ACK  (5)
2022-11-02T17:31:28.144Z zigbee-herdsman:adapter:ezsp:uart --> [8520dd7e]
2022-11-02T17:31:28.144Z zigbee-herdsman:adapter:ezsp:ezsp <== Frame: 2d90013f00000000000004000000400100006e050000
2022-11-02T17:31:28.145Z zigbee-herdsman:adapter:ezsp:ezsp <== 0x3f: {"_id_":63,"_cls_":"messageSentHandler","_isRequest_":false,"type":0,"indexOrDestination":0,"apsFrame":{"profileId":0,"clusterId":4,"sourceEndpoint":0,"destinationEndpoint":0,"options":320,"groupId":0,"sequence":110},"messageTag":5,"status":0,"message":{"type":"Buffer","data":[]}}
2022-11-02T17:31:28.145Z zigbee-herdsman:adapter:ezsp:driv Simple_Desc_rsp  frame: {"type":"Buffer","data":[4,0,0,0,62,1,4,1,239,190,0,7,0,0,3,0,6,0,10,0,25,0,26,0,0,3,20,0,0,3,0,4,0,5,0,6,0,8,0,32,0,0,3,0,4,2,4,5,4,6,4,0,5,1,11,3,11,4,11,2,7,0,16,1,252,2,252]}
2022-11-02T17:31:28.145Z zigbee-herdsman:adapter:ezsp:driv Simple_Desc_rsp parsed: {"_id_":32772,"_cls_":"simpleDescRsp","transId":4,"status":0,"nwkaddr":0,"len":62,"descriptor":{"endpoint":1,"profileid":260,"deviceid":48879,"deviceversion":0,"inclusterlist":[0,3,6,10,25,26,768],"outclusterlist":[0,3,4,5,6,8,32,768,1024,1026,1029,1030,1280,2817,2819,2820,1794,4096,64513,64514]}}
2022-11-02T17:31:28.145Z zigbee-herdsman:adapter:ezsp:driv ZDO Simple_Desc_req params: {"dstaddr":0,"targetEp":242}
2022-11-02T17:31:28.145Z zigbee-herdsman:adapter:ezsp:ezsp ==> setExtendedTimeout: {"remoteEui64":{"_value":{"type":"Buffer","data":[128,75,80,255,254,167,226,255]}},"extendedTimeout":true}
2022-11-02T17:31:28.146Z zigbee-herdsman:adapter:ezsp:ezsp ==> {"_cls_":"setExtendedTimeout","_id_":126,"_isRequest_":true,"remoteEui64":{"_value":{"type":"Buffer","data":[128,75,80,255,254,167,226,255]}},"extendedTimeout":true}
2022-11-02T17:31:28.146Z zigbee-herdsman:adapter:ezsp:uart --> DATA (6,5,0): 2e00017e00ffe2a7feff504b8001
2022-11-02T17:31:28.146Z zigbee-herdsman:adapter:ezsp:uart --> [656c21a92a2aea50fe6ab575e1d59347437e]
2022-11-02T17:31:28.146Z zigbee-herdsman:adapter:ezsp:uart -?- waiting (7)
2022-11-02T17:31:28.523Z zigbee-herdsman:adapter:ezsp:uart <-- [2e6fa1a9602a15dc25867e3e]
2022-11-02T17:31:28.523Z zigbee-herdsman:adapter:ezsp:uart <-- DATA (2,6,1): 2e6fa1a9602a15dc25867e
2022-11-02T17:31:28.523Z zigbee-herdsman:adapter:ezsp:uart --> ACK  (3)
2022-11-02T17:31:28.523Z zigbee-herdsman:adapter:ezsp:uart --> [83401b7e]
2022-11-02T17:31:28.524Z zigbee-herdsman:adapter:ezsp:uart Unexpected packet sequence 6 | 7
2022-11-02T17:31:28.524Z zigbee-herdsman:adapter:ezsp:ezsp <== Frame: 2d80013400006e
2022-11-02T17:31:28.524Z zigbee-herdsman:adapter:ezsp:ezsp <== 0x34: {"_id_":52,"_cls_":"sendUnicast","_isRequest_":false,"status":0,"sequence":110}
2022-11-02T17:31:28.524Z zigbee-herdsman:adapter:ezsp:driv Unhandled frame sendUnicast
2022-11-02T17:31:28.527Z zigbee-herdsman:adapter:ezsp:uart <-- [6fb1a97d312a15b25990ca25]
2022-11-02T17:31:28.529Z zigbee-herdsman:adapter:ezsp:uart <-- [aa1593]
2022-11-02T17:31:28.529Z zigbee-herdsman:adapter:ezsp:uart <-- [499c21d8abedce]
2022-11-02T17:31:28.532Z zigbee-herdsman:adapter:ezsp:uart <-- [9874be]
2022-11-02T17:31:28.535Z zigbee-herdsman:adapter:ezsp:uart <-- [c26389fc40]
2022-11-02T17:31:28.538Z zigbee-herdsman:adapter:ezsp:uart <-- [3ea3ea]
[... cut ...]
Zigbee2MQTT:error 2022-11-02 18:31:30: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start.html for possible solutions
Zigbee2MQTT:error 2022-11-02 18:31:30: Exiting...
Zigbee2MQTT:error 2022-11-02 18:31:30: Error: ZdoRequest error
    at Driver.zdoRequest (/app/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/driver.ts:575:19)
    at Object.func (/app/node_modules/zigbee-herdsman/src/adapter/ezsp/adapter/ezspAdapter.ts:211:36)
    at Queue.executeNext (/app/node_modules/zigbee-herdsman/src/utils/queue.ts:32:32)

kaciker, 2022-11-02:
I restart z2mqtt and system is up againn with this firs log

[19:04:04] INFO: Preparing to start...
[19:04:04] INFO: Socat not enabled
[19:04:05] INFO: Zigbee Herdsman debug logging enabled
[19:04:05] INFO: Starting Zigbee2MQTT...
Zigbee2MQTT:debug 2022-11-02 19:04:07: Loaded state from file /config/zigbee2mqtt/state.json
Zigbee2MQTT:info  2022-11-02 19:04:07: Logging to console and directory: '/config/zigbee2mqtt/log/2022-11-02.19-04-07' filename: log.txt
Zigbee2MQTT:debug 2022-11-02 19:04:07: Removing old log directory '/config/zigbee2mqtt/log/2022-11-02.18-01-48'
Zigbee2MQTT:info  2022-11-02 19:04:07: Starting Zigbee2MQTT version 1.28.1 (commit #unknown)
Zigbee2MQTT:info  2022-11-02 19:04:07: Starting zigbee-herdsman (0.14.68)
Zigbee2MQTT:debug 2022-11-02 19:04:07: Using zigbee-herdsman with settings: '{"adapter":{"concurrent":null,"delay":null,"disableLED":false},"backupPath":"/config/zigbee2mqtt/coordinator_backup.json","databaseBackupPath":"/config/zigbee2mqtt/database.db.backup","databasePath":"/config/zigbee2mqtt/database.db","network":{"channelList":[11],"extendedPanID":[221,221,221,221,221,221,221,221],"networkKey":"HIDDEN","panID":6754},"serialPort":{"adapter":"ezsp","path":"tcp://192.168.31.20:8888"}}'
2022-11-02T18:04:07.706Z zigbee-herdsman:controller:log Starting with options '{"network":{"networkKeyDistribute":false,"networkKey":[1,3,5,7,9,11,13,15,0,2,4,6,8,10,12,13],"panID":6754,"extendedPanID":[221,221,221,221,221,221,221,221],"channelList":[11]},"serialPort":{"path":"tcp://192.168.31.20:8888","adapter":"ezsp"},"databasePath":"/config/zigbee2mqtt/database.db","databaseBackupPath":"/config/zigbee2mqtt/database.db.backup","backupPath":"/config/zigbee2mqtt/coordinator_backup.json","adapter":{"disableLED":false,"concurrent":null,"delay":null}}'
2022-11-02T18:04:07.708Z zigbee-herdsman:adapter:ezsp:uart Opening TCP socket with 192.168.31.20:8888
2022-11-02T18:04:07.726Z zigbee-herdsman:adapter:ezsp:uart Socket connected
2022-11-02T18:04:07.727Z zigbee-herdsman:adapter:ezsp:uart Socket ready
2022-11-02T18:04:07.727Z zigbee-herdsman:adapter:ezsp:uart Uart reseting
2022-11-02T18:04:07.727Z zigbee-herdsman:adapter:ezsp:uart --> Write reset
2022-11-02T18:04:07.729Z zigbee-herdsman:adapter:ezsp:uart --> [1ac038bc7e]
2022-11-02T18:04:08.875Z zigbee-herdsman:adapter:ezsp:uart <-- [c1020b0a527e]
2022-11-02T18:04:08.876Z zigbee-herdsman:adapter:ezsp:uart <-- RSTACK: c1020b0a527e
2022-11-02T18:04:08.876Z zigbee-herdsman:adapter:ezsp:uart RSTACK Version: 2 Reason: RESET_SOFTWARE frame: c1020b0a527e
2022-11-02T18:04:08.877Z zigbee-herdsman:adapter:ezsp:ezsp ==> version: {"desiredProtocolVersion":4}
2022-11-02T18:04:08.878Z zigbee-herdsman:adapter:ezsp:ezsp ==> {"_cls_":"version","_id_":0,"_isRequest_":true,"desiredProtocolVersion":4}
2022-11-02T18:04:08.879Z zigbee-herdsman:adapter:ezsp:uart --> DATA (0,0,0): 00000004
2022-11-02T18:04:08.880Z zigbee-herdsman:adapter:ezsp:uart --> [004221a850ed2c7e]
2022-11-02T18:04:08.880Z zigbee-herdsman:adapter:ezsp:uart -?- waiting (1)
2022-11-02T18:04:08.913Z zigbee-herdsman:adapter:ezsp:uart <-- [0142a1a85c2895]
2022-11-02T18:04:08.915Z zigbee-herdsman:adapter:ezsp:uart <-- [d52e8b7e]
2022-11-02T18:04:08.915Z zigbee-herdsman:adapter:ezsp:uart <-- DATA (0,1,0): 0142a1a85c2895d52e8b7e
2022-11-02T18:04:08.915Z zigbee-herdsman:adapter:ezsp:uart --> ACK  (1)
2022-11-02T18:04:08.916Z zigbee-herdsman:adapter:ezsp:uart --> [8160597e]
2022-11-02T18:04:08.917Z zigbee-herdsman:adapter:ezsp:ezsp <== Frame: 00800008028067
2022-11-02T18:04:08.917Z zigbee-herdsman:adapter:ezsp:ezsp <== 0x0: {"_id_":0,"_cls_":"version","_isRequest_":false,"protocolVersion":8,"stackType":2,"stackVersion":26496}
2022-11-02T18:04:08.918Z zigbee-herdsman:adapter:ezsp:uart -+- waiting (1) success
2022-11-02T18:04:08.918Z zigbee-herdsman:adapter:ezsp:ezsp Switching to eszp version 8
2022-11-02T18:04:08.918Z zigbee-herdsman:adapter:ezsp:ezsp ==> version: {"desiredProtocolVersion":8}
2022-11-02T18:04:08.918Z zigbee-herdsman:adapter:ezsp:ezsp ==> {"_cls_":"version","_id_":0,"_isRequest_":true,"desiredProtocolVersion":8}
2022-11-02T18:04:08.919Z zigbee-herdsman:adapter:ezsp:uart --> DATA (1,1,0): 010001000008
2022-11-02T18:04:08.919Z zigbee-herdsman:adapter:ezsp:uart --> [7d314321a9542a1dc97f7e]
2022-11-02T18:04:08.919Z zigbee-herdsman:adapter:ezsp:uart -?- waiting (2)
2022-11-02T18:04:08.969Z zigbee-herdsman:adapter:ezsp:uart <-- [1243a1a9542a1db0d9f36273]
2022-11-02T18:04:08.969Z zigbee-herdsman:adapter:ezsp:uart <-- [7e]
2022-11-02T18:04:08.970Z zigbee-herdsman:adapter:ezsp:uart <-- DATA (1,2,0): 1243a1a9542a1db0d9f362737e
2022-11-02T18:04:08.970Z zigbee-herdsman:adapter:ezsp:uart --> ACK  (2)
2022-11-02T18:04:08.970Z zigbee-herdsman:adapter:ezsp:uart --> [82503a7e]
2022-11-02T18:04:08.970Z zigbee-herdsman:adapter:ezsp:ezsp <== Frame: 018001000008028067
2022-11-02T18:04:08.970Z zigbee-herdsman:adapter:ezsp:ezsp <== 0x0: {"_id_":0,"_cls_":"version","_isRequest_":false,"protocolVersion":8,"stackType":2,"stackVersion":26496}
2022-11-02T18:04:08.971Z zigbee-herdsman:adapter:ezsp:uart -+- waiting (2) success
2022-11-02T18:04:08.971Z zigbee-herdsman:adapter:ezsp:ezsp Set EzspConfigId.CONFIG_SECURITY_LEVEL = 5
2022-11-02T18:04:08.971Z zigbee-herdsman:adapter:ezsp:ezsp ==> setConfigurationValue: {"configId":13,"value":5}
2022-11-02T18:04:08.972Z zigbee-herdsman:adapter:ezsp:ezsp ==> {"_cls_":"setConfigurationValue","_id_":83,"_isRequest_":true,"configId":13,"value":5}
2022-11-02T18:04:08.972Z zigbee-herdsman:adapter:ezsp:uart --> DATA (2,2,0): 02000153000d0500
2022-11-02T18:04:08.972Z zigbee-herdsman:adapter:ezsp:uart --> [224021a9072a7d38b7596ed17e]
2022-11-02T18:04:08.972Z zigbee-herdsman:adapter:ezsp:uart -?- waiting (3)

kaciker, 2022-11-02:
And now, is stoped, with this static log:
If I touch start, it start to work again some time

9:21:33] INFO: Preparing to start...
[19:21:33] INFO: Socat not enabled
[19:21:33] INFO: Zigbee Herdsman debug logging enabled
[19:21:34] INFO: Starting Zigbee2MQTT...
Zigbee2MQTT:debug 2022-11-02 19:21:36: Loaded state from file /config/zigbee2mqtt/state.json
Zigbee2MQTT:info  2022-11-02 19:21:36: Logging to console and directory: '/config/zigbee2mqtt/log/2022-11-02.19-21-36' filename: log.txt
Zigbee2MQTT:debug 2022-11-02 19:21:36: Removing old log directory '/config/zigbee2mqtt/log/2022-11-02.18-22-40'
Zigbee2MQTT:info  2022-11-02 19:21:36: Starting Zigbee2MQTT version 1.28.1 (commit #unknown)
Zigbee2MQTT:info  2022-11-02 19:21:36: Starting zigbee-herdsman (0.14.68)
Zigbee2MQTT:debug 2022-11-02 19:21:36: Using zigbee-herdsman with settings: '{"adapter":{"concurrent":null,"delay":null,"disableLED":false},"backupPath":"/config/zigbee2mqtt/coordinator_backup.json","databaseBackupPath":"/config/zigbee2mqtt/database.db.backup","databasePath":"/config/zigbee2mqtt/database.db","network":{"channelList":[11],"extendedPanID":[221,221,221,221,221,221,221,221],"networkKey":"HIDDEN","panID":6754},"serialPort":{"adapter":"ezsp","path":"tcp://192.168.31.20:8888"}}'
2022-11-02T18:21:36.317Z zigbee-herdsman:controller:log Starting with options '{"network":{"networkKeyDistribute":false,"networkKey":[1,3,5,7,9,11,13,15,0,2,4,6,8,10,12,13],"panID":6754,"extendedPanID":[221,221,221,221,221,221,221,221],"channelList":[11]},"serialPort":{"path":"tcp://192.168.31.20:8888","adapter":"ezsp"},"databasePath":"/config/zigbee2mqtt/database.db","databaseBackupPath":"/config/zigbee2mqtt/database.db.backup","backupPath":"/config/zigbee2mqtt/coordinator_backup.json","adapter":{"disableLED":false,"concurrent":null,"delay":null}}'
2022-11-02T18:21:36.319Z zigbee-herdsman:adapter:ezsp:uart Opening TCP socket with 192.168.31.20:8888

Vladimir1604, 2022-11-02:
Is this herdsman debug logs?

[22:22:24] INFO: Preparing to start...
[22:22:24] INFO: Socat not enabled
[22:22:24] INFO: Zigbee Herdsman debug logging enabled
[22:22:25] INFO: Starting Zigbee2MQTT...
Zigbee2MQTT:debug 2022-11-02 22:22:27: Loaded state from file /config/zigbee2mqtt/state.json
Zigbee2MQTT:info  2022-11-02 22:22:27: Logging to console and directory: '/config/zigbee2mqtt/log/2022-11-02.22-22-27' filename: log.txt
Zigbee2MQTT:debug 2022-11-02 22:22:27: Removing old log directory '/config/zigbee2mqtt/log/2022-11-02.13-26-40'
Zigbee2MQTT:info  2022-11-02 22:22:27: Starting Zigbee2MQTT version 1.28.1 (commit #unknown)
Zigbee2MQTT:info  2022-11-02 22:22:27: Starting zigbee-herdsman (0.14.68)
Zigbee2MQTT:debug 2022-11-02 22:22:27: Using zigbee-herdsman with settings: '{"adapter":{"concurrent":null,"delay":null,"disableLED":false},"backupPath":"/config/zigbee2mqtt/coordinator_backup.json","databaseBackupPath":"/config/zigbee2mqtt/database.db.backup","databasePath":"/config/zigbee2mqtt/database.db","network":{"channelList":[11],"extendedPanID":[221,221,221,221,221,221,221,221],"networkKey":"HIDDEN","panID":6754},"serialPort":{"adapter":"ezsp","path":"tcp://192.168.101.2:8888"}}'
2022-11-02T19:22:27.373Z zigbee-herdsman:controller:log Starting with options '{"network":{"networkKeyDistribute":false,"networkKey":[1,3,5,7,9,11,13,15,0,2,4,6,8,10,12,13],"panID":6754,"extendedPanID":[221,221,221,221,221,221,221,221],"channelList":[11]},"serialPort":{"path":"tcp://192.168.101.2:8888","adapter":"ezsp"},"databasePath":"/config/zigbee2mqtt/database.db","databaseBackupPath":"/config/zigbee2mqtt/database.db.backup","backupPath":"/config/zigbee2mqtt/coordinator_backup.json","adapter":{"disableLED":false,"concurrent":null,"delay":null}}'
2022-11-02T19:22:27.375Z zigbee-herdsman:adapter:ezsp:uart Opening TCP socket with 192.168.101.2:8888
2022-11-02T19:22:27.383Z zigbee-herdsman:adapter:ezsp:uart Socket connected
2022-11-02T19:22:27.384Z zigbee-herdsman:adapter:ezsp:uart Socket ready
2022-11-02T19:22:27.385Z zigbee-herdsman:adapter:ezsp:uart Uart reseting
2022-11-02T19:22:27.385Z zigbee-herdsman:adapter:ezsp:uart --> Write reset
2022-11-02T19:22:27.386Z zigbee-herdsman:adapter:ezsp:uart --> [1ac038bc7e]
 
Zigbee2MQTT:info  2022-11-02 22:22:30: MQTT publish: topic 'homeassistant/binary_sensor/0xa4c138cf0b034f0f/water_leak/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state"}],"device":{"identifiers":["zigbee2mqtt_0xa4c138cf0b034f0f"],"manufacturer":"TuYa","model":"Water leak detector (TS0207_water_leak_detector)","name":"Датчик протечки на кухне"},"device_class":"moisture","json_attributes_topic":"zigbee2mqtt/Датчик протечки на кухне","name":"Датчик протечки на кухне water leak","payload_off":false,"payload_on":true,"state_topic":"zigbee2mqtt/Датчик протечки на кухне","unique_id":"0xa4c138cf0b034f0f_water_leak_zigbee2mqtt","value_template":"{{ value_json.water_leak }}"}'
Zigbee2MQTT:info  2022-11-02 22:22:30: MQTT publish: topic 'homeassistant/binary_sensor/0xa4c138cf0b034f0f/battery_low/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state"}],"device":{"identifiers":["zigbee2mqtt_0xa4c138cf0b034f0f"],"manufacturer":"TuYa","model":"Water leak detector (TS0207_water_leak_detector)","name":"Датчик протечки на кухне"},"device_class":"battery","entity_category":"diagnostic","json_attributes_topic":"zigbee2mqtt/Датчик протечки на кухне","name":"Датчик протечки на кухне battery low","payload_off":false,"payload_on":true,"state_topic":"zigbee2mqtt/Датчик протечки на кухне","unique_id":"0xa4c138cf0b034f0f_battery_low_zigbee2mqtt","value_template":"{{ value_json.battery_low }}"}'
Zigbee2MQTT:info  2022-11-02 22:22:30: MQTT publish: topic 'homeassistant/sensor/0xa4c138cf0b034f0f/battery/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state"}],"device":{"identifiers":["zigbee2mqtt_0xa4c138cf0b034f0f"],"manufacturer":"TuYa","model":"Water leak detector (TS0207_water_leak_detector)","name":"Датчик протечки на кухне"},"device_class":"battery","enabled_by_default":true,"entity_category":"diagnostic","json_attributes_topic":"zigbee2mqtt/Датчик протечки на кухне","name":"Датчик протечки на кухне battery","state_class":"measurement","state_topic":"zigbee2mqtt/Датчик протечки на кухне","unique_id":"0xa4c138cf0b034f0f_battery_zigbee2mqtt","unit_of_measurement":"%","value_template":"{{ value_json.battery }}"}'
Zigbee2MQTT:info  2022-11-02 22:22:30: MQTT publish: topic 'homeassistant/sensor/0xa4c138cf0b034f0f/linkquality/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state"}],"device":{"identifiers":["zigbee2mqtt_0xa4c138cf0b034f0f"],"manufacturer":"TuYa","model":"Water leak detector (TS0207_water_leak_detector)","name":"Датчик протечки на кухне"},"enabled_by_default":false,"entity_category":"diagnostic","icon":"mdi:signal","json_attributes_topic":"zigbee2mqtt/Датчик протечки на кухне","name":"Датчик протечки на кухне linkquality","state_class":"measurement","state_topic":"zigbee2mqtt/Датчик протечки на кухне","unique_id":"0xa4c138cf0b034f0f_linkquality_zigbee2mqtt","unit_of_measurement":"lqi","value_template":"{{ value_json.linkquality }}"}'
Zigbee2MQTT:info  2022-11-02 22:22:30: MQTT publish: topic 'homeassistant/sensor/0x00158d0007050cd2/battery/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state"}],"device":{"identifiers":["zigbee2mqtt_0x00158d0007050cd2"],"manufacturer":"Xiaomi","model":"MiJia door & window contact sensor (MCCGQ01LM)","name" [... cut ...]
2022-11-02T19:22:30.454Z zigbee-herdsman:adapter:ezsp:ezsp <== 0x64: {"_id_":100,"_cls_":"setMulticastTableEntry","_isRequest_":false,"status":0}
2022-11-02T19:22:30.454Z zigbee-herdsman:adapter:ezsp:uart -+- waiting (3) success
2022-11-02T19:22:30.455Z zigbee-herdsman:adapter:ezsp:cast Set MulticastTableEntry #6 for 65533 multicast id: 0
2022-11-02T19:23:32.030Z zigbee-herdsman:adapter:ezsp:uart Port closed
2022-11-02T19:23:32.031Z zigbee-herdsman:adapter:ezsp:ezsp Close ezsp
2022-11-02T19:23:32.031Z zigbee-herdsman:adapter:ezsp:driv Close driver

kirovilya, 2022-11-03:
@Vladimir1604 Use pastebin type services for large logs. Yes, now the correct log. But I don't see any errors in it.

kaciker, 2022-11-04:
Hello @kirovilya did you see something in my logs ? I see that today we have another update but only talk about Ikea

kirovilya, 2022-11-06:
@kaciker try the current dev/edge version of z2m

kaciker, 2022-11-06:
> @kaciker try the current dev/edge version of z2m

Hello.
Unfortunatelly orked 5' and down. After I activate the log, and only worked 10 seconds

2022-11-06T20:16:38.940Z zigbee-herdsman:adapter:ezsp:uart <-- [6461a1a9372a15b25994d2547e]
2022-11-06T20:16:38.940Z zigbee-herdsman:adapter:ezsp:uart <-- DATA (6,4,0): 6461a1a9372a15b25994d2547e
2022-11-06T20:16:38.940Z zigbee-herdsman:adapter:ezsp:uart --> ACK  (7)
2022-11-06T20:16:38.940Z zigbee-herdsman:adapter:ezsp:uart --> [87009f7e]
2022-11-06T20:16:38.940Z zigbee-herdsman:adapter:ezsp:ezsp <== Frame: 238001630000000000
2022-11-06T20:16:38.940Z zigbee-herdsman:adapter:ezsp:ezsp <== 0x63: {"_id_":99,"_cls_":"getMulticastTableEntry","_isRequest_":false,"value":{"multicastId":0,"endpoint":0,"networkIndex":0}}
2022-11-06T20:16:38.941Z zigbee-herdsman:adapter:ezsp:uart -+- waiting (4) success
2022-11-06T20:16:38.941Z zigbee-herdsman:adapter:ezsp:cast MulticastTableEntry[1] = EmberMulticastTableEntry: {"multicastId":0,"endpoint":0,"networkIndex":0}
2022-11-06T20:16:38.941Z zigbee-herdsman:adapter:ezsp:ezsp ==> getMulticastTableEntry: {"index":2}
2022-11-06T20:16:38.941Z zigbee-herdsman:adapter:ezsp:ezsp ==> {"_cls_":"getMulticastTableEntry","_id_":99,"_isRequest_":true,"index":2}
2022-11-06T20:16:38.941Z zigbee-herdsman:adapter:ezsp:uart --> DATA (4,7,0): 240001630002
2022-11-06T20:16:38.941Z zigbee-herdsman:adapter:ezsp:uart --> [476621a9372a1797b57e]
2022-11-06T20:16:38.941Z zigbee-herdsman:adapter:ezsp:uart -?- waiting (5)
2022-11-06T20:16:38.994Z zigbee-herdsman:adapter:ezsp:uart <-- [7566a1a9]
2022-11-06T20:16:38.995Z zigbee-herdsman:adapter:ezsp:uart <-- [372a15b25994982c7e]
2022-11-06T20:16:38.995Z zigbee-herdsman:adapter:ezsp:uart <-- DATA (7,5,0): 7566a1a9372a15b25994982c7e
2022-11-06T20:16:38.995Z zigbee-herdsman:adapter:ezsp:uart --> ACK  (0)
2022-11-06T20:16:38.995Z zigbee-herdsman:adapter:ezsp:uart --> [8070787e]
2022-11-06T20:16:38.995Z zigbee-herdsman:adapter:ezsp:ezsp <== Frame: 248001630000000000
2022-11-06T20:16:38.995Z zigbee-herdsman:adapter:ezsp:ezsp <== 0x63: {"_id_":99,"_cls_":"getMulticastTableEntry","_isRequest_":false,"value":{"multicastId":0,"endpoint":0,"networkIndex":0}}
2022-11-06T20:16:38.995Z zigbee-herdsman:adapter:ezsp:uart -+- waiting (5) success
2022-11-06T20:16:38.995Z zigbee-herdsman:adapter:ezsp:cast MulticastTableEntry[2] = EmberMulticastTableEntry: {"multicastId":0,"endpoint":0,"networkIndex":0}
2022-11-06T20:16:38.996Z zigbee-herdsman:adapter:ezsp:ezsp ==> getMulticastTableEntry: {"index":3}
2022-11-06T20:16:38.996Z zigbee-herdsman:adapter:ezsp:ezsp ==> {"_cls_":"getMulticastTableEntry","_id_":99,"_isRequest_":true,"index":3}
2022-11-06T20:16:38.996Z zigbee-herdsman:adapter:ezsp:uart --> DATA (5,0,0): 250001630003
2022-11-06T20:16:38.996Z zigbee-herdsman:adapter:ezsp:uart --> [506721a9372a16ec0b7e]
2022-11-06T20:16:38.996Z zigbee-herdsman:adapter:ezsp:uart -?- waiting (6)
2022-11-06T20:16:39.052Z zigbee-herdsman:adapter:ezsp:uart <-- [0667a1a9372a15b259941d41]
2022-11-06T20:16:39.052Z zigbee-herdsman:adapter:ezsp:uart <-- [7e]
2022-11-06T20:16:39.053Z zigbee-herdsman:adapter:ezsp:uart <-- DATA (0,6,0): 0667a1a9372a15b259941d417e
2022-11-06T20:16:39.053Z zigbee-herdsman:adapter:ezsp:uart --> ACK  (1)
2022-11-06T20:16:39.053Z zigbee-herdsman:adapter:ezsp:uart --> [8160597e]
2022-11-06T20:16:39.053Z zigbee-herdsman:adapter:ezsp:ezsp <== Frame: 258001630000000000
2022-11-06T20:16:39.053Z zigbee-herdsman:adapter:ezsp:ezsp <== 0x63: {"_id_":99,"_cls_":"getMulticastTableEntry","_isRequest_":false,"value":{"multicastId":0,"endpoint":0,"networkIndex":0}}
2022-11-06T20:16:39.053Z zigbee-herdsman:adapter:ezsp:uart -+- waiting (6) success
2022-11-06T20:16:39.053Z zigbee-herdsman:adapter:ezsp:cast MulticastTableEntry[3] = EmberMulticastTableEntry: {"multicastId":0,"endpoint":0,"networkIndex":0}
2022-11-06T20:16:39.053Z zigbee-herdsman:adapter:ezsp:ezsp ==> getMulticastTableEntry: {"index":4}
2022-11-06T20:16:39.053Z zigbee-herdsman:adapter:ezsp:ezsp ==> {"_cls_":"getMulticastTableEntry","_id_":99,"_isRequest_":true,"index":4}
2022-11-06T20:16:39.053Z zigbee-herdsman:adapter:ezsp:uart --> DATA (6,1,0): 260001630004
2022-11-06T20:16:39.053Z zigbee-herdsman:adapter:ezsp:uart --> [616421a9372a7d31b3e07e]
2022-11-06T20:16:39.054Z zigbee-herdsman:adapter:ezsp:uart -?- waiting (7)
2022-11-06T20:16:39.318Z zigbee-herdsman:adapter:ezsp:uart <-- [1764a1a9372a15b259]
2022-11-06T20:16:39.319Z zigbee-herdsman:adapter:ezsp:uart <-- [94cbd67e]
2022-11-06T20:16:39.320Z zigbee-herdsman:adapter:ezsp:uart <-- DATA (1,7,0): 1764a1a9372a15b25994cbd67e
2022-11-06T20:16:39.320Z zigbee-herdsman:adapter:ezsp:uart --> ACK  (2)
2022-11-06T20:16:39.320Z zigbee-herdsman:adapter:ezsp:uart --> [82503a7e]
2022-11-06T20:16:39.320Z zigbee-herdsman:adapter:ezsp:ezsp <== Frame: 268001630000000000
2022-11-06T20:16:39.320Z zigbee-herdsman:adapter:ezsp:ezsp <== 0x63: {"_id_":99,"_cls_":"getMulticastTableEntry","_isRequest_":false,"value":{"multicastId":0,"endpoint":0,"networkIndex":0}}
2022-11-06T20:16:39.320Z zigbee-herdsman:adapter:ezsp:uart -+- waiting (7) success
2022-11-06T20:16:39.320Z zigbee-herdsman:adapter:ezsp:cast MulticastTableEntry[4] = EmberMulticastTableEntry: {"multicastId":0,"endpoint":0,"networkIndex":0}
2022-11-06T20:16:39.320Z zigbee-herdsman:adapter:ezsp:ezsp ==> getMulticastTableEntry: {"index":5}
2022-11-06T20:16:39.320Z zigbee-herdsman:adapter:ezsp:ezsp ==> {"_cls_":"getMulticastTableEntry","_id_":99,"_isRequest_":true,"index":5}
[... cut ...]
Zigbee2MQTT:error 2022-11-06 21:16:41: Failed to start zigbee
Zigbee2MQTT:error 2022-11-06 21:16:41: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start.html for possible solutions
Zigbee2MQTT:error 2022-11-06 21:16:41: Exiting...
Zigbee2MQTT:error 2022-11-06 21:16:41: Error: Failure send getMulticastTableEntry:{"type":"Buffer","data":[40,0,1,99,0,6]}
    at /app/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/ezsp.ts:510:23
    at Queue.executeNext (/app/node_modules/zigbee-herdsman/src/utils/queue.ts:32:32)

kirovilya, 2022-11-08:
@kaciker sorry for delay. you have an interesting log - useful. this is the first time I've seen packets being resent from the chip to the host:
```
2022-11-06T20:16:39.320Z zigbee-herdsman:adapter:ezsp:uart <-- DATA (1,7,0): 1764a1a9372a15b25994cbd67e
...
2022-11-06T20:16:39.787Z zigbee-herdsman:adapter:ezsp:uart <-- DATA (1,7,1): 1f64a1a9372a15b25994819d7e
...
2022-11-06T20:16:39.811Z zigbee-herdsman:adapter:ezsp:uart <-- DATA (2,0,0): 2065a1a9372a15b2599491947e
...
2022-11-06T20:16:40.626Z zigbee-herdsman:adapter:ezsp:uart <-- DATA (2,0,1): 2865a1a9372a15b25994dbdf7e
```
I suggest you make backup files and try replace the attached file in your z2m installation and check again. record herdsman log.
[uart.zip](https://github.com/Koenkk/zigbee2mqtt/files/9962913/uart.zip)
the file is in the folder zigbee2mqtt/node_modules/zigbee-herdsman/dist/adapter/ezsp/driver/uart.js

kirovilya, 2022-11-08:
@kaciker it looks like communication problems between the z2m host and your tcp-port on gateway.

kaciker, 2022-11-08:
> @kaciker sorry for delay. you have an interesting log - useful. this is the first time I've seen packets being resent from the chip to the host:
>
> ```
> 2022-11-06T20:16:39.320Z [...]

I'm not sure if I I have this file in my HA installation, I have not this folder/file structure

kaciker, 2022-11-08:
> @kaciker it looks like communication problems between the z2m host and your tcp-port on gateway.

But then, why works in version 1.27 ??
[image]

kirovilya, 2022-11-08:
@kaciker this is also an important question for me - I don't know

kirovilya, 2022-11-12:
@kaciker on dev/edge tried again to handle package errors. can you try?

kaciker, 2022-11-13:
> @kaciker on dev/edge tried again to handle package errors. can you try?

Hello

This is the log, it was working around 2', and after that fails
I deployed a LHX proxmox zigbee2mqtt, and in this, system is stable.
If you need/want I can still doing test in HA version

2022-11-13T18:46:13.548Z zigbee-herdsman:adapter:ezsp:ezsp Set EzspPolicyId.TC_KEY_REQUEST_POLICY = 81
2022-11-13T18:46:13.548Z zigbee-herdsman:adapter:ezsp:ezsp ==> setPolicy: {"policyId":5,"decisionId":81}
2022-11-13T18:46:13.548Z zigbee-herdsman:adapter:ezsp:ezsp ==> {"_cls_":"setPolicy","_id_":85,"_isRequest_":true,"policyId":5,"decisionId":81}
2022-11-13T18:46:13.548Z zigbee-herdsman:adapter:ezsp:uart --> DATA (5,5,0): 0d00015500055100
2022-11-13T18:46:13.548Z zigbee-herdsman:adapter:ezsp:uart --> [554f21a9012a10e3590b3c7e]
2022-11-13T18:46:13.548Z zigbee-herdsman:adapter:ezsp:uart -?- waiting (6)
2022-11-13T18:46:13.625Z zigbee-herdsman:adapter:ezsp:uart <-- [564fa1a901]
2022-11-13T18:46:13.628Z zigbee-herdsman:adapter:ezsp:uart <-- [2a1510d27e]
2022-11-13T18:46:13.628Z zigbee-herdsman:adapter:ezsp:uart <-- DATA (5,6,0): 564fa1a9012a1510d27e
2022-11-13T18:46:13.628Z zigbee-herdsman:adapter:ezsp:uart --> ACK  (6)
2022-11-13T18:46:13.628Z zigbee-herdsman:adapter:ezsp:uart --> [8610be7e]
2022-11-13T18:46:13.628Z zigbee-herdsman:adapter:ezsp:ezsp <== Frame: 0d8001550000
2022-11-13T18:46:13.628Z zigbee-herdsman:adapter:ezsp:ezsp <== 0x55: {"_id_":85,"_cls_":"setPolicy","_isRequest_":false,"status":0}
2022-11-13T18:46:13.628Z zigbee-herdsman:adapter:ezsp:uart -+- waiting (6) success
2022-11-13T18:46:13.628Z zigbee-herdsman:adapter:ezsp:ezsp Set EzspPolicyId.TRUST_CENTER_POLICY = 3
2022-11-13T18:46:13.628Z zigbee-herdsman:adapter:ezsp:ezsp ==> setPolicy: {"policyId":0,"decisionId":3}
2022-11-13T18:46:13.628Z zigbee-herdsman:adapter:ezsp:ezsp ==> {"_cls_":"setPolicy","_id_":85,"_isRequest_":true,"policyId":0,"decisionId":3}
2022-11-13T18:46:13.628Z zigbee-herdsman:adapter:ezsp:uart --> DATA (6,6,0): 0e00015500000300
2022-11-13T18:46:13.628Z zigbee-herdsman:adapter:ezsp:uart --> [664c21a9012a15b15999227e]
2022-11-13T18:46:13.628Z zigbee-herdsman:adapter:ezsp:uart -?- waiting (7)
2022-11-13T18:46:13.722Z zigbee-herdsman:adapter:ezsp:uart <-- [674ca1a9012a153fde7e]
2022-11-13T18:46:13.722Z zigbee-herdsman:adapter:ezsp:uart <-- DATA (6,7,0): 674ca1a9012a153fde7e
2022-11-13T18:46:13.722Z zigbee-herdsman:adapter:ezsp:uart --> ACK  (7)
2022-11-13T18:46:13.722Z zigbee-herdsman:adapter:ezsp:uart --> [87009f7e]
2022-11-13T18:46:13.722Z zigbee-herdsman:adapter:ezsp:ezsp <== Frame: 0e8001550000
2022-11-13T18:46:13.722Z zigbee-herdsman:adapter:ezsp:ezsp <== 0x55: {"_id_":85,"_cls_":"setPolicy","_isRequest_":false,"status":0}
2022-11-13T18:46:13.722Z zigbee-herdsman:adapter:ezsp:uart -+- waiting (7) success
2022-11-13T18:46:13.722Z zigbee-herdsman:adapter:ezsp:ezsp Set EzspValueId.VALUE_END_DEVICE_KEEP_ALIVE_SUPPORT_MODE = 3
2022-11-13T18:46:13.722Z zigbee-herdsman:adapter:ezsp:ezsp ==> setValue: {"valueId":63,"value":3}
2022-11-13T18:46:13.722Z zigbee-herdsman:adapter:ezsp:ezsp ==> {"_cls_":"setValue","_id_":171,"_isRequest_":true,"valueId":63,"value":3}
2022-11-13T18:46:13.722Z zigbee-herdsman:adapter:ezsp:uart --> DATA (7,7,0): 0f0001ab003f0003
2022-11-13T18:46:13.722Z zigbee-herdsman:adapter:ezsp:uart --> [774d21a9ff2a2ab25a7d3a767e]
2022-11-13T18:46:13.723Z zigbee-herdsman:adapter:ezsp:uart -?- waiting (0)
2022-11-13T18:46:13.826Z zigbee-herdsman:adapter:ezsp:uart <-- [704da1a9ff2a15ac127e]
2022-11-13T18:46:13.826Z zigbee-herdsman:adapter:ezsp:uart <-- DATA (7,0,0): 704da1a9ff2a15ac127e
2022-11-13T18:46:13.826Z zigbee-herdsman:adapter:ezsp:uart --> ACK  (0)
2022-11-13T18:46:13.826Z zigbee-herdsman:adapter:ezsp:uart --> [8070787e]
2022-11-13T18:46:13.826Z zigbee-herdsman:adapter:ezsp:ezsp <== Frame: 0f8001ab0000
2022-11-13T18:46:13.826Z zigbee-herdsman:adapter:ezsp:ezsp <== 0xab: {"_id_":171,"_cls_":"setValue","_isRequest_":false,"status":0}
2022-11-13T18:46:13.826Z zigbee-herdsman:adapter:ezsp:uart -+- waiting (0) success
2022-11-13T18:46:13.827Z zigbee-herdsman:adapter:ezsp:ezsp Set EzspValueId.VALUE_CCA_THRESHOLD = 0
2022-11-13T18:46:13.827Z zigbee-herdsman:adapter:ezsp:ezsp ==> setValue: {"valueId":21,"value":0}
2022-11-13T18:46:13.827Z zigbee-herdsman:adapter:ezsp:ezsp ==> {"_cls_":"setValue","_id_":171,"_isRequest_":true,"valueId":21,"value":0}
2022-11-13T18:46:13.827Z zigbee-herdsman:adapter:ezsp:uart --> DATA (0,0,0): 100001ab00150000
2022-11-13T18:46:13.827Z zigbee-herdsman:adapter:ezsp:uart --> [005221a9ff2a00b25995147e]
2022-11-13T18:46:13.827Z zigbee-herdsman:adapter:ezsp:uart -?- waiting (1)
2022-11-13T18:46:14.827Z zigbee-herdsman:adapter:ezsp:uart -!- break waiting (1)
2022-11-13T18:46:14.827Z zigbee-herdsman:adapter:ezsp:uart Can't send DATA frame (0,0,0): 100001ab00150000
2022-11-13T18:46:14.827Z zigbee-herdsman:adapter:ezsp:uart ->> DATA (0,0,1): 100001ab00150000
2022-11-13T18:46:14.828Z zigbee-herdsman:adapter:ezsp:uart --> [085221a9ff2a00b259bceb7e]
2022-11-13T18:46:14.828Z zigbee-herdsman:adapter:ezsp:uart -?- rewaiting (1)
2022-11-13T18:46:15.827Z zigbee-herdsman:adapter:ezsp:uart -!- break rewaiting (1)
2022-11-13T18:46:15.827Z zigbee-herdsman:adapter:ezsp:uart Can't resend DATA frame (0,0,1): 100001ab00150000
2022-11-13T18:46:15.828Z zigbee-herdsman:adapter:ezsp:erro Reset error Error: Failure send setValue:{"type":"Buffer","data":[16,0,1,171,0,21,0,0]}
[... cut ...]
2022-11-13T18:46:26.551Z zigbee-herdsman:adapter:ezsp:uart --> DATA (3,0,0): 1300012200fe
2022-11-13T18:46:26.551Z zigbee-herdsman:adapter:ezsp:uart --> [305121a9762aebe9537e]
2022-11-13T18:46:26.551Z zigbee-herdsman:adapter:ezsp:uart -?- waiting (4)
Error: Failure send addTransientLinkKey:{"type":"Buffer","data":[18,0,1,175,0,255,255,255,255,255,255,255,255,90,105,103,66,101,101,65,108,108,105,97,110,99,101,48,57]}
    at /app/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/ezsp.ts:510:23
    at Queue.executeNext (/app/node_modules/zigbee-herdsman/src/utils/queue.ts:32:32)

aidan-, 2022-11-16:
Not sure if this helps or not - I have just hit the same error being returned which resulted in z2m crashing.  In my situation, my USB device (attached to a USB extension cable) had fallen over and was no longer detected on the system.  Re-plugging it in and restarting z2m seems to have resolved the issue, but I will monitor in-case it happens again and I can share some more insight (logs).

kaciker, 2022-11-16:
Hello @kirovilya although I installed a new fresh z2m in a proxmox lcx, it's not working after some time. I updated the file uart.js sent by you, and switching to Dev branch, finally still not working.
I attach the log
[log 161122 herd.txt](https://github.com/Koenkk/zigbee2mqtt/files/10020855/log.161122.herd.txt)

kirovilya, 2022-11-16:
@kaciker sadly. I can't understand why it's so bad for you. it was possible not to change uart.js any more - the changes have already been made to the dev version.
I see the main problem in a bad connection between z2m and the port. sometimes the data is sent poorly and then errors occur.
will already increase the timeout, has already done some error handling ... and it still falls.

now another try - close the port in case of errors and re-create the connection - on dev|edge version replace the ezsp.js file in the folder zigbee2mqtt/node_modules/zigbee-herdsman/dist/adapter/ezsp/driver/
[ezsp.zip](https://github.com/Koenkk/zigbee2mqtt/files/10020986/ezsp.zip)

kaciker, 2022-11-17:
> @kaciker sadly. I can't understand why it's so bad for you. it was possible not to change uart.js any more - the changes have already been made to the dev version. I see the main problem in a bad [...]

I replaced the file, 2tm appears online from yesterday but sensors are not updated . I'm confused, they you will see as last time seen 1 day, not data is refresh but devices are online ??

[image]

I can not understand why, but now, I can not lauch the herdsman log...

root@zigbee2mqtt:~# DEBUG=zigbee-herdsman* npm start
npm ERR! code ENOENT
npm ERR! syscall open
npm ERR! path /root/package.json
npm ERR! errno -2
npm ERR! enoent ENOENT: no such file or directory, open '/root/package.json'
npm ERR! enoent This is related to npm not being able to find a file.
npm ERR! enoent

npm ERR! A complete log of this run can be found in:
npm ERR!     /root/.npm/_logs/2022-11-17T17_15_39_166Z-debug-0.log
root@zigbee2mqtt:~#

kirovilya, 2022-11-18:
@kaciker you must run commands from the folder where z2m is installed (usually `/opt/zigbee2mqtt`), not from `/root`

kaciker, 2022-11-18:
This is another fresh server lcx in proxmox. In this log you will see that z2m start and works, but when I reuqest to have a new request, z2m down. After Y restar again and got the same origial error

  zigbee-herdsman:adapter:ezsp:uart -+- waiting (7) success +0ms
Zigbee2MQTT:debug 2022-11-18 14:12:12: Received MQTT message on 'homeassistant/b                                                                                                       utton/Dell-Latitude7290/Dell-Latitude7290_apagar/config' with data '{"availabili                                                                                                       ty_topic":"homeassistant/sensor/Dell-Latitude7290/availability","command_topic":                                                                                                       "homeassistant/button/Dell-Latitude7290/Dell-Latitude7290_apagar/set","action_to                                                                                                       pic":"homeassistant/button/Dell-Latitude7290/Dell-Latitude7290_apagar/action","u                                                                                                       nique_id":"5cd61a22-1bdf-4e41-96cf-8665c0f5aa7b","device":{"identifiers":"hass.a                                                                                                       gent-Dell-Latitude7290","manufacturer":"LAB02 Research","model":"Microsoft Windo                                                                                                       ws NT 10.0.22621.0","name":"Dell-Latitude7290","sw_version":"2022.13.0"},"name":                                                                                                       "Dell-Latitude7290_apagar","state_topic":"homeassistant/button/Dell-Latitude7290                                                                                                       /Dell-Latitude7290_apagar/state"}'
Zigbee2MQTT:debug 2022-11-18 14:12:12: Received MQTT message on 'homeassistant/b                                                                                                       utton/Dell-Latitude7290/Dell-Latitude7290_dormir/config' with data '{"availabili                                                                                                       ty_topic":"homeassistant/sensor/Dell-Latitude7290/availability","command_topic":                                                                                                       "homeassistant/button/Dell-Latitude7290/Dell-Latitude7290_dormir/set","action_to                                                                                                       pic":"homeassistant/button/Dell-Latitude7290/Dell-Latitude7290_dormir/action","u                                                                                                       nique_id":"af596d76-95c2-4d85-bdca-064fd19ece44","device":{"identifiers":"hass.a                                                                                                       gent-Dell-Latitude7290","manufacturer":"LAB02 Research","model":"Microsoft Windo                                                                                                       ws NT 10.0.22621.0","name":"Dell-Latitude7290","sw_version":"2022.13.0"},"name":                                                                                                       "Dell-Latitude7290_dormir","state_topic":"homeassistant/button/Dell-Latitude7290                                                                                                       /Dell-Latitude7290_dormir/state"}'
Zigbee2MQTT:debug 2022-11-18 14:12:13: Received MQTT message on 'zigbee2mqtt/bri                                                                                                       dge/request/permit_join' with data '{"device":null,"time":254,"transaction":"2my                                                                                                       4j-1","value":true}'
Zigbee2MQTT:info  2022-11-18 14:12:13: Zigbee: allowing new devices to join.
  zigbee-herdsman:adapter:ezsp:ezsp ==> addTransientLinkKey: {"partner":{"_value                                                                                                       ":{"type":"Buffer","data":[255,255,255,255,255,255,255,255]}},"transientKey":{"c                                                                                                       ontents":{"type":"Buffer","data":[90,105,103,66,101,101,65,108,108,105,97,110,99                                                                                                       ,101,48,57]}}} +3s
  zigbee-herdsman:adapter:ezsp:ezsp ==> {"_cls_":"addTransientLinkKey","_id_":17                                                                                                       5,"_isRequest_":true,"partner":{"_value":{"type":"Buffer","data":[255,255,255,25                                                                                                       5,255,255,255,255]}},"transientKey":{"contents":{"type":"Buffer","data":[90,105,                                                                                                       103,66,101,101,65,108,108,105,97,110,99,101,48,57]}}} +0ms
  zigbee-herdsman:adapter:ezsp:uart --> DATA (7,3,0): 470001af00ffffffffffffffff                                                                                                       5a6967426565416c6c69616e63653039 +3s
[... cut ...]
Zigbee2MQTT:error 2022-11-18 14:13:26: Error: Failure send setPolicy:{"type":"Buffer","data":[11,0,1,85,0,9,0,0]}
    at /opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/ezsp.ts:516:23
    at Queue.executeNext (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/utils/queue.ts:32:32)
  zigbee-herdsman:adapter:ezsp:uart Port closed +32ms
  zigbee-herdsman:adapter:ezsp:ezsp Close ezsp +32ms
  zigbee-herdsman:adapter:ezsp:driv Close driver +32ms
root@zigbee2mqtt:/opt/zigbee2mqtt#

kirovilya, 2022-11-18:
@kaciker  does everything really work for you on z2m version 1.27.2? it doesn't seem to work... I can't repeat similar errors in myself - that's why attempts to fix them are not successful

kirovilya, 2022-11-21:
@kaciker It's me again... another option for you.
find the driver.js file and line 36 in it. should look like this:
`         this.queue = new utils_2.Queue(8);`
change it by removing the number 8:
`         this.queue = new utils_2.Queue();`
is the number of concurrent requests. then restart z2m and observe.
this is one of the changes that makes version 1.27.1 different from the current version. but the chances that it will help - little.

kaciker, 2022-11-21:
> @kaciker It's me again... another option for you. find the driver.js file and line 36 in it. should look like this: ` this.queue = new utils_2.Queue(8);` change it by removing the number 8: ` [...]

Thaks a lot, I will do asap I apreciate your aomitment. I moved all sensors to ZHA, I nee to look the time and restore some sensors to m2z again

kaciker, 2022-11-25:
Hello, @kirovilya after 5 days of stability I tell you what I did. This was before your last antion with the new mod.
I flashed from scratch the sonoff, needed to weld again the pins.. and I installed everything but with a firm Tasmota 10. This was the version running when all sytem was stable. Now z2m in the its last version is working without stops.
Tested in CT Proxmox, and in Hassio addon.
I appreciate your effort and dedication to support, In case of the system become like previusly I'll aply your proposal.

kirovilya, 2022-11-25:
@kaciker You just updated the firmware on esp32 (tasmota) and that's it? now everything works on the latest version of z2m (1.28.2) or on dev / edge?

kaciker, 2022-11-25:
> @kaciker You just updated the firmware on esp32 (tasmota) and that's it? now everything works on the latest version of z2m (1.28.2) or on dev / edge?

Hello, is not sonoff Esp32, not the Pro model (Esp32) is the old one (Esp8266) based. I was in Tasmota 12 version. Is not posible dowgrade to previous by OTA, so I flashed as a new device, with Tasmota 10, after upgraded the zigbee module ncp-uart-sw_6.7.8 version, and set somme Tasmota rules, standard. 
I tested 1.28.2 Dev version in CT Proxmox, and seeing that works, I upgraded HAssio last versioon 1.28.1 and is working from then.
[image]

kirovilya, 2022-11-25:
ok, let esp8266, not esp32... but I still don't understand how it could not work? :)) miracles...
I will be glad if this completely solved your problem

kaciker, 2022-11-25:
> ok, let esp8266, not esp32... but I still don't understand how it could not work? :)) miracles... I will be glad if this completely solved your problem

Yes.. maybe some changes in Tasmota 11/12 did something.. I don't want to restart again testing again 12 :P

MattWestb, 2022-11-25:
@kaciker If you is getting problem with sleeping end device can the reason being that you is using one not hot fixed firmware for your Zigbee module `ncp-uart-sw_6.7.8` .
All Sonoff EZSP must running version 6.7.9..0 or newer for fixing miss tuned radio that is fixed in the latest firmware.

https://github.com/xsp1989/zigbeeFirmware/tree/master/firmware/Zigbee3.0_Dongle-NoSigned/EZSP#679-zigbee-emberznet-679

kaciker, 2022-11-26:
> @kaciker If you is getting problem with sleeping end device can the reason being that you is using one not hot fixed firmware for your Zigbee module `ncp-uart-sw_6.7.8` .
>
> All Sonoff EZSP must [...]

Hello, my device is sonoff wifi, not uss dongle. Can you confirm that those versions updated are compatible ??
I aleáis follow that 
Where the version 6.7.9 is not recomendad and updated 2 years ago 

https://github.com/arendst/Tasmota/tree/development/tools/fw_SonoffZigbeeBridge_ezsp

MattWestb, 2022-11-26:
The tsamot readme.md is 2 years old and if reading the link to the firmware its written that 6.7.9 is having one important fix.
If you like running the Zigbee module on other version you is free to do but dont complaining if getting problems.

kaciker, 2022-11-26:
> The tsamot readme.md is 2 years old and if reading the link to the firmware its written that 6.7.9 is having one important fix. If you like running the Zigbee module on other version you is free to do but dont complaining if getting problems.

@MattWestb , I only tried to follow the recomendations from Tasmota.
This OTA files are not toucehd from 2 years, ant in those time this was the last instructions.
Nevertheless, I  lose nothing by trying it.. ÇI upgraded to this version rightnow, lest to see

[image]

MattWestb, 2022-11-26:
I think i was wrong then thinking you was using one dongle and not one ZBB.

Tasmota is using the Coordinator without security and Z2M and ZHA is using standard Zigbee 3 security and the manufacture is still recommending running on there EZSP 6.7.9 or you can getting problem with radio stability (most with sleeping end devices) as from my link to the manufacture:
> 679 (Zigbee EmberZNet 6.7.9)
Changed configuration to set HFXO CTUNE value to 128,

And as you have saying its being tested for over 2 years and is proven working well and tasmota have not  updating there GIT instructions.
ZHA users have giving good feedback all the time from the updated firmware and we can only recording the EZSP 6.7.8 (Sonoff 6.7.9) for production systems and stable candidate is 6.10.3.0 that is not getting for the ZBB for the moment.

By the way if is running the original ZBB is need the signed firmware from https://github.com/xsp1989/zigbeeFirmware/tree/master/firmware/ZigbeeBridge_SM-011-signed.
and if its not then you shall using the unsigned firmware from https://github.com/xsp1989/zigbeeFirmware/tree/master/firmware/Zigbee3.0_Dongle-NoSigned/EZSP.
Info from: https://github.com/xsp1989/zigbeeFirmware#special-note.

And the linked GIT is from the maker of the firmware from the factory and is the only one that can doing the signed firmware is its 100% secure as long you is using the right type (Signed or not and it shall not being possible flashing the wrong one).

PS: tasmota Zigbee firmware is the same as the signed linked and is copped from the factory GIT.

closed 2023-01-03 by github-actions[bot]