Koenkk/zigbee2mqtt #14573: Unable to add EndDevices to the group

kovaga, 2022-10-21:
### What happened?

Added an EndDevice to the group via GUI.
Z2M reports failure of such action.

I am unable to add any EndDevices to the group. I have tried with 7 different types of EndDevices
However, the routers are added instantly. I have tried with 9 different types of routers

### What did you expect to happen?

Expected the device to become part of the group.

### How to reproduce it (minimal and precise)

add device to the group.

### Zigbee2MQTT version

1.28.0

### Adapter firmware version

development firmware  {"meta":{"maintrel":1,"majorrel":2,"minorrel":7,"product":1,"revision":20220928,"transportrev":2},"type":"zStack3x0"}'

### Adapter

CC2652R development board 

### Debug log

Here are the logs of triying to connect Honeywell smoke sensor to the group, the following steps were taken

send selftest, to which device responded
created a new empty group
added device to the group

info 2022-10-21 10:19:31: MQTT publish: topic 'zigbee2mqtt/sens_smoke2', payload '{"ac_status":false,"battery":100,"battery_low":false,"device_temperature":33,"last_seen":1666340371963,"linkquality":255,"power_outage_count":37,"restore_reports":false,"sensitivity":"high","smoke":false,"smoke_density":0,"supervision_reports":false,"tamper":false,"test":false,"trouble":false,"voltage":3055}'

info 2022-10-21 10:19:46: MQTT publish: topic 'zigbee2mqtt/bridge/response/group/add', payload '{"data":{"friendly_name":"testgroup","id":1},"status":"ok","transaction":"kdq4w-4"}'

info 2022-10-21 10:19:52: Adding 'sens_smoke2' to 'testgroup'

info 2022-10-21 10:20:27: MQTT publish: topic 'zigbee2mqtt/bridge/response/group/members/add', payload '{"data":{"device":"0x00158d0002f37664/1","group":"testgroup"},"error":"Failed to add from group (Command 0x00158d0002f37664/1 genGroups.add({"groupid":1,"groupname":""}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 55803 - 1 - 49 - 4 - 0 after 10000ms))","status":"error","transaction":"kdq4w-5"}'

error 2022-10-21 10:20:27: Failed to add from group (Command 0x00158d0002f37664/1 genGroups.add({"groupid":1,"groupname":""}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 55803 - 1 - 49 - 4 - 0 after 10000ms))

If I manually specify the group in a configuration.yaml and restart z2m, I get the failed response.

Adding 'sens_smoke2' to group 'gsens_smoke02'
Failed to add 'sens_smoke2' from 'gsens_smoke02'

danieledwardgeorgehitchcock, 2022-10-21:
It looks like the group add command is timing out - is there a way that you can keep the devices awake (i.e. by clicking a sync button) on the device you are wanting to add to the group?

kovaga, 2022-10-21:
> It looks like the group add command is timing out - is there a way that you can keep the devices awake (i.e. by clicking a sync button) on the device you are wanting to add to the group?

Tried pressing the button multiple times on the device before and after the group request was sent. Didn't help. still timedout.

MattWestb, 2022-10-23:
End device is normally (some tuya is not) cant being  added to groups with add group commands then they is controllers.
Controllers is being bond to one group and is then sending the commands to the group and the lights is listening to the group and is getting the commands from the bounded controllers.

So controllers (remotes) = binding to one group.
Lights = Add group.
= working also then the coordinator / host system is off line ala Zigbee Light standard.

closed 2022-11-30 by github-actions[bot]

thomasklein1975, 2023-09-12:
Hi,

I have a problem to join a group with devices like lamps from same type, too.

**Lamps:**

- https://www.zigbee2mqtt.io/devices/CK-BL702-AL-01.html
- https://www.zigbee2mqtt.io/devices/LED1923R5.html
- https://www.zigbee2mqtt.io/devices/AC08560-DIM.html
- 

**Log:**
`error	Exception when trying to Add 0xa81710b78ce80000 to group 906. Error: Command 0xa81710b78ce80000/1 genGroups.add({"groupid":906,"groupname":""}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (no response received (23))
warn	calling addDevToGroup with 0xa81710b78ce80000, 906, 1
error	add dev to group Error: "Failed to add 0xa81710b78ce80000 to group 905: {}"
Exception when trying to Add 0xa81710b78ce80000 to group 905. Error: Command 0xa81710b78ce80000/1 genGroups.add({"groupid":905,"groupname":""}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (no response received (22))`

Zigbee Adapter: v1.8.23

Zigbee Coordinator:
type:ConBee2/RaspBee2
version:0-0.38.114.0.
revision:0x26720700