Koenkk/zigbee2mqtt #14579: [New device support]: Tesla Smart Thermostatic Valve TV100

Kedarius, 2022-10-21:
### Link

https://www.teslasmart.com/tesla-smart-thermostatic-valve

### Database entry

{"id":10,"type":"EndDevice","ieeeAddr":"0x84fd27fffe0e70ad","nwkAddr":8156,"manufId":4098,"manufName":"_TZE200_lnbfnyxd","powerSource":"Battery","modelId":"TS0601","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"inClusterList":[0,4,5,61184],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"stackVersion":0,"modelId":"TS0601","manufacturerName":"_TZE200_lnbfnyxd","dateCode":""}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":69,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1666372380756,"defaultSendRequestWhen":"immediate"}

### Comments

It seems that I managed to get it working by adding its product id to tuya.js at zigbee-herdsman-converters. It seems that it is at least similar to already supported valve:
{modelID: 'TS0601', manufacturerName: '_TZE200_husqqvux'}, /* model: 'TSL-TRV-TV01ZG', vendor: 'Tesla Smart' */
so just by adding a line with 
{modelID: 'TS0601', manufacturerName: '_TZE200_lnbfnyxd'}, /* model: 'TSL-TRV-TV01ZG', vendor: 'Tesla Smart' */
it started working. I can not verify if it is missing something as I do not have the original vendor app but the basic settings seems to be ok. And it seems that ZHA did the same in their code...

Koenkk, 2022-10-22:
Added!

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2022-10-22 by Koenkk

fanfuvac, 2023-11-02:
Hello,
I have checked the list of supported devices here: https://www.zigbee2mqtt.io/supported-devices/#v=Tesla%20Smart and the Tesla Smart Thermostatic Valve TV100 is not listed there, but as per this issue, it should now be included? Can you confirm? Thanks,
Vaclav

KarelBrt, 2024-03-06:
Hi Václav,

did you ever get any response to your question? I'm running Zigbee2mqtt in my Home Assistant more for 2 years and I've got 6 of these TRVs recently. They are obviously not integrated yet. I don't want to develop the wheel again and I'm also quite sick so this is why I'm asking...

Zdraví,
Karel

fanfuvac, 2024-03-06:
Hi Karle,

No, unfortunately I have never received an answer for this one. I have not bought this device after all, so I cannot confirm whether there is some way how to make it work...

KarelBrt, 2024-03-07:
Hi Václav, thanks a lot for a fast response. I will give this TRV a try, as I've found something very similar from a different vendor in the database, I just have to have a look over the Internet - I have an idea how to do it. For me this device looks very good from the mechanical POV. (eg. https://www.zigbee2mqtt.io/devices/ZTRV-ZX-TV01-MS.html). It's  not nice, but in the comparison with the other device from Tesla it is a state  of the art to me... BR, Karel From: fanfuvac ***@***.***>Sent: Wednesday, March 6, 2024 10:23:37 pmTo: Koenkk/zigbee2mqtt ***@***.***>Cc: KarelBrt ***@***.***>; Comment ***@***.***>Subject: Re: [Koenkk/zigbee2mqtt] [New device support]: Tesla Smart Thermostatic Valve TV100  (Issue #14579)
Hi Karle,
No, unfortunately I have never received an answer for this one. I have not bought this device after all, so I cannot confirm whether there is some way how to make it work...

KarelBrt, 2024-03-15:
Haha, 

I've tried to initialize this device close to my Zigbee  HA USB gateway and it worked(!), the only issue seems, that it is detected as a different type from the same manufacturer. In his case as  a "TuYa TV02-Zigbee" but the box shows "TSL-TRV100-TV01ZG", the model and picture are of course different. Anyway it seems easy to me, as everything else looks working so far ;8-) I hope that this could help somebody...
Anyway I saw somewhere **that the pairing should be done close to the gateway**. I'm going to move the valve upstairs, now and I will see. Maybe the "fixed" Lidl gateway will find its reason again...

BR.

Karel

KarelBrt, 2025-01-07:
Hi Jan, I have no issues with this device Except of problems with the connectivity, but I'm going to solve it with router from IKEA As it is working smoothly, I've decided not to touch it because I'm still sick. But you can give me a call if you would like to. My setup for Thursday is for instance following: 06:00/17 09:00/19 12:00/20 18:50/22 19:00/17 24:00/17 I can recall the only issue which was a bit not logical for me. Times are the turning point which means that  in this setup there is 17 degrees between 7 in the evening. and 6 in the morning.  
[image]
Karel k2peta@seznam.cz