Koenkk/zigbee2mqtt #14580: AwoX 33957 Wrong Color_Temp

MoosJ80, 2022-10-21:
### What happened?

I added 4 of these ligt bulbs. But the maximum color_temp in Zigbee2MQTT is 370 (Slider) but the actual maximum color_temp is: colorTempPhysicalMax":454

Iam able to enter 454 by hand.

### What did you expect to happen?

When i move the slider to the most right position the color temp value indicates 370. I expected 454.
The bulbs are also to cool

### Zigbee2MQTT version

1.28.01

### Adapter firmware version

20210120 (not able to find)

### Adapter

ZZH!

Koenkk, 2022-10-22:
Fixed!

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2022-10-22 by Koenkk

jdescoux, 2023-09-28:
Hi Koenkk, I think you should do the same for 33943/33944/33946.

It's better with [250,454] range:
```
model: '33943/33944/33946',
vendor: 'AwoX',
description: 'LED RGB & brightness',
extend: extend_1.default.light_onoff_brightness_colortemp_color({ colorTempRange: [250, 454], supportsHueAndSaturation: true }),
```