Koenkk/zigbee2mqtt #14581: Multiple messages when changing state

o0shojo0o, 2022-10-21:
### What happened?

If I set a state to `ON` another one the old state is transmitted twice and then only the new state:
(The logs are read from bottom to top)
```JSON
{"payload":{"brightness":83,"color":{"x":"0.6456","y":"0.3451"},"color_mode":"xy","color_temp":380,"last_seen":"2022-10-21T20:21:03+02:00","linkquality":131,"state":"ON"},"topic":"BU.Licht.Stimmung"}
{"payload":{"brightness":83,"color":{"x":"0.6456","y":"0.3451"},"color_mode":"xy","color_temp":380,"last_seen":"2022-10-21T20:21:03+02:00","linkquality":131,"state":"OFF"},"topic":"BU.Licht.Stimmung"}
{"payload":{"brightness":83,"color":{"x":"0.6456","y":"0.3451"},"color_mode":"xy","color_temp":380,"last_seen":"2022-10-21T20:21:03+02:00","linkquality":131,"state":"OFF"},"topic":"BU.Licht.Stimmung"}
{"payload":{"state":"ON"},"topic":"BU.Licht.Stimmung/set"}
```
If I set a state to `OFF` again the old state is transmitted once and then the new state is transmitted:
```JSON
{"payload":{"brightness":83,"color":{"x":"0.6456","y":"0.3451"},"color_mode":"xy","color_temp":380,"last_seen":"2022-10-21T20:20:33+02:00","linkquality":131,"state":"OFF"},"topic":"BU.Licht.Stimmung"}
{"payload":{"brightness":83,"color":{"x":"0.6456","y":"0.3451"},"color_mode":"xy","color_temp":380,"last_seen":"2022-10-21T20:20:33+02:00","linkquality":131,"state":"ON"},"topic":"BU.Licht.Stimmung"}
{"payload":{"state":"OFF"},"topic":"BU.Licht.Stimmung/set"}
```

### What did you expect to happen?

One message which publishes the new State. :-)

### How to reproduce it (minimal and precise)

This always happens.

### Zigbee2MQTT version

1.28.0-dev commit: a426ee1

### Adapter firmware version

20220219

### Adapter

SONOFF Zigbee 3.0 USB Dongle Plus ZBDongle-P

o0shojo0o, 2022-11-21:
Hi @Koenkk, 

is that the way you want it? 
If that should be so I would like to know why, because that complicates a direct clean processing, and "debouncing" or the same I would not like.

Koenkk, 2022-11-21:
Please provide the debug log so I can determine why these messages are published.

See https://www.zigbee2mqtt.io/guide/usage/debug.html on how to enable debug logging.

o0shojo0o, 2022-11-22:
Ok @Koenkk ,  here a debuglog after a fresh start and a time off and on again of the device `BU.Licht.Stimmung` 
[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/10069467/log.txt)

Koenkk, 2022-11-22:
Are these lights part of a group?

o0shojo0o, 2022-11-22:
Yes it is.

Koenkk, 2022-11-23:
Then this is probably due to https://www.zigbee2mqtt.io/guide/usage/groups.html#state-changes, you can confirm it by temporarily removing the device from the group.

Koenkk, 2022-11-23:
Can you provide the deubg log again when the device is not in the group?

o0shojo0o, 2022-11-23:
Yes of course. 
[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/10077565/log.txt)
[herdsman_log.txt](https://github.com/Koenkk/zigbee2mqtt/files/10077566/herdsman_log.txt)

Koenkk, 2022-11-23:
I see the issue but I cannot explain it yet.

Can you provide me your `configuration.yaml`?

o0shojo0o, 2022-11-23:
Yes, I have only removed the network_key :) 
[configuration.yaml.txt](https://github.com/Koenkk/zigbee2mqtt/files/10077658/configuration.yaml.txt)

o0shojo0o, 2022-11-24:
I can do that, but I don't think it will change anything since this behavior is the same for plug.

o0shojo0o, 2022-11-24:
As suspected, it has unfortunately not improved anything. 
Can still do something to help you troubleshoot?

Koenkk, 2022-11-25:
I found the reason now and the behaviour is as expected.

```
Zigbee2MQTT:debug 2022-11-23 17:43:41: Received MQTT message on 'zigbee2mqtt/BU.Licht.Stimmung/set' with data '{"state":"OFF"}'
Zigbee2MQTT:debug 2022-11-23 17:43:41: Publishing 'set' 'state' to 'BU.Licht.Stimmung'
# ... unrelevant ...
# Here the device responds to the request with a defaultReponse, this will trigger zigbee2mqtt to publish the new linkquality
2022-11-23T16:43:41.219Z zigbee-herdsman:controller:log Received 'zcl' data '{"frame":{"Header":{"frameControl":{"frameType":0,"manufacturerSpecific":false,"direction":1,"disableDefaultResponse":false,"reservedBits":0},"transactionSequenceNumber":18,"manufacturerCode":null,"commandIdentifier":11},"Payload":{"cmdId":4,"statusCode":0},"Command":{"ID":11,"name":"defaultRsp","parameters":[{"name":"cmdId","type":32},{"name":"statusCode","type":32}]}},"address":7826,"endpoint":1,"linkquality":131,"groupID":0,"wasBroadcast":false,"destinationEndpoint":1}'
Zigbee2MQTT:info  2022-11-23 17:43:41: MQTT publish: topic 'zigbee2mqtt/BU.Licht.Stimmung', payload '{"brightness":82,"color":{"x":0.6676,"y":0.3267},"color_mode":"xy","last_seen":"2022-11-23T17:43:41+01:00","linkquality":131,"state":"ON"}'
# Since the command has succeeded, Zigbee2MQTT publishes the new (optimistic) state of the device
Zigbee2MQTT:info  2022-11-23 17:43:41: MQTT publish: topic 'zigbee2mqtt/BU.Licht.Stimmung', payload '{"brightness":82,"color":{"x":0.6676,"y":0.3267},"color_mode":"xy","last_seen":"2022-11-23T17:43:41+01:00","linkquality":131,"state":"OFF"}'
```

o0shojo0o, 2022-11-25:
Do I understand this correctly, that the first message comes with the old state value to update the linkquality?

Koenkk, 2022-11-26:
@o0shojo0o exactly, at that moment the `STATE` has not been changed yet and when `cache_state` is `true` ([docs](https://www.zigbee2mqtt.io/guide/configuration/mqtt.html#server-connection)) (required for HA) it will always publish the full state.

o0shojo0o, 2022-11-27:
Thanks for taking the time to clarify the behavior.
With `cache_state` `false` I get the expected behavior of the messages.

I unfortunately did not come across this in the docs before.

closed 2022-11-27 by o0shojo0o

tomasbedrich, 2023-06-18:
Hello, what is the guidance to resolve the same issue when HA integration is required?