Koenkk/zigbee2mqtt #14582: New Loratap zigbee 1 gang dimmer module

Iroamstaff, 2022-10-21:
### Link

€ 16,53  50%de réduction | LoraTap – interrupteur de lumière à intensité réglable, Module de disjoncteur de relais, Tuya Smart Life App, télécommande, automatisation de la synchronisation https://a.aliexpress.com/_EzMEOYd

### Database entry

Warning 2022-10-21 20:50:04Received message from unsupported device with Zigbee model 'TS0601' and manufacturer name '_TZE200_vucankjx' Warning 2022-10-21 20:50:04Please see: https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html Warning 2022-10-21 20:52:41Received message from unsupported device with Zigbee model 'TS0601' and manufacturer name '_TZE200_vucankjx' Warning 2022-10-21 20:52:41Please see: https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html Info 2022-10-21 20:52:52MQTT publish: topic 'zigbee2mqtt/Prise orange ', payload '{"child_lock":"UNLOCK","current":0,"energy":0.21,"indicator_mode":"off/on","linkquality":142,"power":0,"power_outage_memory":"restore","state":"OFF","update":{"state":"idle"},"update_available":null,"voltage":234}' Info 2022-10-21 20:52:52MQTT publish: topic 'zigbee2mqtt/Prise orange ', payload '{"child_lock":"UNLOCK","current":0,"energy":0.21,"indicator_mode":"off/on","linkquality":138,"power":0,"power_outage_memory":"restore","state":"OFF","update":{"state":"idle"},"update_available":null,"voltage":234}' Info 2022-10-21 20:52:59MQTT publish: topic 'zigbee2mqtt/Prise orange ', payload '{"child_lock":"UNLOCK","current":0,"energy":0.21,"indicator_mode":"off/on","linkquality":138,"power":0,"power_outage_memory":"restore","state":"OFF","update":{"state":"idle"},"update_available":null,"voltage":233}' Info 2022-10-21 20:52:59MQTT publish: topic 'zigbee2mqtt/Prise orange ', payload '{"child_lock":"UNLOCK","current":0,"energy":0.21,"indicator_mode":"off/on","linkquality":138,"power":0,"power_outage_memory":"restore","state":"OFF","update":{"state":"idle"},"update_available":null,"voltage":233}' Warning 2022-10-21 20:54:29Device '0xa4c13866d5569a61' left the network Info 2022-10-21 20:54:30MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0xa4c13866d5569a61","ieee_address":"0xa4c13866d5569a61"},"type":"device_leave"}' Info 2022-10-21 20:54:39Zigbee: allowing new devices to join. Info 2022-10-21 20:54:40MQTT publish: topic 'zigbee2mqtt/bridge/response/permit_join', payload '{"data":{"time":254,"value":true},"status":"ok","transaction":"stvto-1"}' Info 2022-10-21 20:54:44Device '0xa4c13866d5569a61' joined Info 2022-10-21 20:54:44MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0xa4c13866d5569a61","ieee_address":"0xa4c13866d5569a61"},"type":"device_joined"}' Info 2022-10-21 20:54:44Starting interview of '0xa4c13866d5569a61' Info 2022-10-21 20:54:44MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0xa4c13866d5569a61","ieee_address":"0xa4c13866d5569a61","status":"started"},"type":"device_interview"}' Info 2022-10-21 20:54:44MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0xa4c13866d5569a61","ieee_address":"0xa4c13866d5569a61"},"type":"device_announce"}' Info 2022-10-21 20:54:45Successfully interviewed '0xa4c13866d5569a61', device has successfully been paired Warning 2022-10-21 20:54:45Device '0xa4c13866d5569a61' with Zigbee model 'TS0601' and manufacturer name '_TZE200_vucankjx' is NOT supported, please follow https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html Info 2022-10-21 20:54:45MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"definition":null,"friendly_name":"0xa4c13866d5569a61","ieee_address":"0xa4c13866d5569a61","status":"successful","supported":false},"type":"device_interview"}' Warning 2022-10-21 20:54:51Received message from unsupported device with Zigbee model 'TS0601' and manufacturer name '_TZE200_vucankjx' Warning 2022-10-21 20:54:51Please see: https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html Warning 2022-10-21 20:54:51Received message from unsupported device with Zigbee model 'TS0601' and manufacturer name '_TZE200_vucankjx' Warning 2022-10-21 20:54:51Please see: https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html Warning 2022-10-21 20:54:51Received message from unsupported device with Zigbee model 'TS0601' and manufacturer name '_TZE200_vucankjx' Warning 2022-10-21 20:54:52Please see: https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html Warning 2022-10-21 20:54:52Received message from unsupported device with Zigbee model 'TS0601' and manufacturer name '_TZE200_vucankjx' Warning 2022-10-21 20:54:52Please see: https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html Warning 2022-10-21 20:54:52Received message from unsupported device with Zigbee model 'TS0601' and manufacturer name '_TZE200_vucankjx' Warning 2022-10-21 20:54:52Please see: https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html Warning 2022-10-21 20:54:52Received message from unsupported device with Zigbee model 'TS0601' and manufacturer name '_TZE200_vucankjx' Warning 2022-10-21 20:54:52Please see: https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html Warning 2022-10-21 20:54:52Received message from unsupported device with Zigbee model 'TS0601' and manufacturer name '_TZE200_vucankjx' Warning 2022-10-21 20:54:52Please see: https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html Warning 2022-10-21 20:54:58Received message from unsupported device with Zigbee model 'TS0601' and manufacturer name '_TZE200_vucankjx' Warning 2022-10-21 20:54:58Please see: https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html Warning 2022-10-21 20:54:59Received message from unsupported device with Zigbee model 'TS0601' and manufacturer name '_TZE200_vucankjx' Warning 2022-10-21 20:54:59Please see: https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html Info 2022-10-21 20:57:28MQTT publish: topic 'zigbee2mqtt/Interrupteur ', payload '{"action":null,"battery":100,"device_temperature":26,"linkquality":163,"power_outage_count":48,"voltage":3015}'

### Comments

Novice sorry

Koenkk, 2022-10-22:
Could you check if the issue is fixed with the following external converter:

```js
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const ota = require('zigbee-herdsman-converters/lib/ota');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const utils = require('zigbee-herdsman-converters/lib/utils');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    fingerprint: tuya.fingerprint('TS0601', ['_TZE200_ip2akl4w', '_TZE200_1agwnems', '_TZE200_la2c2uo9', '_TZE200_579lguh2', '_TZE200_vucankjx']),
    model: 'TS0601_dimmer_1',
    vendor: 'TuYa',
    description: '1 gang smart dimmer',
    fromZigbee: [tuya.fzDataPoints],
    toZigbee: [tuya.tzDataPoints],
    configure: tuya.configureMagicPacket,
    exposes: [tuya.exposes.lightBrightnessWithMinMax(), tuya.exposes.powerOnBehavior(),
        tuya.exposes.countdown(), tuya.exposes.lightType()],
    meta: {
        tuyaDatapoints: [
            [1, 'state', tuya.valueConverter.onOff, {skip: tuya.skip.stateOnAndBrightnessPresent}],
            [2, 'brightness', tuya.valueConverter.scale0_254to0_1000],
            [3, 'min_brightness', tuya.valueConverter.scale0_254to0_1000],
            [4, 'light_type', tuya.valueConverter.lightType],
            [5, 'max_brightness', tuya.valueConverter.scale0_254to0_1000],
            [6, 'countdown', tuya.valueConverter.countdown],
            [14, 'power_on_behavior', tuya.valueConverter.powerOnBehavior],
        ],
    },
    whiteLabel: [
        {vendor: 'Moes', model: 'MS-105Z'},
        {vendor: 'Lerlink', model: 'X706U'},
        {vendor: 'Moes', model: 'ZS-EUD_1gang'},
    ],
};

module.exports = definition;
```

- save this as file next to `configuration.yaml` as `ext_converter.js`
- add it to `configuration.yaml`:
```yaml
external_converters:
  - ext_converter.js
```
- start z2m, check if device works

Iroamstaff, 2022-10-22:
Merci pour t'as rapidité de réponse
Rien ne change Il n'est toujours pas reconnu
Fonctionne qu'avec l'interrupteur brancher dessus

Iroamstaff, 2022-10-22:
Je suis novice je te montre pour être sûr d'avoir bien suivi ce que tu m'as dit faire pour pas te faire perdre ton temps par ma faute.
[image]
[image]
Et j'ai bien redémarré zigbee2mqtt

Iroamstaff, 2022-10-22:
une erreur de ma part j'ai enlevé les espaces devant external_convecters:
maintenant z2m ne démarre plus voici le log 
[14:06:42] INFO: Preparing to start...
[14:06:43] INFO: Socat not enabled
[14:06:45] INFO: Starting Zigbee2MQTT...
/app/dist/util/externally-loaded.js:20
    exposes: [tuya.exposes.lightBrightnessWithMinMax(), tuya.exposes.powerOnBehavior(),
                           ^
TypeError: tuya.exposes.lightBrightnessWithMinMax is not a function
    at /app/dist/util/externally-loaded.js:20:28
    at Script.runInContext (node:vm:139:12)
    at Script.runInNewContext (node:vm:144:17)
    at Object.runInNewContext (node:vm:298:38)
    at loadModuleFromText (/app/lib/util/utils.ts:148:8)
    at loadModuleFromFile (/app/lib/util/utils.ts:155:12)
    at Object.getExternalConvertersDefinitions (/app/lib/util/utils.ts:165:25)
    at getExternalConvertersDefinitions.next (<anonymous>)
    at new ExternalConverters (/app/lib/extension/externalConverters.ts:12:20)
    at new Controller (/app/lib/controller.ts:84:58)

Iroamstaff, 2022-11-05:
I just tried the external converter again since the last update and now everything works thank you you are on top 👍
[image]

closed 2022-11-05 by Iroamstaff

Koenkk, 2022-11-05:
Great, added!

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)