Koenkk/zigbee2mqtt #14589: Schneider Electric EKO09716 not responding.

therealeldaria, 2022-10-22:
### What happened?

I just bought 3 of the Schneider Electric EKO09716 after seeing on the Supported pages that they should work.
However it is only partially working, it seems to report  power, energy, current, voltage and linkquality.
But it is not possible to set or get switch (state) or power_on_behavior.

Error 2022-10-22 12:22:14Publish 'set' 'state' to 'kitchen_fridge_electric_socket' failed: 'Error: Command 0x04cd15fffec4c26a/242 genOnOff.off({}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (no response received)'

Error 2022-10-22 12:22:23Publish 'set' 'state' to 'kitchen_fridge_electric_socket' failed: 'Error: Command 0x04cd15fffec4c26a/242 genOnOff.on({}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (no response received)'

Error 2022-10-22 12:23:09Publish 'set' 'power_on_behavior' to 'kitchen_fridge_electric_socket' failed: 'Error: Write 0x04cd15fffec4c26a/242 genOnOff({"startUpOnOff":1}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (no response received)'

Error 2022-10-22 12:28:25Publish 'get' 'state' to 'kitchen_fridge_electric_socket' failed: 'Error: Read 0x04cd15fffec4c26a/242 genOnOff(["onOff"], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (no response received)'

Error 2022-10-22 12:28:26Publish 'get' 'power_on_behavior' to 'kitchen_fridge_electric_socket' failed: 'Error: Read 0x04cd15fffec4c26a/242 genOnOff(["startUpOnOff"], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (no response received)'

### What did you expect to happen?

Use all features of the wall switch.

### How to reproduce it (minimal and precise)

Pair device
Try to set or get Switch State, or Power_on_behaviour

### Zigbee2MQTT version

1.27.2

### Adapter firmware version

0x26780700

### Adapter

ConBee2

### Debug log

debug 2022-10-22 12:34:24: Received Zigbee message from 'kitchen_fridge_electric_socket', type 'attributeReport', cluster 'seMetering', data '{"instantaneousDemand":32981}' from endpoint 6 with groupID null
info  2022-10-22 12:34:24: MQTT publish: topic 'zigbee2mqtt/kitchen_fridge_electric_socket', payload '{"current":0.27,"energy":0.1,"linkquality":255,"power":32.981,"power_on_behavior":null,"state":"ON","voltage":233}'
debug 2022-10-22 12:34:28: Received Zigbee message from 'kitchen_fridge_electric_socket', type 'attributeReport', cluster 'seMetering', data '{"instantaneousDemand":33006}' from endpoint 6 with groupID null
info  2022-10-22 12:34:28: MQTT publish: topic 'zigbee2mqtt/kitchen_fridge_electric_socket', payload '{"current":0.27,"energy":0.1,"linkquality":255,"power":33.006,"power_on_behavior":null,"state":"ON","voltage":233}'
debug 2022-10-22 12:34:33: Received Zigbee message from 'kitchen_fridge_electric_socket', type 'attributeReport', cluster 'seMetering', data '{"instantaneousDemand":32988}' from endpoint 6 with groupID null
info  2022-10-22 12:34:33: MQTT publish: topic 'zigbee2mqtt/kitchen_fridge_electric_socket', payload '{"current":0.27,"energy":0.1,"linkquality":255,"power":32.988,"power_on_behavior":null,"state":"ON","voltage":233}'
debug 2022-10-22 12:34:38: Received Zigbee message from 'kitchen_fridge_electric_socket', type 'attributeReport', cluster 'seMetering', data '{"instantaneousDemand":32964}' from endpoint 6 with groupID null
info  2022-10-22 12:34:38: MQTT publish: topic 'zigbee2mqtt/kitchen_fridge_electric_socket', payload '{"current":0.27,"energy":0.1,"linkquality":255,"power":32.964,"power_on_behavior":null,"state":"ON","voltage":233}'
debug 2022-10-22 12:34:39: Received Zigbee message from 'kitchen_freezer_electric_socket', type 'attributeReport', cluster 'seMetering', data '{"instantaneousDemand":71084}' from endpoint 6 with groupID null
debug 2022-10-22 12:34:48: Received Zigbee message from 'kitchen_fridge_electric_socket', type 'attributeReport', cluster 'seMetering', data '{"instantaneousDemand":32916}' from endpoint 6 with groupID null
info  2022-10-22 12:34:48: MQTT publish: topic 'zigbee2mqtt/kitchen_fridge_electric_socket', payload '{"current":0.27,"energy":0.1,"linkquality":255,"power":32.916,"power_on_behavior":null,"state":"ON","voltage":233}'
debug 2022-10-22 12:34:53: Received Zigbee message from 'kitchen_fridge_electric_socket', type 'attributeReport', cluster 'seMetering', data '{"instantaneousDemand":32925}' from endpoint 6 with groupID null
info  2022-10-22 12:34:53: MQTT publish: topic 'zigbee2mqtt/kitchen_fridge_electric_socket', payload '{"current":0.27,"energy":0.1,"linkquality":255,"power":32.925,"power_on_behavior":null,"state":"ON","voltage":233}'
debug 2022-10-22 12:34:55: Received MQTT message on 'zigbee2mqtt/kitchen_fridge_electric_socket/set' with data '{"state":"OFF"}'
debug 2022-10-22 12:34:55: Publishing 'set' 'state' to 'kitchen_fridge_electric_socket'
debug 2022-10-22 12:34:56: Received MQTT message on 'zigbee2mqtt/kitchen_fridge_electric_socket/set' with data '{"state":"ON"}'
debug 2022-10-22 12:34:56: Publishing 'set' 'state' to 'kitchen_fridge_electric_socket'
debug 2022-10-22 12:34:57: Received MQTT message on 'zigbee2mqtt/kitchen_fridge_electric_socket/get' with data '{"state":""}'
debug 2022-10-22 12:34:57: Publishing get 'get' 'state' to 'kitchen_fridge_electric_socket'
debug 2022-10-22 12:34:58: Received Zigbee message from 'kitchen_fridge_electric_socket', type 'attributeReport', cluster 'seMetering', data '{"instantaneousDemand":32956}' from endpoint 6 with groupID null
info  2022-10-22 12:34:58: MQTT publish: topic 'zigbee2mqtt/kitchen_fridge_electric_socket', payload '{"current":0.27,"energy":0.1,"linkquality":255,"power":32.956,"power_on_behavior":null,"state":"ON","voltage":233}'
debug 2022-10-22 12:34:59: Received MQTT message on 'zigbee2mqtt/kitchen_fridge_electric_socket/set' with data '{"power_on_behavior":"on"}'
debug 2022-10-22 12:34:59: Publishing 'set' 'power_on_behavior' to 'kitchen_fridge_electric_socket'
debug 2022-10-22 12:35:01: Received MQTT message on 'zigbee2mqtt/kitchen_fridge_electric_socket/get' with data '{"power_on_behavior":""}'
debug 2022-10-22 12:35:01: Publishing get 'get' 'power_on_behavior' to 'kitchen_fridge_electric_socket'
debug 2022-10-22 12:35:03: Received Zigbee message from 'kitchen_fridge_electric_socket', type 'attributeReport', cluster 'seMetering', data '{"instantaneousDemand":32904}' from endpoint 6 with groupID null
info  2022-10-22 12:35:03: MQTT publish: topic 'zigbee2mqtt/kitchen_fridge_electric_socket', payload '{"current":0.27,"energy":0.1,"linkquality":255,"power":32.904,"power_on_behavior":null,"state":"ON","voltage":233}'
debug 2022-10-22 12:35:08: Received Zigbee message from 'kitchen_fridge_electric_socket', type 'attributeReport', cluster 'seMetering', data '{"instantaneousDemand":32620}' from endpoint 6 with groupID null
info  2022-10-22 12:35:08: MQTT publish: topic 'zigbee2mqtt/kitchen_fridge_electric_socket', payload '{"current":0.27,"energy":0.1,"linkquality":212,"power":32.62,"power_on_behavior":null,"state":"ON","voltage":233}'
debug 2022-10-22 12:35:13: Received Zigbee message from 'kitchen_fridge_electric_socket', type 'attributeReport', cluster 'seMetering', data '{"instantaneousDemand":32985}' from endpoint 6 with groupID null
info  2022-10-22 12:35:13: MQTT publish: topic 'zigbee2mqtt/kitchen_fridge_electric_socket', payload '{"current":0.27,"energy":0.1,"linkquality":255,"power":32.985,"power_on_behavior":null,"state":"ON","voltage":233}'
[... 49 lines cut ...]
[... cut ...]
info  2022-10-22 12:36:14: MQTT publish: topic 'zigbee2mqtt/kitchen_fridge_electric_socket', payload '{"current":0.27,"energy":0.1,"linkquality":255,"power":32.855,"power_on_behavior":null,"state":"ON","voltage":233}'
debug 2022-10-22 12:36:19: Received Zigbee message from 'kitchen_fridge_electric_socket', type 'attributeReport', cluster 'seMetering', data '{"instantaneousDemand":32887}' from endpoint 6 with groupID null
info  2022-10-22 12:36:19: MQTT publish: topic 'zigbee2mqtt/kitchen_fridge_electric_socket', payload '{"current":0.27,"energy":0.1,"linkquality":255,"power":32.887,"power_on_behavior":null,"state":"ON","voltage":233}'

joaquinvacas, 2022-11-01:
Does your device have any install code?

Mine does. And Z2M doesn't accept them.

therealeldaria, 2022-11-01:
> Does your device have any install code?

Not sure what an "Install code is" So I guess mine did not have any?

joaquinvacas, 2022-12-02:
Hey! Not staled. @Koenkk I think this may have the same issue as me. @therealeldaria Have you tried shutting down Zigbee2Mqtt for a while after joining the device? Try it. Solved my problem.

therealeldaria, 2022-12-02:
> Hey! Not staled. @Koenkk I think this may have the same issue as me. @therealeldaria Have you tried shutting down Zigbee2Mqtt for a while after joining the device? Try it. Solved my problem.

I just had a look, I have restarted many times since installing, also updated z2m since, And not sure what made it work, but I just tried to turn off and on the fridge, and it did respond to on/off. This time, I was also able to update the Power_on_behaviour.
I have however notice another issue with these plugs, but not sure if it is a z2m issue. I noticed it sometimes turns off and on for a short duration. From what I could find it did not show up in Home Assistant entity log that this happens. But I had one in use for my computers, and it happened a few times that it would power cycle. Not an issue for most appliances but I can not use it with my computers that are on 24/7

FSund, 2022-12-21:
Strange stuff. I was having the same issue, but stopping the Z2M addon in Home Assistant for a couple of minutes fixed it.

Firmware version 002.006.018 R
Model EKO09716

joaquinvacas, 2022-12-21:
> Strange stuff. I was having the same issue, but stopping the Z2M addon in Home Assistant for a couple of minutes fixed it.
> 
> Firmware version 002.006.018 R Model EKO09716

Yep. That's the only fix I've found.

Rittsel, 2023-01-14:
I can confirm with the same issue and workaround/solution on 3 out of 4 Schneider EKO09716 - FW 002.006.018 R
Only the first one I paired worked straight away, the rest had this behaviour until I stopped Z2M for 5-10min and then started it up again - all work straight away.

joaquinvacas, 2023-02-14:
Not stale. This issue isn't fixed and also device support isn't perfect, can't set previous mode and so on, 0001 ≠ 0002 model

joaquinvacas, 2023-03-18:
Not stale.

closed 2023-05-30 by github-actions[bot]