Koenkk/zigbee2mqtt #14591: [New device support]: Zigbee model 'TS0202' and manufacturer name '_TZ3040_6ygjfyll'

buddy31130, 2022-10-22:
### Link

https://www.aliexpress.com/item/1005004765555342.html?spm=a2g0o.order_list.0.0.4a6c18029IZHtB

### Database entry

{"id":34,"type":"EndDevice","ieeeAddr":"0xa4c1389c17adf387","nwkAddr":33486,"manufId":4417,"manufName":"_TZ3040_6ygjfyll","powerSource":"Battery","modelId":"TS0202","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":1026,"inClusterList":[1,3,4,1280,0],"outClusterList":[4,6,4096,25,10],"clusters":{"genBasic":{"attributes":{"stackVersion":0,"dateCode":""}},"ssIasZone":{"attributes":{"iasCieAddr":"0x00124b001cd4b69e","zoneState":1}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":72,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1666441701198,"defaultSendRequestWhen":"immediate"}

### Comments

Hi,

I've just acquired 4 of these new Human Motion Sensors on Aliexpress but they are not regognized by Z2M.
The product model is said to be IH012-RT01 and compatible with Tuya. Same as this one [https://www.zigbee2mqtt.io/devices/IH012-RT01.html#tuya-ih012-rt01](url).
I lack of knowledge to be able to make external converter working (I tried but I'm lost with the options) but am willing to be active on finding the solution.

Thanks :)

### External converter

```shell
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require("zigbee-herdsman-converters/lib/tuya");

const definition = {
    // Since a lot of Tuya devices use the same modelID, but use different data points
    // it's usually necessary to provide a fingerprint instead of a zigbeeModel
    fingerprint: [
        {
            // The model ID from: Device with modelID 'TS0601' is not supported
            // You may need to add \u0000 at the end of the name in some cases
            modelID: 'TS0202\u0000',
            // The manufacturer name from: Device with modelID 'TS0601' is not supported.
            manufacturerName: '_TZ3040_6ygjfyll'
        },
    ],
    model: 'IH012-RT01',
    vendor: 'CoRui',
    description: 'Human motion sensor',
    fromZigbee: [
        fz.ignore_basic_report, // Add this if you are getting no converter for 'genBasic'
        fz.tuya_data_point_dump, // This is a debug converter, it will be described in the next part
    ],
    toZigbee: [
        tz.tuya_data_point_test, // Another debug converter
    ],
    onEvent: tuya.onEventSetTime, // Add this if you are getting no converter for 'commandMcuSyncTime'
    configure: async (device, coordinatorEndpoint, logger) => {
        const endpoint = device.getEndpoint(1);
        await reporting.bind(endpoint, coordinatorEndpoint, ['genBasic']);
    },
    exposes: [e.occupancy(), e.battery_low(), e.tamper(), e.sensitivity(), e.keep_time(), e.linkquality()],
};

module.exports = definition;
```

Koenkk, 2022-10-22:
Added!

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2022-10-22 by Koenkk

buddy31130, 2022-10-23:
Super ! works like a charm ! Thank you very much !