Koenkk/zigbee2mqtt #14593: [New device support]: TRÄDFRI bulb E14 WS globe 470lm

brunocrvlh, 2022-10-22:
### Link

https://www.ikea.com/pt/en/p/tradfri-led-bulb-e14-470-lumen-smart-wireless-dimmable-white-spectrum-globe-70518179/

### Database entry

I don't know how to see that

### Comments

I just bought a new bulb from ikea but for some reason it shows up as "not supported". Tried to reset the bulb multiple times but nothing changed.

### Supported color modes

Adjustable white tones

Koenkk, 2022-10-23:
Can you make a screenshot of the z2m frontend -> device page? I need the modelID of this bulb.

brunocrvlh, 2022-10-23:
sure! this? 
[image]

Koenkk, 2022-10-24:
Should be supported in z2m 1.28.0

brunocrvlh, 2022-10-24:
Thanks! btw, is there any documentation on how to see my z2m version and how to upgrade it?

Koenkk, 2022-10-25:
Version is under frontend -> settings -> about.
Regarding upgrading it depends on how you run z2m. For e.g. bare metal installation see https://www.zigbee2mqtt.io/guide/installation/01_linux.html#for-later-update-zigbee2mqtt-to-the-latest-version

Since this device is already supported in the latest z2m I will close this.

closed 2022-10-25 by Koenkk