Koenkk/zigbee2mqtt #14598: Map takes forever to load, or doesn't load at all...

pauloon, 2022-10-22:
### What happened?

Map takes a looong time to load and most of the times does not finish loading at all.

[image]

I have only 43 devices.
Using HASS with Z2M and a Sonoff Zig USB adapter.
All updated.

Tryed many solutions, removed power reporting from most devices to "uncllog" the network but it does not appear to work much.
Tried cleaning the USB contacts also.

Can you help?

Thanks
Paulo.

### What did you expect to happen?

The map to show up correctly and in a short time.

### How to reproduce it (minimal and precise)

Start Z2M, go to the admin and request Map do load...

### Zigbee2MQTT version

1.28.0 commit: unknown

### Adapter firmware version

20220219

### Adapter

SONOFF Zigbee 3.0

### Debug log

[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/9845160/log.txt)

pauloon, 2022-11-22:
I know that this is not a priority, but I wouldn't like it to get lost....

mathieu-mp, 2022-11-23:
Hi, same here !
zigbee2mqtt HA addon 1.28.2-1

jmoore377, 2022-12-05:
Hi, I have the same.

```
info  2022-12-05 01:22:20: Logging to console and directory: '/app/data/log/2022-12-05.01-22-20' filename: log.txt
info  2022-12-05 01:22:20: Starting Zigbee2MQTT version 1.28.2 (commit #360a777)
info  2022-12-05 01:22:20: Starting zigbee-herdsman (0.14.68)

```
installed by docker-compose:
`services:
  zigbee2mqtt:
    container_name: zigbee2mqtt
    image: koenkk/zigbee2mqtt                                                                                                                                                                                      
`

closed 2023-01-12 by github-actions[bot]

DenisBY, 2023-03-09:
I have the same issue

merlinpimpim, 2023-11-03:
I now have this problem with zigbee2mqtt 1.33.2 (it wasn't the case before).

timmo65, 2024-01-25:
Hello, following this - mine has been fine for months - i'm on 1.35.1-1 using a sonoff dongle coordinator plugged into usb on a NUC running HA in proxmox.  I'm now getting this behaviour and need to restart  the zigbee2mqtt bridge every day as some devices become unresponsive and the map never loads (unless after a restart).  I am starting to look at usb extension lengths to move it further away from interference but I don't really know if that is the issue or if its the coordinator, any of the devices or something else. Wifi is via Eero which I dont think I can change channels on. Would just like a steer on what I should really be concentrating on first. Thanks

merlinpimpim, 2024-01-25:
If it's any help: when these problems occur, I've found that completely switching off (electrically) my system (Khadas + HA supervised + SkyConnect + Z2M), and not just rebooting it, means that it works for several days!

timmo65, 2024-01-26:
> If it's any help: when these problems occur, I've found that completely switching off (electrically) my system (Khadas + HA supervised + SkyConnect + Z2M), and not just rebooting it, means that it works for several days!

Thanks - I will try anything. I got a 2m usb2 extension (was 1m usb3) and also found a device (smart plug) not powered on at the wall switch in the network  - so enabled that. Will see - thanks for the response. I notice  a few new issues on here for things similar to this with problems with the zigbee2mqtt network.

merlinpimpim, 2024-01-26:
Having said that, we have to accept that not everything works in a stable manner if we don't use an adapter from the "Recommended" section (which is my case with the SkyConnect): 
https://www.zigbee2mqtt.io/guide/adapters/
On the other hand, it would be nice if adapters that have been in the "Experimental" section for a long time could one day be managed without any problems (time flies...)! 😁 @Koenkk

timmo65, 2024-02-08:
> Hello, following this - mine has been fine for months - i'm on 1.35.1-1 using a sonoff dongle coordinator plugged into usb on a NUC running HA in proxmox. I'm now getting this behaviour and need to [...]

An update on this but not 100% sure where the problem was as i have updated z2m a couple of times since this post and also replaced the Sonoff dongle E with a lan connected smlight  slzb06 (  I got the M  experimental version by mistake but it seems ok).
I flashed the sonoff with router firmware and now use that in the network. Actually doing this was a lot easier than i thought it would be. The map loads quickly now and more reliably and I am not having to restart the addon as devices are now not going offline. I think it is fixed so likely was the usb attached dongle causing interference.

merlinpimpim, 2024-02-08:
Personally, I think my issues stemmed from the Lixee device whose adventures are recounted here:
https://github.com/Koenkk/zigbee2mqtt/issues/21198

lperco, 2024-08-21:
> I have the same issue
> 
> UPDATE: it just takes much longer than "2 minutes". Especially when you have "dead" devices.

same here. I've IKEA light that, some times, I manualy switch off, cut AC power, with traditional switch so it became "dead" and generates error on z2m and map doesen't load.

dajomas, 2024-08-28:
It seems indeed that unreachable elements are the culprit here. I have a 7 unreachable elements and it took over half an hour to build the map (I have 86 elements, including the 7 unreachable ones)

drueppela, 2024-10-30:
After I activated "Availability (advanced)" Zigbee2MQTT marks offline devices. GREAT!

Since "Availability (advanced)" is activated the map doesn't load anymore.

@Koenkk: The issue should be reopened. Many thanks.

mathieu-mp, 2024-10-30:
> After I activated "Availability (advanced)" Zigbee2MQTT marks offline devices. GREAT!
> 
> Since "Availability (advanced)" is activated the map doesn't load anymore.

@drueppela, i'm sorry but I did not understand whether this settings "Zigbee2MQTT > Settings > Availability > Availability (Advanced)" solved the map loading issue for you... or not?

merlinpimpim, 2024-10-30:
In addition, a UI problem: if a timeout is set to advanced mode and you click on Availability (simple) then click again on Availability (advanced) (without submitting), the timeout value reverts to the default value (1500 for passive for example), but only visually.

tackin, 2024-11-28:
Found the problem:
If I set docker-compose-file
```
environment:
 - Z2M_WATCHDOG=default

```

The watchdog will kill requests to devices which are not available yet. So the timeout will never reach the card-building processing.
Disable that environment-var solves the card-problem for me immediately.

honzis1364, 2025-01-26:
I can confirm, that if Availability is enabled, Map won't load. Tried on my network with 150 devices. Otherwise, it will load on like 3 minutes.

huntyh, 2025-04-08:
I have availability off and map still doesn't load. I have like 6 devices only.