Koenkk/zigbee2mqtt #14599: Cannot check OTA updates for TuYa SmartPlug (TS011F_plug_3)

vertisan, 2022-10-22:
### What happened?

Unfortunately, I cannot check for the OTA updates due to an error:  
```
Failed to check if update available for 'SmartPlug PC Desk' (Failed to find endpoint which support OTA cluster)
```
Relating to the docs, this SmartPlug has available OTA updates: https://www.zigbee2mqtt.io/devices/TS011F_plug_3.html

### What did you expect to happen?

The OTA updates can be checked and eventually updated.

### How to reproduce it (minimal and precise)

Go to Frontend > OTA > select one of the devices and click "Check for new updates"

### Zigbee2MQTT version

1.28.0-dev commit: 599380d

### Adapter firmware version

20210708

### Adapter

Sonoff Zigbee 3.0 USB Dongle Plus

### Debug log

```
Debug 2022-10-23 00:04:17Received MQTT message on 'zigbee2mqtt/bridge/request/device/ota_update/check' with data '{"id":"SmartPlug PC Desk","transaction":"4cdyy-1"}'
Info 2022-10-23 00:04:17Checking if update available for 'SmartPlug PC Desk'
Debug 2022-10-23 00:04:17Check if update available for '0x70b3d52b60014412' (TS011F)
Info 2022-10-23 00:04:17MQTT publish: topic 'zigbee2mqtt/bridge/response/device/ota_update/check', payload '{"data":{"id":"SmartPlug PC Desk"},"error":"Failed to check if update available for 'SmartPlug PC Desk' (Failed to find endpoint which support OTA cluster)","status":"error","transaction":"4cdyy-1"}'
Error 2022-10-23 00:04:17Failed to check if update available for 'SmartPlug PC Desk' (Failed to find endpoint which support OTA cluster)
Debug 2022-10-23 00:04:17AssertionError [ERR_ASSERTION]: Failed to find endpoint which support OTA cluster at Object.isUpdateAvailable (/app/node_modules/zigbee-herdsman-converters/lib/ota/common.js:248:9) at Object.isUpdateAvailable (/app/node_modules/zigbee-herdsman-converters/lib/ota/zigbeeOTA.js:130:19) at OTAUpdate.onMQTTMessage (/app/lib/extension/otaUpdate.ts:194:67) at EventEmitter.emit (node:events:525:35) at EventBus.emitMQTTMessage (/app/lib/eventBus.ts:109:22) at MQTT.onMessage (/app/lib/mqtt.ts:140:27) at WebSocket.<anonymous> (/app/lib/extension/frontend.ts:94:27) at WebSocket.emit (node:events:513:28) at Receiver.receiverOnMessage (/app/node_modules/ws/lib/websocket.js:1178:20) at Receiver.emit (node:events:513:28)
```

danieledwardgeorgehitchcock, 2022-10-24:
Could you post up the database.db entry for this device?

The error is leading me to believe that Z2M cannot find an endpoint on the device capable of updates so, I want to see if there is one..

vertisan, 2022-10-24:
Sure:

```json
{
  "id":5,
  "type":"Router",
  "ieeeAddr":"0x70b3d52b600141b2",
  "nwkAddr":26522,
  "manufId":4660,
  "manufName":"_TZ3000_r6buo8ba",
  "powerSource":"Mains (single phase)",
  "modelId":"TS011F",
  "epList":[
    1
  ],
  "endpoints":{
    "1":{
      "profId":260,
      "epId":1,
      "devId":81,
      "inClusterList":[
        0,
        6,
        3,
        4,
        5,
        57345,
        2820,
        1794
      ],
      "outClusterList":[
        
      ],
      "clusters":{
        "genBasic":{
          "attributes":{
            "modelId":"TS011F",
            "manufacturerName":"_TZ3000_r6buo8ba",
            "powerSource":1,
            "zclVersion":2,
            "appVersion":160,
            "stackVersion":1,
            "hwVersion":3,
            "dateCode":"20210625",
            "swBuildId":"20B+TZSKT41BS101"
          }
        },
        "haElectricalMeasurement":{
          "attributes":{
            "acCurrentDivisor":1000,
            "acCurrentMultiplier":1,
            "rmsVoltage":236,
            "rmsCurrent":26,
            "activePower":1
          }
        },
        "seMetering":{
          "attributes":{
            "divisor":100,
            "multiplier":1
          }
        },
        "genOnOff":{
          "attributes":{
            "32768":0,
            "onOff":1,
            "moesStartUpOnOff":0,
            "tuyaBacklightMode":255
          }
        }
      },
      "binds":[
        
      ],
      "configuredReportings":[
        
      ],
      "meta":{
        
      }
    }
  },
  "appVersion":160,
  "stackVersion":1,
  "hwVersion":3,
  "dateCode":"20210625",
  "swBuildId":"20B+TZSKT41BS101",
  "zclVersion":2,
  "interviewCompleted":true,
  "meta":{
    "configured":-522437996
  },
  "lastSeen":1666601985888,
  "defaultSendRequestWhen":"immediate"
}
```

danieledwardgeorgehitchcock, 2022-10-24:
Your device does not support OTA updating - if it did, you'd see it in the `outClusterList`

vertisan, 2022-10-24:
Ok, so it would need to be corrected in the documentation because it currently indicates that they are available.
https://www.zigbee2mqtt.io/devices/TS011F_plug_3.html#ota-updates

danieledwardgeorgehitchcock, 2022-10-24:
Yes, and also the converter for the device too

vertisan, 2022-10-24:
@danieledwardgeorgehitchcock 
Just a little offtop - do you have some solution for missing energy reporting for this plug? I've checked already some of the solutions and converters but none of them work.

closed 2022-12-01 by github-actions[bot]

MvdB1982, 2022-12-04:
Since 1.28.4 Home Assistant shows "update available".
However, zig2mqtt logs shows "_Update of '0x.....' failed (Failed to find endpoint which support OTA cluster)_"

Since my OCD can't handle a "yellow 1" for update icon - is there any way to solve this issue?

HabelStefan, 2022-12-21:
Is there a way to update this device if not OTA?

Aresitoo, 2023-03-27:
> Is there a way to update this device if not OTA?

I'd like to know as well. Can only think of adding it to Tuya see if there is an update and then bind again..