Koenkk/zigbee2mqtt #14607: Zemismart ZM-CSW032-D - No state update after upgrade to 1.27.2-1, problem remains in 1.28.0-1

Gebruikertje, 2022-10-23:
### What happened?

The state of the cover does not change in Home Assistant after upgrade from 1.27.0-1 to 1.27.2-1 or 1.28.0-1. When Home Assistant is restarted the state will be unknown. When I look the log I see in 1.27.0-1 the following:

info 2022-10-23 10:38:18MQTT publish: topic 'zigbee2mqtt/WK_Rolluik_Achter', payload '{"last_seen":"2022-10-23T10:38:17+02:00","linkquality":43,"position":100}'

After upgrade, also the "state" is published to MQTT:

info 2022-10-23 10:22:10MQTT publish: topic 'zigbee2mqtt/WK_Rolluik_Achter', payload '{"last_seen":"2022-10-23T10:22:08+02:00","linkquality":142,"position":100,"state":null}'

A rollback to 1.27.0-1 and restart of HA will fix the problem.

### Zigbee2MQTT version

1.27.2-1 and 1.28.0-1

### Adapter firmware version

20220219

### Adapter

Sonoff Zigbee 3.0 USB Dongle Plus (ZBDongle-P)

Hydci, 2022-11-07:
I also have this same problem

Hydci, 2022-12-08:
@Koenkk the problem and still relevant if it is possible to look thank you

Hydci, 2022-12-08:
@Gebruikertje yes i tried the latest version but it doesn't change anything

Koenkk, 2022-12-08:
Could you check if the issue is fixed with the following external converter: https://gist.github.com/Koenkk/2835843081f821e0d01d5a892c5be3cf

- save this as file next to `configuration.yaml` as `ext_converter.js`
- add it to `configuration.yaml`:
```yaml
external_converters:
  - ext_converter.js
```
- start z2m, check if issue is fixed

Hydci, 2022-12-08:
after trying the file the error however disappears in the history the position and reverse when the pane and opened it and declared close and closed and declared open

Koenkk, 2022-12-08:
Updated https://gist.github.com/Koenkk/2835843081f821e0d01d5a892c5be3cf, please try again.

Hydci, 2022-12-08:
this does not change the state in the history and always reverse

Koenkk, 2022-12-08:
Updated https://gist.github.com/Koenkk/2835843081f821e0d01d5a892c5be3cf again.

Gebruikertje, 2022-12-08:
I took the 2nd file and added the 2 lines to the configuration.yaml. invert_cover is false. I closed the cover and get the following line in the log:

info  2022-12-08 18:31:37: MQTT publish: topic 'zigbee2mqtt/WK_Rolluik_Achter', payload '{"last_seen":"2022-12-08T18:31:37+01:00","linkquality":32,"position":0,"state":"OPEN"}'

Then I opened the cover and these 2 lines are written to the log:

info  2022-12-08 18:36:25: MQTT publish: topic 'zigbee2mqtt/WK_Rolluik_Voor', payload '{"last_seen":"2022-12-08T18:36:25+01:00","linkquality":32,"position":0,"state":null}'
info  2022-12-08 18:36:38: MQTT publish: topic 'zigbee2mqtt/WK_Rolluik_Achter', payload '{"last_seen":"2022-12-08T18:36:38+01:00","linkquality":120,"position":100,"state":"CLOSE"}'

I now will try the 3rd file

Hydci, 2022-12-08:
The last test works perfectly the states goes back perfectly thank you

Gebruikertje, 2022-12-08:
Now tested the 3rd file and opened the cover:

info  2022-12-08 18:44:46: MQTT publish: topic 'zigbee2mqtt/WK_Rolluik_Achter', payload '{"last_seen":"2022-12-08T18:44:46+01:00","linkquality":142,"position":-1,"state":"OPEN"}'

And then closed it.

info  2022-12-08 18:45:47: MQTT publish: topic 'zigbee2mqtt/WK_Rolluik_Achter', payload '{"last_seen":"2022-12-08T18:45:47+01:00","linkquality":32,"position":0,"state":"CLOSE"}'

And setting it to a specific position works also:

info  2022-12-08 18:48:08: MQTT publish: topic 'zigbee2mqtt/WK_Rolluik_Achter', payload '{"last_seen":"2022-12-08T18:48:08+01:00","linkquality":29,"position":53,"state":"OPEN"}'

Thank you - Bedankt

Gebruikertje, 2022-12-08:
In the previous post, I saw that the position was -1 when the cover was open. Now after opening the cover completely I see:

info  2022-12-08 18:56:26: MQTT publish: topic 'zigbee2mqtt/WK_Rolluik_Achter', payload '{"last_seen":"2022-12-08T18:56:26+01:00","linkquality":65,"position":100,"state":"OPEN"}'

So the position seems correct now.

Koenkk, 2022-12-08:
Pushed the fix! You can keep using the external converter until the next release.

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2022-12-08 by Koenkk