Koenkk/zigbee2mqtt #14612: Perenio PECLS01 shows 200% battery instead of 100%

yndtrud, 2022-10-23:
### What happened?

zigbee2mqtt: **1.28.0-dev**
sensore: **Perenio PECLS01**
Zigbee-OTA FW: **LeakSensor_v5.OTA.zigbee**

I noticed that a few days ago, updates for Perenio devices were added to the OTA repository. I really wanted to update the leak sensors, as they did not correctly show the battery status. Always 100%, even when the battery was 2.6V instead of 3V and, in fact, the sensor did not respond to contact closure. Perenio support recommended updating the firmware, but I don't have a hub.

I used the **dev v1.28.0-dev (21oct)** version of the program and updated the sensors (release 1.28.0 cannot update perenio). All sensors have been updated, but now the battery status is 200% at full battery and 96% at 2.6V. How do I fix to 100% with a full battery?

[image]

### What did you expect to happen?

correct battery level

### How to reproduce it (minimal and precise)

Update sensor with latest firmware from Zigbee2mqtt

### Zigbee2MQTT version

1.28.0.dev

### Adapter firmware version

20220726

### Adapter

Zigbee 3.0 USB Dongle Plus (P version)

Koenkk, 2022-10-24:
Fixed! after updating to the latest dev you need to wait until a new battery % report is send by the device, then it will show the correct value.

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2022-10-24 by Koenkk

mocksoul, 2024-03-09:
i have the opposite now.
50% for fully charged battery (2 devices tested, 3.28V both batteries)