Koenkk/zigbee2mqtt #14622: Zigbee2MQTT Start Error: No path provided and failed to auto detect pat

AarneAarebye, 2022-10-23:
### What happened?

- Raspi bullseye
- ConBee 2
- udev is installed
- /opt/zigbee2mqtt/data/configuration.yaml
homeassistant: false
permit_join: true
mqtt:
  base_topic: zigbee2mqtt
  server: mqtt://192.168.178.78
  user: <myuser>
  password: <mypasss>
serial:
  adapter: deconz
frontend: false
advanced:
  network_key: <has been generated>

- console output:
admin@node-000:/opt/zigbee2mqtt $ npm start

> zigbee2mqtt@1.28.0 start /opt/zigbee2mqtt
> node index.js

Zigbee2MQTT:info  2022-10-23 23:21:16: Logging to console and directory: '/opt/zigbee2mqtt/data/log/2022-10-23.23-21-16' filename: log.txt
Zigbee2MQTT:info  2022-10-23 23:21:16: Starting Zigbee2MQTT version 1.28.0 (commit #03ba647)
Zigbee2MQTT:info  2022-10-23 23:21:16: Starting zigbee-herdsman (0.14.62)
Zigbee2MQTT:error 2022-10-23 23:21:16: Error while starting zigbee-herdsman
Zigbee2MQTT:error 2022-10-23 23:21:16: Failed to start zigbee
Zigbee2MQTT:error 2022-10-23 23:21:16: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start.html for possible solutions
Zigbee2MQTT:error 2022-10-23 23:21:16: Exiting...
Zigbee2MQTT:error 2022-10-23 23:21:17: Error: No path provided and failed to auto detect path
    at Function.create (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/adapter/adapter.ts:81:23)
    at processTicksAndRejections (internal/process/task_queues.js:95:5)
    at Controller.start (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/controller/controller.ts:129:24)
    at Zigbee.start (/opt/zigbee2mqtt/lib/zigbee.ts:58:27)
    at Controller.start (/opt/zigbee2mqtt/lib/controller.ts:101:27)
    at start (/opt/zigbee2mqtt/index.js:109:5)
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! zigbee2mqtt@1.28.0 start: `node index.js`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the zigbee2mqtt@1.28.0 start script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     /home/admin/.npm/_logs/2022-10-23T21_21_17_081Z-debug.log

### What did you expect to happen?

just starting ...

### Zigbee2MQTT version

1.28.0 (commit #03ba647)

### Adapter firmware version

latest

### Adapter

ConBee 2

### Debug log

dmin@node-000:/opt/zigbee2mqtt $ EBUG=zigbee-herdsman* npm start

> zigbee2mqtt@1.28.0 start /opt/zigbee2mqtt
> node index.js

internal/modules/cjs/loader.js:905
  throw err;
  ^

**Error: Cannot find module 'semver'**
Require stack:
- /opt/zigbee2mqtt/index.js
    at Function.Module._resolveFilename (internal/modules/cjs/loader.js:902:15)
    at Function.Module._load (internal/modules/cjs/loader.js:746:27)
    at Module.require (internal/modules/cjs/loader.js:974:19)
    at require (internal/modules/cjs/helpers.js:101:18)
    at Object.<anonymous> (/opt/zigbee2mqtt/index.js:3:16)
    at Module._compile (internal/modules/cjs/loader.js:1085:14)
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:1114:10)
    at Module.load (internal/modules/cjs/loader.js:950:32)
    at Function.Module._load (internal/modules/cjs/loader.js:790:12)
    at Function.executeUserEntryPoint [as runMain] (internal/modules/run_main.js:75:12) {
  code: 'MODULE_NOT_FOUND',
  requireStack: [ '/opt/zigbee2mqtt/index.js' ]
}
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! zigbee2mqtt@1.28.0 start: `node index.js`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the zigbee2mqtt@1.28.0 start script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     /home/admin/.npm/_logs/2022-10-23T21_37_12_999Z-debug.log

AarneAarebye, 2022-10-23:
Adapter is RaspBee II

AarneAarebye, 2022-10-26:
On my raspi, no /dev/ttyAMA0 is available. Also, no /dev/serial/by-id/* is working, as there is no "by-id". The only serial device, besides /dev/ttyS0 is /dev/ttyAMA0.

```
admin@node-000:/opt/zigbee2mqtt $ udevadm info -a -n /dev/ttyAMA0 | grep 'serial'
  looking at device '/devices/platform/soc/fe201000.serial/tty/ttyAMA0':
  looking at parent device '/devices/platform/soc/fe201000.serial':
    KERNELS=="fe201000.serial"

```
So, I added /dev/ttyAMA0 as port in the yaml config file.
```
homeassistant: false
permit_join: true
mqtt:
  base_topic: zigbee2mqtt
  server: mqtt://192.168.178.78
  user: XXX
  password: XXX
serial:
  port: /dev/ttyAMA0
  adapter: deconz
frontend: false
log_level: debug
advanced:
  network_key: XXXX
```

AarneAarebye, 2022-10-26:
`admin@node-000:/opt/zigbee2mqtt $ DEBUG=zigbee-herdsman* npm start

> zigbee2mqtt@1.28.0 start /opt/zigbee2mqtt
> node index.js

internal/modules/cjs/loader.js:905
  throw err;
  ^

Error: Cannot find module 'semver'
Require stack:
- /opt/zigbee2mqtt/index.js
    at Function.Module._resolveFilename (internal/modules/cjs/loader.js:902:15)
    at Function.Module._load (internal/modules/cjs/loader.js:746:27)
    at Module.require (internal/modules/cjs/loader.js:974:19)
    at require (internal/modules/cjs/helpers.js:101:18)
    at Object.<anonymous> (/opt/zigbee2mqtt/index.js:3:16)
    at Module._compile (internal/modules/cjs/loader.js:1085:14)
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:1114:10)
    at Module.load (internal/modules/cjs/loader.js:950:32)
    at Function.Module._load (internal/modules/cjs/loader.js:790:12)
    at Function.executeUserEntryPoint [as runMain] (internal/modules/run_main.js:75:12) {
  code: 'MODULE_NOT_FOUND',
  requireStack: [ '/opt/zigbee2mqtt/index.js' ]
}
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! zigbee2mqtt@1.28.0 start: `node index.js`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the zigbee2mqtt@1.28.0 start script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     /home/admin/.npm/_logs/2022-10-26T10_17_09_139Z-debug.log
`

AarneAarebye, 2022-10-26:
so, as you can see, I did some research, update and another npm ci... Need some hint in which direction to go further to analyze the root case...

AarneAarebye, 2022-10-29:
OK, found solution: deConz service was still up and running. Now working fine.

closed 2022-10-29 by AarneAarebye

wavedeluxmaster, 2023-12-28:
I have also an RaspBee II and using the  snoff zigbee 3.0 USB Dongle Plus and got this error,
`
    Zigbee2MQTT:info  2023-12-28 21:55:13: Starting Zigbee2MQTT version 1.34.0 (commit #aae7312)
    Zigbee2MQTT:info  2023-12-28 21:55:13: Starting zigbee-herdsman (0.25.0)
    
    Zigbee2MQTT:error 2023-12-28 21:55:14: Error while starting zigbee-herdsman
    
    Zigbee2MQTT:error 2023-12-28 21:55:14: Failed to start zigbee
    
    Zigbee2MQTT:error 2023-12-28 21:55:14: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start.html for possible solutions
    
    Zigbee2MQTT:error 2023-12-28 21:55:14: Exiting...
    
    Zigbee2MQTT:error 2023-12-28 21:55:14: Error: Error while opening serialport 'Error: Error: No such file or directory, cannot open /dev/ttyAMA0'
    
    at SerialPort.<anonymous> (/app/node_modules/zigbee-herdsman/src/adapter/z-stack/znp/znp.ts:146:28)
    
    at SerialPort._error (/app/node_modules/@serialport/stream/dist/index.js:82:22)
    
    at /app/node_modules/@serialport/stream/dist/index.js:118:18

`