Koenkk/zigbee2mqtt #14623: backlight_mode not responding correctly

lethalwallabies, 2022-10-24:
### What happened?

Rebranded Tuya TS0013 (https://www.zemismart.com/products/tb21-zb-1) wall switches do not respond when changing the backlight LED behavior.

Zigbee2MQTT reports success when changing the state but no change is effective on the device.

### What did you expect to happen?

Changing LED backlight status to LOW would disable the LED on the switch.

### How to reproduce it (minimal and precise)

Pair Device
Add custom code
`const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const ota = require('zigbee-herdsman-converters/lib/ota');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    zigbeeModel: ['TS0013'],
    model: 'TS0013',
    vendor: 'Moes',
    description: 'Smart light switch - 3 gang without neutral wire',
    extend: extend.switch(),
    toZigbee: extend.switch().toZigbee.concat([tz.moes_power_on_behavior, tz.tuya_backlight_mode]),
    fromZigbee: extend.switch().fromZigbee.concat([tz.moes_power_on_behavior, tz.tuya_backlight_mode]),
    exposes: [e.switch().withEndpoint('left'), e.switch().withEndpoint('center'), e.switch().withEndpoint('right'), 
	exposes.enum('power_on_behavior', ea.ALL, Object.values(tuya.moesSwitch.powerOnBehavior)),
        exposes.enum('backlight_mode', ea.ALL, ['LOW', 'MEDIUM', 'HIGH']).withDescription('Indicator light status: LOW: Off | MEDIUM: On| HIGH: Inverted')],
    endpoint: (device) => {
        return {'left': 1, 'center': 2, 'right': 3};
    },
    whiteLabel: [{vendor: 'TUYATEC', model: 'GDKES-03TZXD'}],
    meta: {multiEndpoint: true},
    configure: async (device, coordinatorEndpoint, logger) => {
        await device.getEndpoint(1).read('genBasic', ['manufacturerName', 'zclVersion', 'appVersion', 'modelId', 'powerSource', 0xfffe]);
        try {
            for (const ID of [1, 2, 3]) {
                const endpoint = device.getEndpoint(ID);
                await reporting.bind(endpoint, coordinatorEndpoint, ['genOnOff']);
            }
        } catch (e) {
            // Fails for some: https://github.com/Koenkk/zigbee2mqtt/issues/4872
        }
        device.powerSource = 'Mains (single phase)';
        device.save();
    },

};

module.exports = definition;`

Update backlight setting through Zigbee2MQTT or HA

### Zigbee2MQTT version

1.28.0-1

### Adapter firmware version

0x26720700

### Adapter

Decon Conbee 2

closed 2022-12-01 by github-actions[bot]

osnwt, 2022-12-21:
Same here. Shouldn't it be reopen?

devkaiwang, 2025-03-02:
I met this issue also, what's wrong with this project?