Koenkk/zigbee2mqtt #14628: QS-Zigbee-D02-TRIAC-2C-L Dimmer switching works, but dimming not

FizykPSX, 2022-10-24:
### What happened?

I have added [QS-Zigbee-D02-TRIAC-2C-L] dimmer to zigbee2mqtt without any issues, but it appears that dimming function doesn't work, either from zigbee2mqtt web interface nor Home Assistant. It does work from physical interface though.
I do not intend to use it with physical switch (I use scene switcher) and dim function should be triggered by automations.

Issue was tested on both stable and dev branches of docker image.

### What did you expect to happen?

I would expect to be able to adjust brightness from Home Assistant/Zigbee2mqtt web interface

### How to reproduce it (minimal and precise)

Add dimmer to zigbee2mqtt and try to dim or brighten bulbs

### Zigbee2MQTT version

 1.28.0-dev commit: 08ceae2

### Adapter firmware version

20220219

### Adapter

Sonof Zigbee USB 3.0 Dongle 

### Debug log

No errors are produced, logs shows normal operation:

2022-10-24 13:49:21LQI succeeded for 'Ściemniacz Kuchnia'
2022-10-24 13:50:47Received MQTT message on 'zigbee2mqtt/Ściemniacz Kuchnia/set' with data '{"brightness_l1":65}'
2022-10-24 13:50:47Publishing 'set' 'brightness' to 'Ściemniacz Kuchnia'

Yari117, 2022-11-30:
I would like to bump up this topic!
I also have a QS-Zigbee-D02-TRIAC-2C-L dimmer, recognized as _TZ3210_4ubylghk
It pairs with z2m perfectly, but I cannot control the dim level with the slider. Physical buttons on the dimmer do work correctly.

Yari117, 2022-12-01:
Updating z2m to 1.28.3 did not help....
The brightnesss value changes when I move the slider, but actual brightness does not.
Changing the brightness by physical button changes it but does not lead to any published events. Turning on and off by a physical button leads to posting the state change. The brightnesss value in the published message does not correspond to the actual value.

Here is how pairing looks like:

Info 2022-12-01 22:12:39MQTT publish: topic 'homeassistant/light/0xa4c138b3ab6f45cc/light_l2/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state"}],"brightness":true,"brightness_scale":254,"command_topic":"zigbee2mqtt/0xa4c138b3ab6f45cc/l2/set","device":{"identifiers":["zigbee2mqtt_0xa4c138b3ab6f45cc"],"manufacturer":"Lonsonho","model":"2 gang smart dimmer switch module without neutral (QS-Zigbee-D02-TRIAC-2C-)","name":"0xa4c138b3ab6f45cc"},"json_attributes_topic":"zigbee2mqtt/0xa4c138b3ab6f45cc/l2","name":"0xa4c138b3ab6f45cc l2","schema":"json","state_topic":"zigbee2mqtt/0xa4c138b3ab6f45cc/l2","unique_id":"0xa4c138b3ab6f45cc_light_l2_zigbee2mqtt"}'

Info 2022-12-01 22:12:39MQTT publish: topic 'homeassistant/sensor/0xa4c138b3ab6f45cc/linkquality/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state"}],"device":{"identifiers":["zigbee2mqtt_0xa4c138b3ab6f45cc"],"manufacturer":"Lonsonho","model":"2 gang smart dimmer switch module without neutral (QS-Zigbee-D02-TRIAC-2C-L)","name":"0xa4c138b3ab6f45cc"},"enabled_by_default":false,"entity_category":"diagnostic","icon":"mdi:signal","json_attributes_topic":"zigbee2mqtt/0xa4c138b3ab6f45cc","name":"0xa4c138b3ab6f45cc linkquality","state_class":"measurement","state_topic":"zigbee2mqtt/0xa4c138b3ab6f45cc","unique_id":"0xa4c138b3ab6f45cc_linkquality_zigbee2mqtt","unit_of_measurement":"lqi","value_template":"{{ value_json.linkquality }}"}'

Info 2022-12-01 22:12:39Starting interview of '0xa4c138b3ab6f45cc'

Info 2022-12-01 22:12:40MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0xa4c138b3ab6f45cc","ieee_address":"0xa4c138b3ab6f45cc","status":"started"},"type":"device_interview"}'

Info 2022-12-01 22:12:40MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_started","meta":{"friendly_name":"0xa4c138b3ab6f45cc"},"type":"pairing"}'

Info 2022-12-01 22:12:40MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0xa4c138b3ab6f45cc","ieee_address":"0xa4c138b3ab6f45cc"},"type":"device_announce"}'
Info 2022-12-01 22:12:40MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"announce","meta":{"friendly_name":"0xa4c138b3ab6f45cc"},"type":"device_announced"}'

Info 2022-12-01 22:12:40Successfully interviewed '0xa4c138b3ab6f45cc', device has successfully been paired

Info 2022-12-01 22:12:40Device '0xa4c138b3ab6f45cc' is supported, identified as: Lonsonho 2 gang smart dimmer switch module without neutral (QS-Zigbee-D02-TRIAC-2C-L)

Info 2022-12-01 22:12:41MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"definition":{"description":"2 gang smart dimmer switch module without neutral","exposes":[{"endpoint":"l1","features":[{"access":7,"description":"On/off state of this light","endpoint":"l1","name":"state","property":"state_l1","type":"binary","value_off":"OFF","value_on":"ON","value_toggle":"TOGGLE"},{"access":7,"description":"Brightness of this light","endpoint":"l1","name":"brightness","property":"brightness_l1","type":"numeric","value_max":254,"value_min":0}],"type":"light"},{"endpoint":"l2","features":[{"access":7,"description":"On/off state of this light","endpoint":"l2","name":"state","property":"state_l2","type":"binary","value_off":"OFF","value_on":"ON","value_toggle":"TOGGLE"},{"access":7,"description":"Brightness of this light","endpoint":"l2","name":"brightness","property":"brightness_l2","type":"numeric","value_max":254,"value_min":0}],"type":"light"},{"access":1,"description":"Link quality (signal strength)","name":"linkquality","property":"linkquality","type":"numeric","unit":"lqi","value_max":255,"value_min":0}],"model":"QS-Zigbee-D02-TRIAC-2C-L","options":[{"access":2,"description":"Controls the transition time (in seconds) of on/off, brightness, color temperature (if applicable) and color (if applicable) changes. Defaults to `0` (no transition).","name":"transition","property":"transition","type":"numeric","value_min":0}],"supports_ota":false,"vendor":"Lonsonho"},"friendly_name":"0xa4c138b3ab6f45cc","ieee_address":"0xa4c138b3ab6f45cc","status":"successful","supported":true},"type":"device_interview"}'

Info 2022-12-01 22:12:41Configuring '0xa4c138b3ab6f45cc'

Info 2022-12-01 22:12:41MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_successful","meta":{"description":"2 gang smart dimmer switch module without neutral","friendly_name":"0xa4c138b3ab6f45cc","model":"QS-Zigbee-D02-TRIAC-2C-L","supported":true,"vendor":"Lonsonho"},"type":"pairing"}'

Info 2022-12-01 22:12:42Successfully configured '0xa4c138b3ab6f45cc'

Koenkk, 2022-12-02:
@Yari117  
Could you check if the issue is fixed with the following external converter: https://gist.github.com/Koenkk/0316c10873dcc48905c6e8f63fa4b09c
- save this as file next to `configuration.yaml` as `ext_converter.js`
- add it to `configuration.yaml`:
```yaml
external_converters:
  - ext_converter.js
```
- start z2m, check if issue is fixed

Yari117, 2022-12-03:
Thank you very much for the quick response!
I installed the external converter and paired the dimmer once again. New attributes appeared like max and min brightness, so the converter is definetly taking over. 
Yet, the brightness slider does not work still! The issue is not fixed....

Koenkk, 2022-12-04:
Does dimming of the first channel work with this converter?

```js
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const ota = require('zigbee-herdsman-converters/lib/ota');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const utils = require('zigbee-herdsman-converters/lib/utils');
const globalStore = require('zigbee-herdsman-converters/lib/store');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    fingerprint: tuya.fingerprint('TS110E', ['_TZ3210_4ubylghk']),
    model: 'TS110E_dimmer_1',
    vendor: 'TuYa',
    description: '1 gang smart dimmer',
    fromZigbee: [tuya.fz.datapoints],
    toZigbee: [tuya.tz.datapoints],
    configure: tuya.configureMagicPacket,
    exposes: [tuya.exposes.lightBrightnessWithMinMax(), e.power_on_behavior(),
        tuya.exposes.countdown(), tuya.exposes.lightType()],
    meta: {
        tuyaDatapoints: [
            [1, 'state', tuya.valueConverter.onOff, {skip: tuya.skip.stateOnAndBrightnessPresent}],
            [2, 'brightness', tuya.valueConverter.scale0_254to0_1000],
            [3, 'min_brightness', tuya.valueConverter.scale0_254to0_1000],
            [4, 'light_type', tuya.valueConverter.lightType],
            [5, 'max_brightness', tuya.valueConverter.scale0_254to0_1000],
            [6, 'countdown', tuya.valueConverter.countdown],
            [14, 'power_on_behavior', tuya.valueConverter.powerOnBehavior],
        ],
    },
};

module.exports = definition;

```

If this doesn't work this device uses a (yet another) different TuYa api. To reverse engineer this you need to sniff the traffic when setting the brightness via the TuYa gateway: https://www.zigbee2mqtt.io/advanced/zigbee/04_sniff_zigbee_traffic.html

Yari117, 2022-12-04:
Ok, maybe I've done this the wrong way. I have copied the code above to a new file, named it "ext_converter2.js", put it in the z2m folder and pasted a line in the configuration.yaml (I deleted the previous ext_converter.js from there).

After this, z2m fails to start with the following log:

[00:48:38] INFO: Preparing to start...
[00:48:38] INFO: Socat not enabled
[00:48:40] INFO: Starting Zigbee2MQTT...
/app/dist/util/externally-loaded.js:18
    fromZigbee: [tuya.fz.datapoints],
                         ^
TypeError: Cannot read properties of undefined (reading 'datapoints')
    at /app/dist/util/externally-loaded.js:18:26
    at Script.runInContext (node:vm:141:12)
    at Script.runInNewContext (node:vm:146:17)
    at Object.runInNewContext (node:vm:306:38)
    at loadModuleFromText (/app/lib/util/utils.ts:148:8)
    at loadModuleFromFile (/app/lib/util/utils.ts:155:12)
    at Object.getExternalConvertersDefinitions (/app/lib/util/utils.ts:165:25)
    at getExternalConvertersDefinitions.next (<anonymous>)
    at new ExternalConverters (/app/lib/extension/externalConverters.ts:12:20)
    at new Controller (/app/lib/controller.ts:84:58)

Koenkk, 2022-12-06:
Make sure you are on z2m 1.28.4.

FizykPSX, 2022-12-07:
> OK, sorry for that..... I updated to 1.28.4-1 and the converter kicked in Still, with this converter I was not able not only to change the dim level but also to turn the light on and off. All controls are disfuctional with this converter. Strangely enough, when I tried to go back to converter you mentioned in https://gist.github.com/Koenkk/0316c10873dcc48905c6e8f63fa4b09c z2m failed to start with it. I can provide logs
> 
> Then I reverted back to no converters, and the situation returned to where it started: only on/off control and no dim level control. I don't have a tuya gateway, so I guess I will not be able to sniff the traffic...

I have the tuya gateway, I bought it to have the possibility of dimming the lights, but my config is janky as hell right now ;)
Home assistant manages that though. Problem is I'm not very technical gifted regarding z2m, so if someone provide some instruction I can do the sniffing part and bring logs here.

Koenkk, 2022-12-07:
@FizykPSX just a sniff where you control the brightness via the TuYa gateway is enough. I also need the network key of your TuYa gateway. See https://www.zigbee2mqtt.io/advanced/zigbee/04_sniff_zigbee_traffic.html

Yari117, 2022-12-07:
@FizykPSX Wow, that would be great if you could do the sniffing! Thank you!

FizykPSX, 2022-12-08:
> @FizykPSX just a sniff where you control the brightness via the TuYa gateway is enough. I also need the network key of your TuYa gateway. See https://www.zigbee2mqtt.io/advanced/zigbee/04_sniff_zigbee_traffic.html

It's very nice tutorial. I'll take care of getting this data during the weekend.
Thank you!

Yari117, 2022-12-21:
@FizykPSX Any luck with sniffing?

FizykPSX, 2022-12-28:
> @FizykPSX Any luck with sniffing?

Sorry not yet :( Had crazy last couple of weeks. But finally I'm starting short vacation, hopefully give an update here soon

midiwidi, 2022-12-28:
I'm facing the exact same issue with the Lonsonho QS-Zigbee-D02-TRIAC-2C-LN 2 channel dimmer.
On/Off control is working fine, but brightness adjustment is only working with the hardware buttons, not via z2m.
It would be awesome if the dimming functionality could be added to z2m.

midiwidi, 2023-01-06:
@FizykPSX Did you had a chance to do the sniffing. I don't want to push you, but it would be really helpful.

daviddewaard, 2023-01-14:
I have the same problem .
I have a  Lonsonho QS-Zigbee-D02-TRIAC-2C-LN 2 

im using zigbeeMqtt edge   with Mqtt broker. both the most resent version

Zigbee Model: TS110E
Zigbee Manufacturer : _TZ3210_pagajpog

also the converters mentioned earlier does not solve te problem.

ik have a lot of lthis type in my system ,  but this batch seems to have a dimmer problem.

the ones with sigbee model TS110F. are working fine,   they only don't work with transition times

jangcy, 2023-01-23:
Hi everyone! 
I am also experiencing the same problem, I can turn on and off both L1 and L2 using HA, but brightness adjustment doesn't work from the app, only when using hardware switches.

Can I help in any way? I don't know if I can do some sniffing using my Sonoff gateway.

P.S. The possibility to adjust these lights was my first and only reason to invest money in HA :D

jangcy, 2023-01-23:
I have been also trying with some additional converters, for example:

https://github.com/Koenkk/zigbee2mqtt/issues/12763#issuecomment-1151908806

But when restarting Z2MQTT I get the following error:
`[20:42:47] INFO: Starting Zigbee2MQTT...
/app/node_modules/zigbee-herdsman-converters/index.js:91
        if (converter.options) {
                      ^
TypeError: Cannot read properties of undefined (reading 'options')
    at Object.addDefinition [as addDeviceDefinition] (/app/node_modules/zigbee-herdsman-converters/index.js:91:23)
    at new ExternalConverters (/app/lib/extension/externalConverters.ts:15:17)
    at new Controller (/app/lib/controller.ts:84:58)
    at start (/app/index.js:106:18)`

Any ideas?

Koenkk, 2023-01-24:
@jangcy  Could you check if the issue is fixed with the following external converter: https://gist.github.com/Koenkk/eeb32620579515f5645ca8f79d16757a

- save this as file next to `configuration.yaml` as `ext_converter.js`
- add it to `configuration.yaml`:
```yaml
external_converters:
  - ext_converter.js
```
- start z2m, check if issue is fixed (it will be recognized as a 1 gang, we can fix that later if the dimming works now)

Koenkk, 2023-01-25:
@jangcy can you provide the debug log when it jumps back after dimming?

See https://www.zigbee2mqtt.io/guide/usage/debug.html on how to enable debug logging.

jangcy, 2023-01-25:
@Koenkk 

I have turned light on by changing brightness to 168 and after few seconds turned it off. Log of that (please let me know if I can provide some more information):

```
debug 2023-01-25 17:02:16Received MQTT message on 'zigbee2mqtt/0xa4c1388a3ec6c249/set' with data '{"brightness":168}'
debug 2023-01-25 17:02:16Publishing 'set' 'brightness' to '0xa4c1388a3ec6c249'
debug 2023-01-25 17:02:17Received Zigbee message from '0xa4c1388a3ec6c249', type 'attributeReport', cluster 'genOnOff', data '{"onOff":1}' from endpoint 1 with groupID 0
info 2023-01-25 17:02:17MQTT publish: topic 'zigbee2mqtt/0xa4c1388a3ec6c249', payload '{"brightness":100,"brightness_l1":193,"brightness_l2":155,"light_type":null,"linkquality":192,"max_brightness":254,"min_brightness":1,"power_on_behavior":"on","state":"ON","state_l1":"OFF","state_l2":"OFF"}'
info 2023-01-25 17:02:17MQTT publish: topic 'zigbee2mqtt/0xa4c1388a3ec6c249/type', payload '{"light":null}'
debug 2023-01-25 17:02:17Received Zigbee message from '0xa4c1388a3ec6c249', type 'attributeReport', cluster 'genOnOff', data '{"onOff":1}' from endpoint 1 with groupID 0
info 2023-01-25 17:02:17MQTT publish: topic 'zigbee2mqtt/0xa4c1388a3ec6c249', payload '{"brightness":100,"brightness_l1":193,"brightness_l2":155,"light_type":null,"linkquality":200,"max_brightness":254,"min_brightness":1,"power_on_behavior":"on","state":"ON","state_l1":"OFF","state_l2":"OFF"}'
info 2023-01-25 17:02:17MQTT publish: topic 'zigbee2mqtt/0xa4c1388a3ec6c249/type', payload '{"light":null}'
debug 2023-01-25 17:02:18Received Zigbee message from '0xa4c1388a3ec6c249', type 'attributeReport', cluster 'genOnOff', data '{"onOff":1}' from endpoint 1 with groupID 0
info 2023-01-25 17:02:18MQTT publish: topic 'zigbee2mqtt/0xa4c1388a3ec6c249', payload '{"brightness":100,"brightness_l1":193,"brightness_l2":155,"light_type":null,"linkquality":192,"max_brightness":254,"min_brightness":1,"power_on_behavior":"on","state":"ON","state_l1":"OFF","state_l2":"OFF"}'
info 2023-01-25 17:02:18MQTT publish: topic 'zigbee2mqtt/0xa4c1388a3ec6c249/type', payload '{"light":null}'
debug 2023-01-25 17:02:18Received Zigbee message from '0xa4c1388a3ec6c249', type 'attributeReport', cluster 'genOnOff', data '{"onOff":1}' from endpoint 1 with groupID 0
info 2023-01-25 17:02:18MQTT publish: topic 'zigbee2mqtt/0xa4c1388a3ec6c249', payload '{"brightness":100,"brightness_l1":193,"brightness_l2":155,"light_type":null,"linkquality":192,"max_brightness":254,"min_brightness":1,"power_on_behavior":"on","state":"ON","state_l1":"OFF","state_l2":"OFF"}'
info 2023-01-25 17:02:18MQTT publish: topic 'zigbee2mqtt/0xa4c1388a3ec6c249/type', payload '{"light":null}'
debug 2023-01-25 17:02:18Received Zigbee message from '0xa4c1388a3ec6c249', type 'attributeReport', cluster 'genOnOff', data '{"onOff":1}' from endpoint 1 with groupID 0
info 2023-01-25 17:02:18MQTT publish: topic 'zigbee2mqtt/0xa4c1388a3ec6c249', payload '{"brightness":100,"brightness_l1":193,"brightness_l2":155,"light_type":null,"linkquality":180,"max_brightness":254,"min_brightness":1,"power_on_behavior":"on","state":"ON","state_l1":"OFF","state_l2":"OFF"}'
info 2023-01-25 17:02:18MQTT publish: topic 'zigbee2mqtt/0xa4c1388a3ec6c249/type', payload '{"light":null}'
debug 2023-01-25 17:02:18Received Zigbee message from '0xa4c1388a3ec6c249', type 'attributeReport', cluster 'genOnOff', data '{"onOff":1}' from endpoint 1 with groupID 0
info 2023-01-25 17:02:18MQTT publish: topic 'zigbee2mqtt/0xa4c1388a3ec6c249', payload '{"brightness":100,"brightness_l1":193,"brightness_l2":155,"light_type":null,"linkquality":180,"max_brightness":254,"min_brightness":1,"power_on_behavior":"on","state":"ON","state_l1":"OFF","state_l2":"OFF"}'
info 2023-01-25 17:02:18MQTT publish: topic 'zigbee2mqtt/0xa4c1388a3ec6c249/type', payload '{"light":null}'
info 2023-01-25 17:02:19MQTT publish: topic 'zigbee2mqtt/0xa4c1388a3ec6c249', payload '{"brightness":168,"brightness_l1":193,"brightness_l2":155,"light_type":null,"linkquality":200,"max_brightness":254,"min_brightness":1,"power_on_behavior":"on","state":"ON","state_l1":"OFF","state_l2":"OFF"}'
info 2023-01-25 17:02:19MQTT publish: topic 'zigbee2mqtt/0xa4c1388a3ec6c249/type', payload '{"light":null}'
debug 2023-01-25 17:02:19Received Zigbee message from '0xa4c1388a3ec6c249', type 'attributeReport', cluster 'genLevelCtrl', data '{"61440":661}' from endpoint 1 with groupID 0
info 2023-01-25 17:02:19MQTT publish: topic 'zigbee2mqtt/0xa4c1388a3ec6c249', payload '{"brightness":168,"brightness_l1":193,"brightness_l2":155,"light_type":null,"linkquality":200,"max_brightness":254,"min_brightness":1,"power_on_behavior":"on","state":"ON","state_l1":"OFF","state_l2":"OFF"}'
info 2023-01-25 17:02:19MQTT publish: topic 'zigbee2mqtt/0xa4c1388a3ec6c249/type', payload '{"light":null}'
debug 2023-01-25 17:02:20Received Zigbee message from '0xa4c1388a3ec6c249', type 'attributeReport', cluster 'genLevelCtrl', data '{"61440":661}' from endpoint 1 with groupID 0
info 2023-01-25 17:02:20MQTT publish: topic 'zigbee2mqtt/0xa4c1388a3ec6c249', payload '{"brightness":168,"brightness_l1":193,"brightness_l2":155,"light_type":null,"linkquality":196,"max_brightness":254,"min_brightness":1,"power_on_behavior":"on","state":"ON","state_l1":"OFF","state_l2":"OFF"}'
info 2023-01-25 17:02:20MQTT publish: topic 'zigbee2mqtt/0xa4c1388a3ec6c249/type', payload '{"light":null}'
[... cut ...]
debug 2023-01-25 17:02:35Received Zigbee message from '0xa4c1388a3ec6c249', type 'attributeReport', cluster 'genOnOff', data '{"onOff":0}' from endpoint 1 with groupID 0
info 2023-01-25 17:02:35MQTT publish: topic 'zigbee2mqtt/0xa4c1388a3ec6c249', payload '{"brightness":100,"brightness_l1":193,"brightness_l2":155,"light_type":null,"linkquality":184,"max_brightness":254,"min_brightness":1,"power_on_behavior":"on","state":"OFF","state_l1":"OFF","state_l2":"OFF"}'
info 2023-01-25 17:02:35MQTT publish: topic 'zigbee2mqtt/0xa4c1388a3ec6c249/type', payload '{"light":null}'
```

Koenkk, 2023-01-25:
@jangcy 

- any change you set `optimistic` to `false` for this device in your z2m `configuration.yaml`? If yes, remove that.
- I've updated the [converter](https://gist.github.com/Koenkk/0316c10873dcc48905c6e8f63fa4b09c)
  - Both gangs should work now
  - light type is removed (not supported by the device as can be seen in your logs)
- I've never used a TuYa device with max/min brightness support, but I expect it limits the brightness. E.g. when you set `max_brightness` `50`, I would expect that setting `brightness` `255` makes the light **physically** only output `brightness` `50` **but** you will see `255` in the MQTT message.

jangcy, 2023-01-25:
@Koenkk 
Thanks a lot! Dimming works now and the slider stays in the set position.
However I have just changed the converter, hasn't touched anything else, I didn't have the optimistic flag set anywhere eiter, so it's interesting. Maybe it was something with previous converter?

Anyways, the new converter works fine, so I can continue playing around with my adjustable lights, thanks again!

midiwidi, 2023-01-25:
That's fantastic that we have a solution with that external converter now. Thank you very much @Koenkk.
Will the functionality of the external converter make it into z2m at a future update or would the converter always be required? Do you have a rough idea about an ETA?

Koenkk, 2023-01-26:
I've integrated the changes,  you can keep using the external converter until the next release (1 February) which will contain this fix.

Note that the fix has only been applied to `_TZ3210_pagajpog`, if you have another version which is not working in Z2M 1.29.2 let me know.

Yari117, 2023-01-26:
@Koenkk @FizykPSX  @midiwidi Dear Koen! I tried using your new converter with my _TZ3210_4ubylghk
I changed _TZ3210_pagajpog to _TZ3210_4ubylghk and to my surprise the dimmer finally worked!!!

Thus, I guess, you can include _TZ3210_4ubylghk in the final release as well

Thank you!

Koenkk, 2023-01-29:
@Yari117 done!

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

FizykPSX, 2023-01-31:
Hey Guys, some update from me, I apologize for being silent for such a long time. I tried to prep for sniffing the traffic, but one of my two sonoff dongles died and I was scarred to try flashing on the other working one. I have to order a new one.
What I can confirm that after changing the name in converter to _TZ3210_4ubylghk my dimmer started to work as desired, dimming works perfect. @Koenkk you're the BOSS.

Once again sorry for being silent, I had lot's of personal stuff to handle :(

Koenkk, 2023-02-01:
@FizykPSX great.

This fix is now in z2m 1.30.0 so I assume I can close this.

closed 2023-02-01 by Koenkk

VentusGallus, 2023-02-18:
Dear all - I am really happy to see that there is a solution now! I am using the same dimmer and have the same issue. Based on the above conversation, it is not clear to me what I need to change to my HA settings to make it work. 

I am running 2023.2.4 Home assistant. For zigbee, I am using a conbee II usb stick and I am using the Zigbee Home Intregation. 

Would be great if someone can share the steps I need to take

FizykPSX, 2023-02-20:
> Dear all - I am really happy to see that there is a solution now! I am using the same dimmer and have the same issue. Based on the above conversation, it is not clear to me what I need to change to [...]

Hi there, first of all you have toi ditch ZHA integration. It is the simplest but IMHO the most limited in terms of device support.
Conbee stick - I have no experience here, but I know that those problematic outside of DeConz intergation.
What I would suggest is to switch entirely into zigbee2mqtt, there is a full installation guide how to use it as addon in HA, or how to get it separately. You will not be dissapointed :)
For coordinator I'd recommend Sonoff dongle +, it's cheap, reliable and has a good range. Or try to use the new SkyConnect from Nabucasa.

midiwidi, 2023-02-20:
> Conbee stick - I have no experience here, but I know that those problematic outside of DeConz intergation.
I use a ConBee II stick together with zigbee2mqtt for over a year now without any problems.

sanderjohan, 2023-04-12:
@Koenkk, I have now QS-Zigbee-D02-TRIAC-L Zigbee dimmers with failing dimming functionality. Can you please add _TZ3210_weaqkhab to the next release.

Thanks in advance.

Sir-Pandalot, 2023-04-12:
> @Koenkk, I have now QS-Zigbee-D02-TRIAC-L Zigbee dimmers with failing dimming functionality. Can you please add _TZ3210_weaqkhab to the next release.
> 
> Thanks in advance.

Same here. I actually have 4 and they are all 4 not dimmable through z2m/home assistant:

Zigbee Model TS110E
Zigbee Manufacturer _TZ3210_weaqkhab
Manufacturer [Lonsonho](https://www.zigbee2mqtt.io/supported-devices/#v=Lonsonho)
Model [QS-Zigbee-D02-TRIAC-L](https://www.zigbee2mqtt.io/devices/QS-Zigbee-D02-TRIAC-L.html#lonsonho-qs-zigbee-d02-triac-l)

sanderjohan, 2023-04-13:
@Koenkk Thanks and it works.

But a small correction is needed because the dimmer shows up in z2m as TS110E_2gang_2. And this should be TS110E_1gang_1.

Sir-Pandalot, 2023-04-14:
@Koenkk It actually did _not_ work for me. I figured it's because it is a 1 channel dimmer and only exposes 1 topic not 2 like the 2 channel dimmer (l1 and l2). 

I have never written any code for z2m nor do I have much experience coding in general, but I took a look at the code you provided and I edited things that looked logical to me as a noob...

The code for ext_converter.js below works for me, but I have no idea if this is written "properly"

```
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const ota = require('zigbee-herdsman-converters/lib/ota');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const utils = require('zigbee-herdsman-converters/lib/utils');
const globalStore = require('zigbee-herdsman-converters/lib/store');
const e = exposes.presets;
const ea = exposes.access;

const fzLocal = {
    TS110E: {
        cluster: 'genLevelCtrl',
        type: ['attributeReport', 'readResponse'],
        convert: (model, msg, publish, options, meta) => {
            const result = {};
            if (msg.data.hasOwnProperty('64515')) {
                result['min_brightness'] = utils.mapNumberRange(msg.data['64515'], 0, 1000, 1, 255);
            }
            if (msg.data.hasOwnProperty('64516')) {
                result['max_brightness'] = utils.mapNumberRange(msg.data['64516'], 0, 1000, 1, 255);
            }
            if (msg.data.hasOwnProperty('61440')) {
                result['brightness'] = utils.mapNumberRange(msg.data['61440'], 0, 1000, 0, 255);
            }
            return result;
        },
    },
    TS110E_light_type: {
        cluster: 'genLevelCtrl',
        type: ['attributeReport', 'readResponse'],
        convert: (model, msg, publish, options, meta) => {
            const result = {};
            if (msg.data.hasOwnProperty('64514')) {
                const lookup = {0: 'led', 1: 'incandescent', 2: 'halogen'};
                result['light_type'] = lookup[msg.data['64514']];
            }
            return result;
        },
    },
};

const tzLocal = {
    TS110E_onoff_brightness: {
        key: ['state', 'brightness'],
        options: [],
        convertSet: async (entity, key, value, meta) => {
            const {message, state} = meta;
            if (message.state === 'OFF' || (message.hasOwnProperty('state') && !message.hasOwnProperty('brightness'))) {
                return await tz.on_off.convertSet(entity, key, value, meta);
            } else if (message.hasOwnProperty('brightness')) {
                // set brightness
                if (state.state === 'OFF') {
                    await entity.command('genOnOff', 'on', {}, utils.getOptions(meta.mapped, entity));
                }

                const level = utils.mapNumberRange(message.brightness, 0, 254, 0, 1000);
                await entity.command('genLevelCtrl', 'moveToLevelTuya', {level, transtime: 100}, utils.getOptions(meta.mapped, entity));
                return {state: {state: 'ON', brightness: message.brightness}};
            }
        },
[... 28 lines cut ...]

const definition = {
    fingerprint: tuya.fingerprint('TS110E', ['_TZ3210_weaqkhab']),
    model: 'TS110E_1gang_1',
    vendor: 'TuYa',
    description: '1 channel dimmer',
    fromZigbee: [fzLocal.TS110E, fzLocal.TS110E_light_type, tuya.fz.power_on_behavior_1, fz.on_off],
    toZigbee: [tzLocal.TS110E_onoff_brightness, tzLocal.TS110E_options, tuya.tz.power_on_behavior_1, tz.light_brightness_move],
    exposes: [
        e.light_brightness().withMinBrightness().withMaxBrightness(),
        e.power_on_behavior().withAccess(ea.ALL)],
    configure: async (device, coordinatorEndpoint, logger) => {
        await tuya.configureMagicPacket(device, coordinatorEndpoint, logger);
        await reporting.bind(device.getEndpoint(1), coordinatorEndpoint, ['genOnOff', 'genLevelCtrl']);
        await reporting.onOff(device.getEndpoint(1));
    },
};

module.exports = definition;
```
Thanks for all the work!

Also, according to [QS-Zigbee-D02-TRIAC-L](https://www.zigbee2mqtt.io/devices/QS-Zigbee-D02-TRIAC-L.html#lonsonho-qs-zigbee-d02-triac-l) this device should also expose effect. I have no idea if it should and I don't see any Tuya device exposing effect. Should this be added? Is there a way for me to find out if this Lonsonho version is supposed to expose effect?

@sanderjohan On another note: are you by any chance selling these devices in the Netherlands? I bought mine from a guy named Sander and on the website it states this device doesn't work with HA now.

Koenkk, 2023-04-14:
Updated https://gist.github.com/Koenkk/a3ee9a02d62943418e3c7caae0b06519, can you check again?

sanderjohan, 2023-04-14:
Thanks for the update and the dimmer is working for me.

When I try to use the  light_type and power_on_behavior I get the error that these are unsupported attributes.

Koenkk, 2023-04-14:
Does min/max brightness work?

sanderjohan, 2023-04-14:
Yes the min/max brightness works. I checked the min and max setting and that is between 65 - 120

Koenkk, 2023-04-15:
Great pushed the fix!

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

gdbd, 2023-05-03:
@Koenkk hello, i am tried  latest version `[1.30.4]`  with `_TZ3210_weaqkhab`

with this version brightness does not change physically(in state it seems changing ok)

but converter above (https://github.com/Koenkk/zigbee2mqtt/issues/14628#issuecomment-1509053252) works great 

looks like converter and production fix works in different way?

Koenkk, 2023-05-04:
@gdbd I made a mistake in my fix, fixed it now.

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

gdbd, 2023-05-10:
@Koenkk hello, i am tested docker `latest-dev` and confirm that now it works without converters. thank you!

gdbd, 2023-05-15:
@Koenkk hello, i am tested docker latest-dev and confirm that now it works without converters.

But device not provide brightness control and status to home assistant.
this is responsibilty of device plugin in zigbee2mqtt or something else?

Koenkk, 2023-05-15:
@gdbd does it work correctly from the z2m frontend?

gdbd, 2023-05-15:
> @gdbd does it work correctly from the z2m frontend?

yes, correctly change brightness

in HA it also can control brightness manually but the is no actions to set it directly to the value by automation

only increase and decrease brightness posssible

Koenkk, 2023-05-15:
> in HA it also can control brightness manually but the is no actions to set it directly to the value by automation

I don't understand what you mean with this, can you explain it further?

gdbd, 2023-05-15:
> in a simple words: there is no way to create automation from external software which can get/set brightness value.

device does not privide controls to change brightness:

[image]   

but via this widget it can be changed manually and work as expected:

[image]

so, because if this, available automation actions not contain something like 'set brightness to value', only these:

[image]

not sure this is responsibility of z2m, sorry if this  true

Koenkk, 2023-05-15:
None of my lights in HA provide this, you can set the brightness with "call a service" -> "brightness value".

> not sure this is responsibility of z2m, sorry if this true

Indeed out of scope for Z2M, for further help use the HA support channels.

AndreasMaverick, 2023-10-08:
Hello Koenkk,
I'm using your dev branch and figured out that "TOGGLE" is not functional with my QS-Zigbee-D02-TRIAC-2C-L.
Is this expected from you?
Thank you in advance for your assistance.

adekloet, 2023-10-20:
Hi,

Has anybody tried the above converter for the TS110F _TZ3000_ktuoyvt5? it looks like this Loshono dimmer has the same issue with the Brightness not responding to the light turn on service call in HA. 

Thanks