Koenkk/zigbee2mqtt #14629: Blitzwolf BW-SHP13 plug (TS011F_plug_1) does not detect power below 10 watt

manny199, 2022-10-24:
### What happened?

My Blitzwolf BW-SHP13 plug does not detect watt below 10 watt. Basically, the value it shows with consumption below 8/10 watts is equal to ZERO.
For example, if I put a webcam it does not detect its consumption in watt because webcam have low consumption

### What did you expect to happen?

I expect power detection below 10 watt

### How to reproduce it (minimal and precise)

qqqq

### Zigbee2MQTT version

1.28.0

### Adapter firmware version

2021

### Adapter

Sonoff ZigBee 3.0

### Debug log

I have a bit of a weird problem. My Blitzwolf BW-SHP13 plug does not detect watt below 10 watt. Basically, the value it shows with consumption below 8/10 watts is equal to ZERO.
For example, if I put a webcam it does not detect its consumption in watt.
What the solution to this problem? 
The Plug is detected as TS011F_plug_1 and the firmware has been updated. The production is _TZ3000_amdymr7l

ster1um, 2022-10-24:
is not problem a zigbee2mqtt...
The plug it's not much sensitive is cheap plug.

manny199, 2022-10-24:
> The plug it's not much sensitive is cheap plug

Can you suggest me a quality plug that also detects below 2 watts?

ster1um, 2022-10-24:
I have 2 same plug and start measurement to 6W, for my scope it is not needed.
You wait other suggestion or try different plug in the device list

Koenkk, 2022-10-24:
Closing as this is an hardware issue of the plug, not something we can solve from z2m.

closed 2022-10-24 by Koenkk

manny199, 2022-10-25:
Although these plugs have the same name BW-SHP13, they have different hardware. Convenient to buy TS0121 to have reading above 1.5 watts. The model with hardware TS011F_plug_1 does not detect below 10 watts