Koenkk/zigbee2mqtt #14632: [New device support]: MOES MS-104CZ 3 gang switch

bazz46, 2022-10-24:
### Link

https://it.aliexpress.com/item/1005001593033528.html?spm=a2g0o.productlist.0.0.2d803775wKPzj8&algo_pvid=49ce8788-a926-4c27-afc3-873d133e83db&algo_exp_id=49ce8788-a926-4c27-afc3-873d133e83db-5&pdp_ext_f=%7B%22sku_id%22%3A%2212000030750239155%22%7D&pdp_npi=2%40dis%21EUR%2126.03%2115.62%21%21%21%21%21%400b0a050b16666294413806121e0b51%2112000030750239155%21sea&curPageLogUid=NA0Bg1XlpZAM

### Database entry

{"id":19,"type":"Router","ieeeAddr":"0xXXXXXXX","nwkAddr":14830,"manufId":4417,"manufName":"_TZ3000_pfc7i3kt","powerSource":"Mains (single phase)","modelId":"TS0003","epList":[1,2,3,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":256,"inClusterList":[3,4,5,6,57344,57345,0],"outClusterList":[25,10],"clusters":{"57344":{"attributes":{"53251":"AAAAAgAABAAA"}},"genOnOff":{"attributes":{"onOff":0,"onTime":0,"offWaitTime":0,"moesStartUpOnOff":2}},"manuSpecificTuya_3":{"attributes":{"switchType":0}},"genBasic":{"attributes":{"65506":54,"65508":0,"65534":0,"modelId":"TS0003","manufacturerName":"_TZ3000_pfc7i3kt","powerSource":1,"zclVersion":3,"appVersion":68,"stackVersion":0,"hwVersion":1,"dateCode":""}}},

### Comments

It is now paired and recognized as a Zemismart model ZM-L03E-Z

akakoori, 2022-11-16:
+1 for the same request

Trickytree1984, 2022-11-25:
also add "switch type" to Moes MS-105BZ please

Koenkk, 2022-11-26:
Does it work with:

```js
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const ota = require('zigbee-herdsman-converters/lib/ota');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const utils = require('zigbee-herdsman-converters/lib/utils');
const globalStore = require('zigbee-herdsman-converters/lib/store');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    fingerprint: [{modelID: 'TS0003', manufacturerName: '_TZ3000_vsasbzkf'},
        {modelID: 'TS0003', manufacturerName: '_TZ3000_pfc7i3kt'},
        {modelID: 'TS0003', manufacturerName: '_TZ3000_odzoiovu'}],
    model: 'TS0003_switch_module',
    vendor: 'TuYa',
    description: '3 gang switch module',
    whiteLabel: [{vendor: 'OXT', model: 'SWTZ23'}],
    toZigbee: extend.switch().toZigbee.concat([tz.moes_power_on_behavior, tz.tuya_switch_type]),
    fromZigbee: extend.switch().fromZigbee.concat([fz.moes_power_on_behavior, fz.tuya_switch_type]),
    exposes: [
        e.switch().withEndpoint('l1'),
        e.switch().withEndpoint('l2'),
        e.switch().withEndpoint('l3'),
        e.power_on_behavior(),
        e.switch_type_2(),
    ],
    endpoint: (device) => {
        return {'l1': 1, 'l2': 2, 'l3': 3};
    },
    meta: {multiEndpoint: true},
    configure: async (device, coordinatorEndpoint, logger) => {
        await tuya.configureMagicPacket(device, coordinatorEndpoint, logger);
        await reporting.bind(device.getEndpoint(1), coordinatorEndpoint, ['genOnOff']);
        await reporting.bind(device.getEndpoint(2), coordinatorEndpoint, ['genOnOff']);
        await reporting.bind(device.getEndpoint(3), coordinatorEndpoint, ['genOnOff']);
    },
};

module.exports = definition;

```

bazz46, 2022-12-02:
Hi, with the converter i have this errore:
[12:00:55] INFO: Preparing to start...
[12:00:55] INFO: Socat not enabled
[12:00:56] INFO: Starting Zigbee2MQTT...
/app/node_modules/zigbee-herdsman-converters/index.js:91
        if (converter.options) {
                      ^
TypeError: Cannot read properties of undefined (reading 'options')
    at Object.addDefinition [as addDeviceDefinition] (/app/node_modules/zigbee-herdsman-converters/index.js:91:23)
    at new ExternalConverters (/app/lib/extension/externalConverters.ts:15:17)
    at new Controller (/app/lib/controller.ts:84:58)
    at start (/app/index.js:106:18)

Koenkk, 2022-12-02:
Updated https://github.com/Koenkk/zigbee2mqtt/issues/14632#issuecomment-1328007281 (works with z2m 1.28.4 only)

mjrama, 2022-12-30:
Hi,

I had the same issue, but I added this block to moes.js and it seems to be working and it is recognized as Moes MS-104CZ
 
```
    {
        fingerprint: [{modelID: 'TS0003', manufacturerName: '_TZ3000_vsasbzkf'},{modelID: 'TS0003', manufacturerName: '_TZ3000_pfc7i3kt'},
            {modelID: 'TS0003', manufacturerName: '_TZ3000_odzoiovu'}],
        model: 'MS-104CZ',
        description: 'Smart light switch module (3 gang)',
        vendor: 'Moes',
        extend: tuya.extend.switch({switchType: true, endpoints: ['l1', 'l2', 'l3']}),
        meta: {multiEndpoint: true},
        endpoint: (device) => {
            return {l1: 1, l2: 2, l3: 3};
        },
        configure: async (device, coordinatorEndpoint, logger) => {
            await tuya.configureMagicPacket(device, coordinatorEndpoint, logger);
            const endpoint1 = device.getEndpoint(1);
            await reporting.bind(endpoint1, coordinatorEndpoint, ['genOnOff']);
            await reporting.onOff(endpoint1);
            const endpoint2 = device.getEndpoint(2);
            await reporting.bind(endpoint2, coordinatorEndpoint, ['genOnOff']);
            await reporting.onOff(endpoint2);
            const endpoint3 = device.getEndpoint(3);
            await reporting.bind(endpoint3, coordinatorEndpoint, ['genOnOff']);
            await reporting.onOff(endpoint3);
        },
    },
    
```
  
  The only issue is the image in the frontpage that it is using 
  https://slsys.github.io/Gateway/devices/png/TS0003.png
  
  instead of this
  
[image]

HobbyKonijn, 2023-01-11:
I have 2 of these switches as well, both report out _TZ3000_pfc7i3kt, manufactuur code: 0x1141.
Using Conbee II stick with latest software.

switch 1:
- Can be controlled one by one output.
- Power on of module, no relais switching

switch 2:
- If I switch 1 output (1) via phoscon both others will switch as well (2) and (3).
- Same as if I switch 2, (1) and (3) will swith as well.

How can I switch each output independent.
- Did a factory reset of switch 2, when power on all outputs will be activated directly.

bazz46, 2023-02-05:
Where can i find moes.js?

> Hi,
> 
> I had the same issue, but I added this block to moes.js and it seems to be working. It is recognized as Moes MS-104CZ
> 
> ```
>     {
>         fingerprint: [{modelID: 'TS0003', manufacturerName: '_TZ3000_vsasbzkf'},{modelID: 'TS0003', manufacturerName: '_TZ3000_pfc7i3kt'},
>             {modelID: 'TS0003', manufacturerName: '_TZ3000_odzoiovu'}],
>         model: 'MS-104CZ',
>         description: 'Smart light switch module (3 gang)',
>         vendor: 'Moes',
>         extend: tuya.extend.switch({switchType: true, endpoints: ['l1', 'l2', 'l3']}),
>         meta: {multiEndpoint: true},
>         endpoint: (device) => {
>             return {l1: 1, l2: 2, l3: 3};
>         },
>         configure: async (device, coordinatorEndpoint, logger) => {
>             await tuya.configureMagicPacket(device, coordinatorEndpoint, logger);
>             const endpoint1 = device.getEndpoint(1);
>             await reporting.bind(endpoint1, coordinatorEndpoint, ['genOnOff']);
>             await reporting.onOff(endpoint1);
>             const endpoint2 = device.getEndpoint(2);
>             await reporting.bind(endpoint2, coordinatorEndpoint, ['genOnOff']);
>             await reporting.onOff(endpoint2);
>             const endpoint3 = device.getEndpoint(3);
>             await reporting.bind(endpoint3, coordinatorEndpoint, ['genOnOff']);
>             await reporting.onOff(endpoint3);
>         },
>     },
>     
> ```
> 
> The only issue is the image in the frontpage that it is using. https://slsys.github.io/Gateway/devices/png/TS0003.png
> 
> instead of this
> 
> [image]

closed 2023-03-16 by github-actions[bot]