Koenkk/zigbee2mqtt #14635: Miboxer LED Controller fails to turn on after Home Assistant Restart

Curtisjk, 2022-10-24:
### What happened?

After a restart of Home Assistant, I am unable to turn on Miboxer [FUT039Z](https://www.zigbee2mqtt.io/devices/FUT039Z.html#miboxer-fut039z) RGB+CCT LED controller (_TZ3210_jicmoite). The LED strip transitions on, but immediately turns off. This is either via toggling on, or turning on via a scene in Home Assistant.

The only way I can get it to turn on is by manually changing the brightness. Following this it works as expected.

### Zigbee2MQTT version

1.28.0

### Adapter firmware version

20190425

### Adapter

Texas Instruments CC2531

### Debug log

Attempting to toggle the led strip on/off after a HA restart:

```
debug 2022-10-24 19:52:39: Received MQTT message on 'zigbee2mqtt/en_suite_mirror_led/set' with data '{"state":"OFF"}'
debug 2022-10-24 19:52:39: Publishing 'set' 'state' to 'en_suite_mirror_led'
info  2022-10-24 19:52:39: MQTT publish: topic 'zigbee2mqtt/en_suite_mirror_led', payload '{"brightness":120,"color":{"h":35,"hue":35,"s":72,"saturation":72,"x":0.4286,"y":0.4011},"color_mode":"color_temp","color_power_on_behavior":"previous","color_temp":320,"do_not_disturb":false,"linkquality":31,"state":"OFF"}'
debug 2022-10-24 19:52:44: Received MQTT message on 'zigbee2mqtt/en_suite_mirror_led/set' with data '{"state":"ON"}'
debug 2022-10-24 19:52:44: Publishing 'set' 'state' to 'en_suite_mirror_led'
debug 2022-10-24 19:52:44: Received Zigbee message from 'en_suite_mirror_led', type 'readResponse', cluster 'genLevelCtrl', data '{}' from endpoint 1 with groupID 0
info  2022-10-24 19:52:44: MQTT publish: topic 'zigbee2mqtt/en_suite_mirror_led', payload '{"brightness":120,"color":{"h":35,"hue":35,"s":72,"saturation":72,"x":0.4286,"y":0.4011},"color_mode":"color_temp","color_power_on_behavior":"previous","color_temp":320,"do_not_disturb":false,"linkquality":94,"state":"ON"}'
debug 2022-10-24 19:52:47: Received MQTT message on 'zigbee2mqtt/en_suite_mirror_led/set' with data '{"state":"OFF"}'
debug 2022-10-24 19:52:47: Publishing 'set' 'state' to 'en_suite_mirror_led'
info  2022-10-24 19:52:47: MQTT publish: topic 'zigbee2mqtt/en_suite_mirror_led', payload '{"brightness":120,"color":{"h":35,"hue":35,"s":72,"saturation":72,"x":0.4286,"y":0.4011},"color_mode":"color_temp","color_power_on_behavior":"previous","color_temp":320,"do_not_disturb":false,"linkquality":94,"state":"OFF"}'
debug 2022-10-24 19:52:48: Received MQTT message on 'zigbee2mqtt/en_suite_mirror_led/set' with data '{"state":"ON"}'
debug 2022-10-24 19:52:48: Publishing 'set' 'state' to 'en_suite_mirror_led'
debug 2022-10-24 19:52:48: Received Zigbee message from 'en_suite_mirror_led', type 'readResponse', cluster 'genLevelCtrl', data '{}' from endpoint 1 with groupID 0
info  2022-10-24 19:52:49: MQTT publish: topic 'zigbee2mqtt/en_suite_mirror_led', payload '{"brightness":120,"color":{"h":35,"hue":35,"s":72,"saturation":72,"x":0.4286,"y":0.4011},"color_mode":"color_temp","color_power_on_behavior":"previous","color_temp":320,"do_not_disturb":false,"linkquality":94,"state":"ON"}'
```

Attempting to set the brightness via Home Assistant (works after this)

```
debug 2022-10-24 19:52:50: Received MQTT message on 'zigbee2mqtt/en_suite_mirror_led/set' with data '{"state":"ON","brightness":134}'
debug 2022-10-24 19:52:51: Publishing 'set' 'brightness' to 'en_suite_mirror_led'
info  2022-10-24 19:52:51: MQTT publish: topic 'zigbee2mqtt/en_suite_mirror_led', payload '{"brightness":134,"color":{"h":35,"hue":35,"s":72,"saturation":72,"x":0.4286,"y":0.4011},"color_mode":"color_temp","color_power_on_behavior":"previous","color_temp":320,"do_not_disturb":false,"linkquality":94,"state":"ON"}'
debug 2022-10-24 19:53:08: Received MQTT message on 'zigbee2mqtt/en_suite_mirror_led/set' with data '{"state":"OFF"}'
debug 2022-10-24 19:53:08: Publishing 'set' 'state' to 'en_suite_mirror_led'
info  2022-10-24 19:53:08: MQTT publish: topic 'zigbee2mqtt/en_suite_mirror_led', payload '{"brightness":134,"color":{"h":35,"hue":35,"s":72,"saturation":72,"x":0.4286,"y":0.4011},"color_mode":"color_temp","color_power_on_behavior":"previous","color_temp":320,"do_not_disturb":false,"linkquality":94,"state":"OFF"}'
debug 2022-10-24 19:53:09: Received MQTT message on 'zigbee2mqtt/en_suite_mirror_led/set' with data '{"state":"ON"}'
debug 2022-10-24 19:53:09: Publishing 'set' 'state' to 'en_suite_mirror_led'
debug 2022-10-24 19:53:09: Received Zigbee message from 'en_suite_mirror_led', type 'readResponse', cluster 'genLevelCtrl', data '{}' from endpoint 1 with groupID 0
info  2022-10-24 19:53:09: MQTT publish: topic 'zigbee2mqtt/en_suite_mirror_led', payload '{"brightness":134,"color":{"h":35,"hue":35,"s":72,"saturation":72,"x":0.4286,"y":0.4011},"color_mode":"color_temp","color_power_on_behavior":"previous","color_temp":320,"do_not_disturb":false,"linkquality":94,"state":"ON"}'
```

Xitro01, 2023-01-03:
> 

I have a similair issue with a Miboxer FUT035Z, although it's not after a HA restart, that would not be such a big issue. It happens after being off for 1-2 hours. It requires me to turn on-off-on the LED controller in HA, for it to turn on. Strangely doing it with a button or a door-sensor isn't fixing it.

Have you managed to fix this issue?

Curtisjk, 2023-01-03:
No I've not been able to fix it. I need to raise a post of the Home Asssistant forums, but not got round to it yet.

Xitro01, 2023-01-03:
> No I've not been able to fix it. I need to raise a post of the Home Asssistant forums, but not got round to it yet.

I have Gledopto LED controllers as well without this issue. So I think I will change to that brand, might fix the issue.

Curtisjk, 2023-01-03:
> > No I've not been able to fix it. I need to raise a post of the Home Asssistant forums, but not got round to it yet.
> 
> I have Gledopto LED controllers as well without this issue. So I think I will change to that brand, might fix the issue.

I was considering that, or just creating an automation in HA which turns the LEDs on at 1%, then off again when it is started.

Xitro01, 2023-01-03:
After some troubleshooting I finally found out what causes the problem for my Miboxer.

Do you by any chance send a turn_off service call with a transition to your LEDs while closing or starting HA?

thatslolo, 2023-02-20:
Ran into the same problem. As far as I’m aware the issue seems to occur after sending a turn off command when the light is already off. 
Did you fix it in HA? My proposed solution would be to create a template light which doesn’t relay the "turn off" commands when the state is already "off". But I think it would be better to fix this on z2m side.

Would it be possible to add a filter in z2m for this device to not send the off command when the device is already off?

Xitro01, 2023-02-20:
> Ran into the same problem. As far as I’m aware the issue seems to occur after sending a turn off command when the light is already off. Did you fix it in HA? My proposed solution would be to create [...]

No, did not fix it. Needs to be resolved in Z2M. All I did for now is in my automations create a check if the lights are on or not. But in some cases it still happens, when you trigger an automation twice within 1 second. So what I did was basically just remove the transition and that solves it.

Curtisjk, 2023-02-20:
> > Ran into the same problem. As far as I’m aware the issue seems to occur after sending a turn off command when the light is already off. Did you fix it in HA? My proposed solution would be to create a template light which doesn’t relay the "turn off" commands when the state is already "off". But I think it would be better to fix this on z2m side.
> > Would it be possible to add a filter in z2m for this device to not send the off command when the device is already off?
> 
> No, did not fix it. Needs to be resolved in Z2M. All I did for now is in my automations create a check if the lights are on or not. But in some cases it still happens, when you trigger an automation twice within 1 second. So what I did was basically just remove the transition in my turn_off calls and that solves it.

How are you setting the transition in HA? I am turning the LED Strip off via an automation action but see no options for transition? I have transition turned off (set to 0) in Z2M.

Xitro01, 2023-02-20:
> How are you setting the transition in HA? I am turning the LED Strip off via an automation action but see no options for transition? I have transition turned off (set to 0) in Z2M.

It only happened for me while turning off the device with a transition in my automation. So normal turn on/turn offs didn’t give me any issues. So it shouldn’t happen if you don’t use transition in your automations. Maybe a different problem then?

thatslolo, 2023-02-20:
Tried the service light.turn_off with and without transition: same result. When turn_off is called while the driver is already off, the next turn_on is broken. 
For now I fixed it in the most used automations with a check before calling light.turn_off, but it would be nice if there could be a more general fix in z2m.

thatslolo, 2023-02-20:
Actually I do use the FUT036Z. But I assume the issue is the same as with the 039Z an 035Z.

Curtisjk, 2023-02-28:
Since adding the condition that the LED Controller is off before turning on, my issue has gone away. It appears that my automation to turn off the LED Controller as being triggered on HA start which put it in the broken state. 

This is because I use the state of a switch to turn the LED Strip off/on. When the off state was being synced with HA on boot, it was triggering the automation.

I guess there's still the risk of triggering it by other means, but for now this has fixed my main issue. Would still be nice if there was some mitigation in Z2M to fix the flaw in the Miboxer firmware.