Koenkk/zigbee2mqtt #14636: [Feature request]: Use Postgres as backing storage instead of disk

Cellivar, 2022-10-25:
### Is your feature request related to a problem? Please describe

Currently Z2M uses local disk for storage, including device information and device state. I run my Home Assistant and related services in a [Nomad cluster](https://www.nomadproject.io/) (like k8s but less complex) which works excellent, except for the additional complexity of dealing with local disk. While I can use [CSI storage volumes](https://learn.hashicorp.com/tutorials/nomad/stateful-workloads-csi-volumes) for mounting disk storage to jobs in the cluster this is a lot of effort for what appears to be relatively little actual on-disk information. 

### Describe the solution you'd like

I would love to be able to use my existing Postgres database instance for Z2M data storage. Z2M doesn't seem to generate a ton of data on its internal disk storage, in a clustered environment (or any environment where local disk may not be guaranteed or easy) a network database connection is much preferred.

The more things in my cluster running network-only services with no local state the better the overall cluster can operate.

### Describe alternatives you've considered

As my Zigbee setup does not change often I have currently _hardcoded_ the` dataase.db`, `configuration.yaml`, and `coordinator_backup.json` files directly into my Nomad job spec after generating them once. This works mostly fine, the only issue when something goes wrong and the container restarts is that my door sensors default to `Unknown` in Home Assistant. I have to go open and close all my windows to reset them. This has prevented me from feeling comfortable moving my IKEA smart home equipment off of the gateway and into Z2M though.

### Additional context

I'm comfortable enough with TypeScript to take a crack at this to _scratch my own itch_ if you can provide pointers on where to get started and the kind of direction you'd like me to take for such a feature. I understand implementing a database abstraction layer is a _non-trivial_ amount of effort for an uncommon configuration scenario.

Cellivar, 2022-11-25:
Not stale, still interested :)

Cellivar, 2022-12-26:
I'm still interested in this.

closed 2023-02-02 by github-actions[bot]

waleed-alharthi, 2025-01-06:
Not stale, still interested :)

NitriKx, 2025-11-02:
Would be interested as well