Koenkk/zigbee2mqtt #14644: Duplicate remote messages after fresh upgrade to CC2652p

KirovAir, 2022-10-25:
### What happened?

Simply set up my network, new panId and new channel. (Moved away from CC2531)
Repaired all my devices freshly. I do have quite a few router devices. 
I noticed my dimmer automation was not working. Noticed in my logs I get three messages from a single click of my Hue Dimmer:

```
Info 2022-10-25 13:47:34MQTT publish: topic 'zigbee2mqtt/Hue Dimmer', payload '{"action":"up_press_release","action_duration":null,"battery":43.5,"linkquality":112,"update":{"state":null},"update_available":null}'
Info 2022-10-25 13:47:34MQTT publish: topic 'zigbee2mqtt/Hue Dimmer/action', payload 'up_press_release'
Info 2022-10-25 13:47:34MQTT publish: topic 'zigbee2mqtt/Hue Dimmer/action', payload 'up_press_release'
Info 2022-10-25 13:47:34MQTT publish: topic 'zigbee2mqtt/Hue Dimmer/action', payload 'up_press_release'
```

My network map: [Here](https://www.imgdumper.nl/uploads9/6357cdd2ae22e/6357cdd2a82c1-Screenshot_2022-10-25_at_13.51.01.png)

### What did you expect to happen?

I would expect a single click event within the logs.

### How to reproduce it (minimal and precise)

Simply pair/re-pair the dimmer and it happens every single time.

### Zigbee2MQTT version

1.28.0 commit: unknown

### Adapter firmware version

20220219

### Adapter

Sonoff Dongle-P

### Debug log

```
Debug 2022-10-25 13:49:21Received Zigbee message from 'Hue Dimmer', type 'commandOffWithEffect', cluster 'genOnOff', data '{"effectid":0,"effectvariant":0}' from endpoint 1 with groupID 0
Debug 2022-10-25 13:49:21Received Zigbee message from 'Hue Dimmer', type 'commandHueNotification', cluster 'manuSpecificPhilips', data '{"button":4,"time":1,"type":2,"unknown1":3145728,"unknown2":0}' from endpoint 2 with groupID 0
Info 2022-10-25 13:49:21MQTT publish: topic 'zigbee2mqtt/Hue Dimmer', payload '{"action":"off_press_release","action_duration":null,"battery":43.5,"linkquality":116,"update":{"state":null},"update_available":null}'
Info 2022-10-25 13:49:21MQTT publish: topic 'zigbee2mqtt/Hue Dimmer/action', payload 'off_press_release'
Info 2022-10-25 13:49:21MQTT publish: topic 'zigbee2mqtt/Hue Dimmer/action', payload 'off_press_release'
Info 2022-10-25 13:49:21MQTT publish: topic 'zigbee2mqtt/Hue Dimmer/action', payload 'off_press_release'
Debug 2022-10-25 13:49:23Received Zigbee message from 'Hue Dimmer', type 'commandHueNotification', cluster 'manuSpecificPhilips', data '{"button":1,"time":0,"type":0,"unknown1":3145728,"unknown2":0}' from endpoint 2 with groupID 0
Info 2022-10-25 13:49:23MQTT publish: topic 'zigbee2mqtt/Hue Dimmer', payload '{"action":"on_press","action_duration":null,"battery":43.5,"linkquality":102,"update":{"state":null},"update_available":null}'
Info 2022-10-25 13:49:23MQTT publish: topic 'zigbee2mqtt/Hue Dimmer/action', payload 'on_press'
Info 2022-10-25 13:49:23MQTT publish: topic 'zigbee2mqtt/Hue Dimmer/action', payload 'on_press'
Info 2022-10-25 13:49:23MQTT publish: topic 'zigbee2mqtt/Hue Dimmer/action', payload 'on_press'
Debug 2022-10-25 13:49:23Received Zigbee message from 'Hue Dimmer', type 'commandOn', cluster 'genOnOff', data '{}' from endpoint 1 with groupID 0
Debug 2022-10-25 13:49:23Received Zigbee message from 'Hue Dimmer', type 'commandHueNotification', cluster 'manuSpecificPhilips', data '{"button":1,"time":1,"type":2,"unknown1":3145728,"unknown2":0}' from endpoint 2 with groupID 0
Info 2022-10-25 13:49:23MQTT publish: topic 'zigbee2mqtt/Hue Dimmer', payload '{"action":"on_press_release","action_duration":null,"battery":43.5,"linkquality":102,"update":{"state":null},"update_available":null}'
Info 2022-10-25 13:49:23MQTT publish: topic 'zigbee2mqtt/Hue Dimmer/action', payload 'on_press_release'
Info 2022-10-25 13:49:23MQTT publish: topic 'zigbee2mqtt/Hue Dimmer/action', payload 'on_press_release'
Info 2022-10-25 13:49:23MQTT publish: topic 'zigbee2mqtt/Hue Dimmer/action', payload 'on_press_release'
```

KirovAir, 2022-10-25:
Update: The problem seems to have resolved itself after a short while. Interesting! :)

closed 2022-10-25 by KirovAir

ster1um, 2022-10-25:
try to 
**stop z2m
delete database.db and coordinator_backup.json
start z2m
repair all device**