Koenkk/zigbee2mqtt #14645: OWON PC321 Can't modify report timing

renaiku, 2022-10-25:
### What happened?

The device reports change between 3 and 5 times per minutes so my Home Assistant database is giant because of this.

I tried to change the report frequency in z2m.

Setting this in the UI:
[image]

Gives this error.
```
error 2022-10-25 14:21:22Request 'zigbee2mqtt/bridge/request/device/configure_reporting' failed with error: 'ConfigureReporting 0x3c6a2cfffed21266/1 seMetering([{"attribute":"owonL1PhasePower","minimumReportInterval":60,"maximumReportInterval":3600,"reportableChange":"1"}], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 6252 - 1 - 238 - 1794 - 7 after 10000ms)'
```

### Zigbee2MQTT version

1.28.0

### Adapter firmware version

20220103

### Adapter

zzh

danieledwardgeorgehitchcock, 2022-10-26:
If your device is battery powered, it needs to be awake while setting the reporting interval

renaiku, 2022-10-26:
It is not https://www.zigbee2mqtt.io/devices/PC321.html

cddu33, 2022-10-30:
Hello, I have the same problem, I think the action does not exist in the convertissuer but I do not have the skill to modify it

Here is the info:
[image]

renaiku, 2022-10-30:
Oh perfect! If nobody tried in a few days I'll check it. I don't have time this weekend !

renaiku, 2022-11-04:
I don't find examples on how to add writeable config.
The file to update would be this one: https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/devices/owon.js

@cddu33 where did you find this file ?

cddu33, 2022-11-04:
Hi, here https://github.com/zigpy/zha-device-handlers/issues/1657

renaiku, 2022-12-05:
Still an issue.

closed 2023-01-13 by github-actions[bot]