Koenkk/zigbee2mqtt #14646: IKEA LED1624G9 with old firmware 1.3.013 - (No image available for imageType '10241') when checking for updates via OTA

marthubner, 2022-10-25:
### What happened?

I have several LED1624G9 bulbs. Most recent firmware update for this type of bulb is V-2.3.093 from september 21st 2022. Before that there was V-2.3.086 from 13th december 2021 and before that there was 1.3.013 from 29th june 2020.

All of my bulbs which were on V-2.3.086 got updated to V-2.3.093 without any problem. Two of my LED1624G9s which are still on firmware 1.3.013 now give me `Failed to check if update available for 'XY' (No image available for imageType '10241')` error message when I try to check for updates via OTA. I can see here http://fw.ota.homesmart.ikea.net/feed/version_info.json that IKEA for some reason probably has changed the imageType to "fw_image_type":10243 (probably from 10241) which is why I am now unable to update my two bulbs with 1.3.013 which is now 2 versions behind the newest firmware version.

I tried to play around [this](https://www.zigbee2mqtt.io/guide/usage/ota_updates.html#local-ota-index-and-firmware-files) guide but to no avail. I´m not 100% sure how to use it, I tried to add my_index.json  `[
    {
        "manufacturerCode": 4476,
        "imageType": 10241,
        "url": "http://fw.ota.homesmart.ikea.net/global/GW1.0/01.19.032/bin/10035515-TRADFRI-bulb-cws-2.3.093.ota.ota.signed",
        "force": true
    }
]` which did not work and gave me the same error.

I understand that the issue is not on Zigbee2MQTT side but I´m looking for every help I can get.
Is there a way to:
1) install the latest firmware directly?
2) install V-2.3.086 first? - (but I did´t find the file anywhere)

The reason why I want to update the firmware is that the 1.3.013 does not support power_on_behavior.

### What did you expect to happen?

firmware to update to the latest version

### How to reproduce it (minimal and precise)

Try to perform OTA update on LED1624G9 with firmware 1.3.013

### Zigbee2MQTT version

1.28.0-dev commit: 08ceae2

### Adapter firmware version

 20210319

### Adapter

CC2652R stick

MattWestb, 2022-10-26:
I have seen that the image ID is changed and it must being made in the 2.3.086 OTA update then the device was converted from ZLL to Zigbee 3.
I only have one one the CWS1 and it have updated OK in ZHA.

In ZHA you can putting OTA files local and also doing  downgrade with little hacking and it have working good with Zigbee 3 light but i have newer downgrading from Zigbee 3 to ZLL do its unknown territory for my.

Attacking the OTA i have found but it can being it getting older but then must using web archive for finding them in the cloud but i think its not needed.
[CWS1-OTA.zip](https://github.com/Koenkk/zigbee2mqtt/files/9866591/CWS1-OTA.zip)

I dont know if the image ID is in the real firmware or is it being store in the user data that shall not being written after leaving the factory and IKEA is patching it with the 2.3.086 and your device have only updating the main flash (the real firmware) and not patching the user data.

If you can try getting the 2.3.086 flashed one more time and see if its changing the image ID.

MattWestb, 2022-10-26:
One more thing is that the IKEA CWS1 is one pre first gen device.
Its is not having the normal IKEA Zigbee module its one that is only ben seen in the CWS1. Normal first gen devices was potted and having ZLL firmware and the CSW1 is it 2 so its nearly impossible opening it without permanent damage it for getting the SWD pads for flashing and dumping the firmware.

If it was not potted i have open my and dumping the main flash (its the bootloader + OTA firmware) plus the `user data` (its the factory setting with name and other hardware parameters is storage is).

Still one warning if trying rolling back to 1.3.013 it not tested if the device can handling it and its unknown territory for us.

marthubner, 2022-10-26:
@MattWestb Thank you so so much for the files you provided and for pointing me towards ZHA! I was able to update my two bulbs to the 2.3.086 using [this](https://www.reddit.com/r/homeassistant/comments/fak430/how_to_update_your_ikea_or_ledevance_firmware/) guide.

For information: I found out that with firmware 1.3.013 on LED1624G9 there was a value 10241 in cluster ota, image_type_id attribute:
`Ota': {"image_type_id":10241}`
Whether with 2.3.086 and 2.3.093 on LED1624G9 and even on all other IKEA bulbs I have and their firmware is up to date, the value is 65535
`Ota': {"image_type_id":65535}`

So problem solved and once again thank you very much for such great help!

closed 2022-10-26 by marthubner

MattWestb, 2022-10-26:
Great !!

I was seen that the image ID was changed but was updating it in ZHA discussion of IKEA Firmware resale but i think i shall doing one new wit the old ID and writing that the new (Zigbee3) firmware have one new ID.

Was you getting the service working for requesting images ?
I was trying but was not working for my and i have written most of the guide in Zigpy Wiki that your guide one copy of.

The real ID you only getting then the system is requesting OTA update from the device and the ZLL firmware is functioning little different so you was getting `10241` but perhaps you is getting the real one after one successful request to the device.

I think `65535` (0xffff) is the same as not know or not defined / requested.

marthubner, 2022-10-26:
I am sorry but I don't really know what Image requesting service is. I don't have any deep knowledge about how this all really works, I did just set up ZHA, followed the guide, put the 2.3.086 firmware file in a local folder, wrote everything what was needed to configuration.yaml, used the zha.issue_zigbee_cluster_command as described in the guide and it worked... :)

MattWestb, 2022-10-26:
Its working i have testing with the old CWS1 600 Lm and new CWS3 800 Lm and both is working if doing it right (that i was not then doing the first test).

ZHA is always checking for firmware update then device is joining and it working great.

psitem, 2022-11-05:
I'd love some help getting this working without resorting to ZHA.

In my configuration.yaml I have:

```
ota:
  zigbee_ota_override_index_location: override.json
  ikea_ota_use_test_url: false

```

In override.json:

```
[
    {
        "fileVersion": 587806257,
        "fileSize": 227500,
        "manufacturerCode": 4476,
        "imageType": 10243,
        "sha512": "fcdcc6198cd5f41d3602000207fa4a4c7678279dce4c2e8467d2044293e27d8d9d63e0aa71b5125af62a74b599bfb0dd28645961fb034c803722b732d20802d0",
        "url": "10035515-TRADFRI-bulb-cws-2.3.093.ota.ota.signed"
    },
    {
        "fileVersion": 587753009,
        "fileSize": 226244,
        "manufacturerCode": 4476,
        "imageType": 10241,
        "sha512": "e9826a19df240cd3793fe1a2d8e99ea62941379663da2447bb1b0fae4340dd268647c9dc21d790b2cf42a23e59d04011d8db71ec8f086cee7699663ae8c884a7",
        "url": "10035515-TRADFRI-bulb-cws-2.3.086.ota.ota.signed"
    }
]
```

I put the `110035515-TRADFRI-bulb-cws-2.3.086.ota.ota.signed` and `10035515-TRADFRI-bulb-cws-2.3.093.ota.ota.signed` files from the .zip you'd attached in the `data/` directory. z2m version is `1.28.2 commit: e66fe85`, LED1624G9 bulb is on 1.3.013, coordinator is CC2652R USB stick, and I continue to get the message that there's no image available for imageType '10241'.

damien-pang, 2022-11-06:
would like some way to resolve this without ZHA too..

damien-pang, 2022-11-08:
Have conceded and ran up a HA instance on VirtualBox (passing through my USB coordinator). Got the bulbs updated with the method above via ZHA, can confirm the imageType updated to 10243.

MattWestb, 2022-11-08:
@damien-pang Z2M is normally not updating the record for firmware version then have self doing the OTA and also if you have doing it "behind its back".
Also is doing function upgrade like ZLL -> ZB3 or the device is changing its endpoints and cluster it can being good deleting the device and adding it agen so the old information is going away and the new is being red OK (also the version is being fixed then).

MattWestb, 2022-11-08:
If some need the OTA files i have Zipping all i have found (not the very old but the last ZLL and to current ZB3).
[IKEA-OTA-CWS1.zip](https://github.com/Koenkk/zigbee2mqtt/files/9958404/IKEA-OTA-CWS1.zip)

kezmdor, 2022-12-04:
I ran into this problem too. I powered up a local http server an relocated the ota-URL to it in my local network. I edited the file "version_info.json" to "fw_image_type" = 10241.
Z2M ota update grabbed the "version_info.json" from my webserver and downloaded the desired firmware too. Sadly it looks to me that the "fw_image_type" is also coded within the firmware image file, because Z2M refuses to install the update:

`2022-12-04 12:28:35Received Zigbee message from '0x588e81fffe41a16b', type 'readResponse', cluster 'genBasic', data '{"dateCode":"20170315","swBuildId":"1.3.002"}' from endpoint 1 with groupID 0
2022-12-04 12:28:35Updating to latest '0x588e81fffe41a16b' (TRADFRI bulb E27 CWS opal 600lm)
2022-12-04 12:28:35Using endpoint '1'
2022-12-04 12:28:35Received Zigbee message from '0x588e81fffe41a16b', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":1,"fileVersion":318767730,"imageType":10241,"manufacturerCode":4476}' from endpoint 1 with groupID 0
2022-12-04 12:28:35Got OTA request '{"fieldControl":1,"manufacturerCode":4476,"imageType":10241,"fileVersion":318767730}'
2022-12-04 12:28:35getNewImage for '0x588e81fffe41a16b', meta {"fileVersion":587806257,"url":"http://fw.ota.homesmart.ikea.net/global/GW1.0/01.19.043/bin/10035515-TRADFRI-bulb-cws-2.3.093.ota.ota.signed"}
2022-12-04 12:28:35getNewImage for '0x588e81fffe41a16b', image header {"otaUpgradeFileIdentifier":{"type":"Buffer","data":[30,241,238,11]},"otaHeaderVersion":256,"otaHeaderLength":56,"otaHeaderFieldControl":0,"manufacturerCode":4476,"imageType":10243,"fileVersion":587806257,"zigbeeStackVersion":2,"otaHeaderString":"EBL tradfri_wrgb\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000","totalImageSize":227500}
2022-12-04 12:28:35Update of '0x588e81fffe41a16b' failed (AssertionError [ERR_ASSERTION]: Image type mismatch)`

There must exist an update file that changes the "fw_image_type" from 10241 to 10243. I am not sure if it is even possible to force the installation of a firmware file with 10243 or if it will break the bulb ...
The update is refused in zigbee2mqtt/node_modules/zigbee-herdsman-converters/lib/ota/common.js: `assert(image.header.imageType === current.imageType, 'Image type mismatch');`

Any ideas? Is anyone brave enough to risk one bulb?

MattWestb, 2022-12-04:
If you is using the right OTA file it shall working.
```
    {
        "fileVersion": 587753009,
        "fileSize": 226244,
        "manufacturerCode": 4476,
        "imageType": 10241,
        "sha512": "e9826a19df240cd3793fe1a2d8e99ea62941379663da2447bb1b0fae4340dd268647c9dc21d790b2cf42a23e59d04011d8db71ec8f086cee7699663ae8c884a7",
        "url": "10035515-TRADFRI-bulb-cws-2.3.086.ota.ota.signed"
    }
```
Its the "transformer".

Si1entDave, 2022-12-04:
In case it's useful to anyone:
I tried everything in this thread that didn't require setting up ZHA (I'm using HA/Mosquitto/Zigbee2MQTT in separate docker containers, and nothing worked. The use of of a local or "override" index did nothing - it's as if Tradfri doesn't even bother using the index, either central or local, because it has its own repository.

What DID work however, as of 10 minutes ago from time of this post, was enabling the ikea test server with
`ikea_ota_use_test_url: true`

I tried that just in case it worked, and everything happened seamlessly!

I'm good at this point, but hey, maybe my experience will help someone else.

Si1entDave, 2022-12-04:
Addendum:
Enabling the IKEA test server took me from 1.3.009 to 1.3.013, but now I can't take it any further. with IKEA test server either enabled or disabled, and with or without the local index and files suggested above.

I'm stumped.

kezmdor, 2022-12-05:
Thanks to @MattWestb, it worked for me. All my bulbs are upgraded to the latest firmware.

@Si1entDave, I hope this helps you to get it working:
As I wrote earlier I started a local apache webserver and redirected the original Ikea update URL to this webserver in my local network. Since I am running Z2M on a RPi it was nothing more than adding this to the `/etc/hosts` file:
`192.168.0.100 fw.ota.homesmart.ikea.net`
If you are running Z2M inside a container you may have to shell into the container. Check the access logs of your webserver to see if Z2M is redirected to it.

Next I placed the file `version_info.json` in a folder named `feed` within the webroot:
`[
    {
        "fw_binary_url": "http://fw.ota.homesmart.ikea.net/global/GW1.0/01.19.043/bin/10035515-TRADFRI-bulb-cws-2.3.086.ota.ota.signed",
        "fw_file_version_LSB": 26161,
        "fw_file_version_MSB": 8968,
        "fw_filesize": 227244,
        "fw_image_type": 10241,
        "fw_manufacturer_id": 4476,
        "fw_type": 2
    }
]`

I placed the firmware (available as download in this issue) in the path described in the `version_info.json` and started the update of the bulbs. After the update of all bulbs was done I removed the line from `/etc/hosts` and checked again for updates and installed them from the original Ikea update server. Now all my bulbs are on the latest available firmware. :)

Cheers

louiecaulfield, 2023-01-31:
Thanks @kezmdor for writing it up so clearly.

Minor simplification is to change the `fw_binary_url` field and place the firmware image next to the version.json like so

```
[
    {
        "fw_binary_url": "http://fw.ota.homesmart.ikea.net/feed/10035515-TRADFRI-bulb-cws-2.3.086.ota.ota.signed",
        "fw_file_version_LSB": 26161,
        "fw_file_version_MSB": 8968,
        "fw_filesize": 227244,
        "fw_image_type": 10241,
        "fw_manufacturer_id": 4476,
        "fw_type": 2
    }
]
```

stigglitch, 2023-02-22:
I cant get this to work. Trying to update from 1.3.013

1. Run webserver
2. Put firmware in feed dir
3. Put version.info in feed dir
[
    {
        "fw_binary_url": "http://fw.ota.homesmart.ikea.net/feed/10035515-TRADFRI-bulb-cws-2.3.086.ota.ota.signed",
        "fw_file_version_LSB": 26161,
        "fw_file_version_MSB": 8968,
        "fw_filesize": 227244,
        "fw_image_type": 10241,
        "fw_manufacturer_id": 4476,
        "fw_type": 2
    }
]
4. Ikea TEST URL disabled
5. NO override.json

ERROR: Failed to check if update available for '0xd0cf5efffe24a2d8' (Request failed with status code 404)

The webserver is accessible at  http://fw.ota.homesmart.ikea.net/feed/10035515-TRADFRI-bulb-cws-2.3.086.ota.ota.signed