Koenkk/zigbee2mqtt #14647: Stuck at Starting zigbee-herdsman with CC2651P

algovera, 2022-10-25:
### What happened?

Testing between different chips, everything was okay with efr32, and back to testing CC2652P and got stuck at
Starting zigbee-herdsman.

Switching back to erf32 stick works again, apparently I am not able to use CC2652P anymore.
I have one Sonoff ZBdongle-P and another ETH based hub using CC2652P.

I tried the methods here, neither works.
Please assist
https://www.zigbee2mqtt.io/guide/configuration/zigbee-network.html
https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start.html#error-coordinator-failed-to-start-probably-the-panid-is-already-in-use-try-a-different-panid-or-channel

**Config**
data_path: /config/z2m_zbdongle
socat:
  enabled: false
  master: pty,raw,echo=0,link=/tmp/ttyZ2M,mode=777
  slave: tcp-listen:8485,keepalive,nodelay,reuseaddr,keepidle=1,keepintvl=1,keepcnt=5
  options: "-d -d"
  log: false
mqtt:
  base_topic: zigbee2mqtt
  server: mqtt://core-mosquitto:1883
serial:
  port: >-
    /dev/serial/by-id/usb-ITead_Sonoff_Zigbee_3.0_USB_Dongle_Plus_e8a0b4b0401aec11bab874e5f01c6278-if00-port0
permit_join: true
zigbee_herdsman_debug: true
advanced:
  log_level: debug
  pan_id: GENERATE
  ext_pan_id: GENERATE
  channel: 25
  network_key:
    - 1
    - 3
    - 5
    - 7
    - 9
    - 11
    - 13
    - 15
    - 0
    - 2
    - 4
    - 6
    - 8
    - 10
    - 12
    - 13

### What did you expect to happen?

On returning to CC2652P, it should continue to work

### Zigbee2MQTT version

 1.28.0-1 

### Adapter firmware version

Unable to see find, but latest for ZBDongle-P

### Adapter

Sonoff ZBdongle-P, and another no brand from China, both  CC2651P

### Debug log

**DEBUG LOG with CC2562P Hub**
[22:37:17] INFO: Preparing to start...
[22:37:17] INFO: Socat not enabled
[22:37:18] INFO: Zigbee Herdsman debug logging enabled
[22:37:19] INFO: Starting Zigbee2MQTT...
Zigbee2MQTT:debug 2022-10-25 22:37:23: Loaded state from file /config/zigbee2mqtt/state.json
Zigbee2MQTT:info  2022-10-25 22:37:23: Logging to console and directory: '/config/zigbee2mqtt/log/2022-10-25.22-37-22' filename: log.txt
Zigbee2MQTT:debug 2022-10-25 22:37:23: Removing old log directory '/config/zigbee2mqtt/log/2022-10-25.20-06-09'
Zigbee2MQTT:info  2022-10-25 22:37:23: Starting Zigbee2MQTT version 1.28.0 (commit #unknown)
Zigbee2MQTT:info  2022-10-25 22:37:23: Starting zigbee-herdsman (0.14.62)
Zigbee2MQTT:debug 2022-10-25 22:37:23: Using zigbee-herdsman with settings: '{"adapter":{"concurrent":null,"delay":null,"disableLED":false},"backupPath":"/config/zigbee2mqtt/coordinator_backup.json","databaseBackupPath":"/config/zigbee2mqtt/database.db.backup","databasePath":"/config/zigbee2mqtt/database.db","network":{"channelList":[17],"extendedPanID":[221,221,221,221,221,221,221,221],"networkKey":"HIDDEN","panID":6754},"serialPort":{"adapter":"ezsp","path":"tcp://192.168.1.9:1024"}}'
2022-10-25T14:37:23.417Z zigbee-herdsman:controller:log Starting with options '{"network":{"networkKeyDistribute":false,"networkKey":[1,3,5,7,9,11,13,15,0,2,4,6,8,10,12,13],"panID":6754,"extendedPanID":[221,221,221,221,221,221,221,221],"channelList":[17]},"serialPort":{"path":"tcp://192.168.1.9:1024","adapter":"ezsp"},"databasePath":"/config/zigbee2mqtt/database.db","databaseBackupPath":"/config/zigbee2mqtt/database.db.backup","backupPath":"/config/zigbee2mqtt/coordinator_backup.json","adapter":{"disableLED":false,"concurrent":null,"delay":null}}'
2022-10-25T14:37:23.421Z zigbee-herdsman:adapter:ezsp:uart Opening TCP socket with 192.168.1.9:1024
2022-10-25T14:37:23.428Z zigbee-herdsman:adapter:ezsp:uart Socket connected
2022-10-25T14:37:23.430Z zigbee-herdsman:adapter:ezsp:uart Socket ready
2022-10-25T14:37:23.430Z zigbee-herdsman:adapter:ezsp:uart Uart reseting
2022-10-25T14:37:23.431Z zigbee-herdsman:adapter:ezsp:uart --> Write reset
2022-10-25T14:37:23.433Z zigbee-herdsman:adapter:ezsp:uart --> [1ac038bc7e]

**DEBUG LOG with Sonoff ZBdongle-P**
[22:43:39] INFO: Preparing to start...
[22:43:40] INFO: Socat not enabled
[22:43:40] INFO: Zigbee Herdsman debug logging enabled
[22:43:41] INFO: Starting Zigbee2MQTT...
Zigbee2MQTT:debug 2022-10-25 22:43:45: Loaded state from file /config/zigbee2mqtt/state.json
Zigbee2MQTT:info  2022-10-25 22:43:45: Logging to console and directory: '/config/zigbee2mqtt/log/2022-10-25.22-43-45' filename: log.txt
Zigbee2MQTT:debug 2022-10-25 22:43:45: Removing old log directory '/config/zigbee2mqtt/log/2022-10-25.20-10-42'
Zigbee2MQTT:info  2022-10-25 22:43:45: Starting Zigbee2MQTT version 1.28.0 (commit #unknown)
Zigbee2MQTT:info  2022-10-25 22:43:45: Starting zigbee-herdsman (0.14.62)
Zigbee2MQTT:debug 2022-10-25 22:43:45: Using zigbee-herdsman with settings: '{"adapter":{"concurrent":null,"delay":null,"disableLED":false},"backupPath":"/config/zigbee2mqtt/coordinator_backup.json","databaseBackupPath":"/config/zigbee2mqtt/database.db.backup","databasePath":"/config/zigbee2mqtt/database.db","network":{"channelList":[17],"extendedPanID":[221,221,221,221,221,221,221,221],"networkKey":"HIDDEN","panID":6754},"serialPort":{"adapter":"ezsp","path":"/dev/ttyUSB0"}}'
2022-10-25T14:43:46.039Z zigbee-herdsman:controller:log Starting with options '{"network":{"networkKeyDistribute":false,"networkKey":[1,3,5,7,9,11,13,15,0,2,4,6,8,10,12,13],"panID":6754,"extendedPanID":[221,221,221,221,221,221,221,221],"channelList":[17]},"serialPort":{"path":"/dev/ttyUSB0","adapter":"ezsp"},"databasePath":"/config/zigbee2mqtt/database.db","databaseBackupPath":"/config/zigbee2mqtt/database.db.backup","backupPath":"/config/zigbee2mqtt/coordinator_backup.json","adapter":{"disableLED":false,"concurrent":null,"delay":null}}'
2022-10-25T14:43:46.042Z zigbee-herdsman:adapter:ezsp:uart Opening SerialPort with /dev/ttyUSB0 and {"baudRate":115200,"rtscts":false,"autoOpen":false}
2022-10-25T14:43:46.051Z zigbee-herdsman:adapter:ezsp:uart Serialport opened
2022-10-25T14:43:46.051Z zigbee-herdsman:adapter:ezsp:uart Uart reseting
2022-10-25T14:43:46.051Z zigbee-herdsman:adapter:ezsp:uart --> Write reset
2022-10-25T14:43:46.053Z zigbee-herdsman:adapter:ezsp:uart --> [1ac038bc7e]

algovera, 2022-10-26:
Actually this is resolved after setting a high pan_id

https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start.html#error-coordinator-failed-to-start-probably-the-panid-is-already-in-use-try-a-different-panid-or-channel

algovera, 2022-10-26:
let me try reflashing and see

algovera, 2022-10-31:
tried re-flashing and still no luck
can anyone help?

Glazzanz, 2022-11-25:
Ill add a unhelpful "me too" comment.

closed 2023-01-02 by github-actions[bot]