Koenkk/zigbee2mqtt #14648: TuYa sensor (ZG-204ZL) Illuminance Unavailable in Home Assistant

KirovAir, 2022-10-25:
### What happened?

Hi!

It seems that illuminance stays on 'unavailable' within home assistant. It's working fine in the z2m dashboard and it's also reporting values fine as well within the home assistant json logs. So maybe a mapping error?
[State in home assistant](https://www.imgdumper.nl/uploads9/6358002d05b97/6358002d02ec0-Screenshot_2022-10-25_at_17.26.16.png)
[Received messages in home assistant](https://www.imgdumper.nl/uploads9/635800e7e3433/635800e7dca7e-Screenshot_2022-10-25_at_17.29.32.png)

Log from z2m when sensor is activated:
```
Info 2022-10-25 17:33:23MQTT publish: topic 'zigbee2mqtt/Kitchen Sensor', payload '{"battery":100,"illuminance":292,"keep_time":"30","linkquality":91,"occupancy":true,"sensitivity":"high"}'
```

### What did you expect to happen?

I would expect the Illuminance sensor to be displaying a value. 
Currently the state from home assistant is: `This entity is unavailable.`

### How to reproduce it (minimal and precise)

Simply add the sensor and try to use the value in home assistant.

### Zigbee2MQTT version

1.28.0

### Adapter firmware version

20220219

### Adapter

Sonoff Dongle-P

### Debug log

```
Debug 2022-10-25 17:33:23Received Zigbee message from 'Kitchen Sensor', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0],"type":"Buffer"},"datatype":4,"dp":1}],"seq":12032}' from endpoint 1 with groupID 0
Info 2022-10-25 17:33:23MQTT publish: topic 'zigbee2mqtt/Kitchen Sensor', payload '{"battery":100,"illuminance":292,"keep_time":"30","linkquality":91,"occupancy":true,"sensitivity":"high"}'
Debug 2022-10-25 17:33:26Received Zigbee message from 'Kitchen Sensor', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0,0,0,244],"type":"Buffer"},"datatype":2,"dp":12}],"seq":12288}' from endpoint 1 with groupID 0
Info 2022-10-25 17:33:26MQTT publish: topic 'zigbee2mqtt/Kitchen Sensor', payload '{"battery":100,"illuminance":244,"keep_time":"30","linkquality":102,"occupancy":true,"sensitivity":"high"}'
Debug 2022-10-25 17:33:57Received Zigbee message from 'Kitchen Sensor', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0,0,1,24],"type":"Buffer"},"datatype":2,"dp":12}],"seq":12544}' from endpoint 1 with groupID 0
Info 2022-10-25 17:33:57MQTT publish: topic 'zigbee2mqtt/Kitchen Sensor', payload '{"battery":100,"illuminance":280,"keep_time":"30","linkquality":105,"occupancy":true,"sensitivity":"high"}'
Debug 2022-10-25 17:33:57Received Zigbee message from 'Kitchen Sensor', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[1],"type":"Buffer"},"datatype":4,"dp":1}],"seq":12800}' from endpoint 1 with groupID 0
Info 2022-10-25 17:33:57MQTT publish: topic 'zigbee2mqtt/Kitchen Sensor', payload '{"battery":100,"illuminance":280,"keep_time":"30","linkquality":102,"occupancy":false,"sensitivity":"high"}'
```

KirovAir, 2022-10-26:
Hmm, it seems I am an idiot. For some reason the entity was disabled by default in Home Assistant. By simply enabling it it started to work. Sorry for wasting anyone's time.

closed 2022-10-26 by KirovAir

DrMikeyS, 2023-06-29:
I think this is worth reopening as an issue. @KirovAir - your fix was the right one, but it is not obvious to me why it was disabled by default. I only figured out how to fix it based on your comment.

aleSaleKa, 2023-06-29:
This thread helped me too.
Added this device through z2mqtt and i could not see the illuminance in homeassistant entities.
I did as OP said, went to home assistant devices -> entities -> entered my new device entity name fragment and applied filter to show disabled entities. I enbled the entity and then i could user and see it in HA dashboards.
Thanks!
i thought this was z2mqtt or HA bug.

benin-work, 2023-09-16:
Thanks, this definitely helped me find out the reason I didn't have illuminance sensor data available in Home Assistant.

cdensel, 2024-01-11:
> Hmm, it seems I am an idiot. For some reason the entity was disabled by default in Home Assistant. By simply enabling it it started to work. Sorry for wasting anyone's time.

Thanks! You helped me with this too.

imthernet, 2024-08-01:
Thanks for this thread. Helped me a lot!